aws/amazon-freertos #826: Issue with Delta Callback function Registered to Update Topic

tombresson, 2019-06-21:
**Describe the bug**
In v1 of the Shadow library, the changes that were made caused the Delta Callback to be registered to the Update Topic. This was due to the variable `callbackInfo` being passed into the `AwsIotShadow_SetUpdatedCallback` function instead of the pointer, `pCallbackInfo`.

**System information**
- Version of Amazon FreeRTOS: master (c334212b824)

**Expected behavior**

Delta callback should only be called with Delta topic data, not with Update topic data

**Additional context**

See PR #825 for the change that needs to be made.

dnoal, 2019-06-24:
Tom, Thanks very much for this.  We are working on merging the PR.

alexa-noxon, 2019-06-25:
Closing the issue as #825 has been merged.   Thanks for pointing this out.

closed 2019-06-25 by alexa-noxon