aws/amazon-freertos #831: Microchip curiosity demo WiFi only?

htibosch, 2019-06-22:
**Describe the bug**

The demo can not work with the Ethernet adapter anymore.
Somewhere it says:

    #define configSUPPORTED_NETWORKS    ( AWSIOT_NETWORK_TYPE_WIFI  )

Has the Microchip curiosity demo become WiFi only? Why is that?

**System information**

Board: pic32mz_ef_curiosity with both WiFi and Ethernet
MPLAB X
Today's version ( june 22, 2019 )
Sub pub Demo
I did define `PIC32_USE_ETHERNET`, just like before.

Also I adapted `aws_iot_network_config.h`:

    #define configSUPPORTED_NETWORKS    ( AWSIOT_NETWORK_TYPE_WIFI | AWSIOT_NETWORK_TYPE_ETH )
    #if defined(PIC32_USE_ETHERNET)
        #define configENABLED_NETWORKS      ( AWSIOT_NETWORK_TYPE_ETH )
    #else
        #define configENABLED_NETWORKS      ( AWSIOT_NETWORK_TYPE_WIFI )
    #endif

Also I added `NetworkInterface_eth.c` to the project.

**Expected behavior**

That the Ethernet adapter can still be used.

**Screenshots or console output**

These messsages are written to the debug port:

    [iot_thread] [INFO ][INIT][231] SDK successfully initialized.
    [iot_thread] [INFO ][DEMO][231] No networks connected for the demo.

Thanks, Hein
        Waiting for a network connection.

htibosch, 2019-06-22:
Some text got mixed-up. The output was:

    [iot_thread] [INFO ][INIT][231] SDK successfully initialized.
    [iot_thread] [INFO ][DEMO][231] No networks connected for the demo. Waiting for a network connection.

alexa-noxon, 2019-06-28:
Hi Hein,

Thanks for pointing this out.  I am also encountering this issue and will check on the status of the Ethernet port for PIC32.  

Alexa

sarenameas, 2019-07-06:
Hello Hein,

I have made the updates to all Microchip projects and CMAKE configuration to support Ethernet again. It was missed in the directory restructuring. Please see  https://github.com/aws/amazon-freertos/pull/907.

Thansk for bringing this to the light!

sarenameas, 2019-07-08:
Closing this issue, since the fix has been merged to master.

closed 2019-07-08 by sarenameas