aws/amazon-freertos #841: Error while running 'python SetupAWS.py setup'

KunjalRupala, 2019-06-25:
Hey,
I am new to AWS and am trying to get the latest version of FreeRTOS into my windows system to operate along with ESP32-DevKitC. Whenever I try to run the 'python SetupAWS.py setup' to run the configuration script, it shows me the following error:

[image]

The content of my 'configure.json' file is as follows:
```cpp
{
    "afr_source_dir":"C:/AmazonFreeRTOS/",
    "thing_name":"first",
    "wifi_ssid":"XXXXXXX",
    "wifi_password":"XXXXXXXX",
    "wifi_security":"eWiFiSecurityWPA2"
}
``` 
My 'aws configure' is also in sync with my access keys and I have no thing with the same name

Please help me resolve the issue.

Thank you!

tgsong, 2019-06-25:
Hi, can you double check if you're looking at the correct region? You can also try list things from AWS CLI by running `aws iot list-things` from your commandline. This should give you the same result, the python script is just calling the AWS SDK under the hood.

KunjalRupala, 2019-06-26:
Hey tgsong,
I have double checked my region its the same. In addition to that i also tried ```aws iot list-things``` as suugested by you but still it gives me the following error:

```An error occurred (UnrecognizedClientException) when calling the ListThings operation: The security token included in the request is invalid.```

sheldon123z, 2019-06-27:
Hi, exactly the same errors happened here. Any suggestions ? I have fixed the buggy code mentioned in the [f9c987a](https://github.com/aws/amazon-freertos/commit/f9c987a4537a7f4abd684348c5194ed672eaaadf) but the same error still happens.

alexa-noxon, 2019-06-27:
Do you have multi-factor authentication enabled for your account? 
If so, maybe this article could help https://aws.amazon.com/premiumsupport/knowledge-center/authenticate-mfa-cli/

One more thing to check might be whether the user that you set during aws configure has permissions to perform all the actions (For example, I have a user with 'AWSIoTFullAccess'), although I would expect a message along the lines of User "is not authorized to perform: iot:ListThings on resource" if that were the case.

alexa-noxon, 2019-07-02:
One more idea- I was able to replicate the error message you are getting by making the user who's credentials I was using in IAM "Inactive"

[image]

Can you also check that your IAM user is "active"?

If you are not using MFA and your user is 'active' with permissions that include the "AWSIoTFullAccess", try upgrading to the most recent aws cli version.

Let me know if this fixes your issue!

ravibhagavandas, 2019-07-22:
Hi @KunjalRupala,

Does the suggestions fixed the issue ?

pvyawaha, 2019-07-30:
Hi KunjalRupala,

Closing this issue now, please reopen if you have more questions.

closed 2019-07-30 by pvyawaha