aws/amazon-freertos #842: .gitignore doesn't work. and outdated

povi2946, 2019-06-25:
Hi, it seems that .gitignore does work for AWS files and folders. It works only on files created additionally. 

Also to note that 
demos/include/aws_clientcredential.h
directory changed from 
demos/common/include/aws_clientcredential.h
So, .gitignore file paths should be updated. 

My goal is to add credentials files to .gitignore but it doesn't work.
Thank you.

tgsong, 2019-06-26:
Hi, thanks for reporting this, our `.gitignore` file has been modified by mistake, we will roll back to what it was before. It shouldn't reference any file that's already tracked by git.

alexa-noxon, 2019-06-29:
I have revised our .gitignore paths and modified/added new .gitignores to match with the new directory structure.

Thanks again for pointing this out.

closed 2019-06-29 by alexa-noxon