aws/amazon-freertos #847: Issue while generating build file

KunjalRupala, 2019-06-26:
Hey,
Whenever I try to generate the demo application's build files with CMake for ESP32 in Ubuntu using the following command I get a CMake error.

```cpp
cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B your-build-directory
```
**console output**
[image]

Please help me resolve the issue.

Thank you!

alexa-noxon, 2019-06-26:
Hi Kunjal,

If the path to the toolchain is not in your path, you can add this as an option in the cmake build by adding 
`-DAFR_TOOLCHAIN_PATH=/path/to/toolchain/`
to the build command.

For example, 
`cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -S . -B build -DAFR_TOOLCHAIN_PATH=/opt/xtensa-esp32-elf/`

I will update our documentation to provide an example of specifying the tool chain path using the command line.

Does this resolve the issue?

Thanks,
Alexa

KunjalRupala, 2019-06-27:
Hey Alexa,
Thanks for your reply but the Cmake errors still persist. Here, I am attaching a screenshot of the issue.

[image]

alexa-noxon, 2019-06-27:
I notice that you are running from a folder called amazon-freertos-master, but it seems that the script is looking for the toolchain in amazon-freertos/tools/cmake/toolchains/xtensa-esp32.cmake instead of amazon-freerotos-master/tools/cmake/toolchains/xtensa-esp32.cmake.  The CMake build should determine the correct relative paths, but in your case it looks like there is some issue.

If you make a new clone into "testFolder" and then call that same command from inside the folder, do you still have the same issue?

`git clone git@github.com:aws/amazon-freertos.git testFolder`   
`cd testFolder`   
`cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -S . -B build -DAFR_TOOLCHAIN_PATH=/esp/xtensa-esp32-elf/`

KunjalRupala, 2019-06-28:
Thank you very much. It worked for me!

ahtashambaig, 2019-12-26:
Hi 
I am working Amazon FreeRTOS , i am new to Amazon Web services .. i am following the documentation
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html
from scratch i have been following each step ... the issue i am getting after creating build directory using
`cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -GNinja -S . -B your-build-directory`
the build directory is created but when i try later on steps as for my OS I am using Windows10 Which donot support 

> make 

command i used the compiler Ninja but when i try steps as below in the screenshot
[image]
i get following errors:
[image]

KunjalRupala, 2020-03-08:
I am getting the error again. It was working absolutely fine and I dont know how but it started again.
[image]

closed 2020-03-08 by KunjalRupala