aws/amazon-freertos #848: heap usage release 1.5 vs ble-beta branch

chegewara, 2019-06-26:
Hi,
i have esp32 app for client that i wrote using ble-test branch and it works fine, even with custom BLE scan and client functionality. Recently client asked me to port it to release 1.5. I did it, but for my surprise i found that i cant run it anymore because of short of heap.
To have as much the same code running during test i am performing ble de-init and memory release and then testing memory usage. Here is result:
1. ble-beta branch:
```
Heap summary for capabilities 0x00001000:
  At 0x3ffbdb28 len 9432 free 0 allocated 9328 min_free 0
    largest_free_block 0 alloc_blocks 18 free_blocks 0 total_blocks 18
  At 0x3ffb8000 len 15144 free 8348 allocated 6728 min_free 4728
    largest_free_block 6548 alloc_blocks 7 free_blocks 2 total_blocks 9
  At 0x3ffb0000 len 10032 free 9956 allocated 36 min_free 9956
    largest_free_block 9956 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3ffae6e0 len 6192 free 6116 allocated 36 min_free 6116
    largest_free_block 6116 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3ffbbb28 len 8192 free 7552 allocated 560 min_free 0
    largest_free_block 6120 alloc_blocks 9 free_blocks 3 total_blocks 12
  At 0x3ffb2730 len 22736 free 20252 allocated 2312 min_free 16
    largest_free_block 17428 alloc_blocks 24 free_blocks 11 total_blocks 35
  At 0x3ffaff10 len 240 free 0 allocated 196 min_free 0
    largest_free_block 0 alloc_blocks 3 free_blocks 0 total_blocks 3
  At 0x3ffd7f38 len 32968 free 3984 allocated 28544 min_free 0
    largest_free_block 3940 alloc_blocks 97 free_blocks 5 total_blocks 102
  At 0x3ffe0440 len 15296 free 180 allocated 15032 min_free 0
    largest_free_block 112 alloc_blocks 11 free_blocks 2 total_blocks 13
  At 0x3ffe4350 len 113840 free 108008 allocated 5780 min_free 86876
    largest_free_block 108008 alloc_blocks 4 free_blocks 1 total_blocks 5
  Totals:
    free 164396 allocated 68552 min_free 107692 largest_free_block 108008
```
2. master branch with release 1.5 merged:
```
Heap summary for capabilities 0x00001000:
  At 0x3ffbdb28 len 9432 free 4588 allocated 4640 min_free 0
    largest_free_block 4448 alloc_blocks 38 free_blocks 5 total_blocks 43
  At 0x3ffb8000 len 15144 free 448 allocated 14608 min_free 0
    largest_free_block 448 alloc_blocks 13 free_blocks 1 total_blocks 14
  At 0x3ffb0000 len 10032 free 836 allocated 9116 min_free 0
    largest_free_block 480 alloc_blocks 9 free_blocks 3 total_blocks 12
  At 0x3ffae6e0 len 6192 free 3980 allocated 2048 min_free 0
    largest_free_block 1060 alloc_blocks 26 free_blocks 7 total_blocks 33
  At 0x3ffbbb28 len 8192 free 3376 allocated 4716 min_free 0
    largest_free_block 2332 alloc_blocks 13 free_blocks 4 total_blocks 17
  At 0x3ffb2730 len 22736 free 164 allocated 22348 min_free 0
    largest_free_block 28 alloc_blocks 39 free_blocks 9 total_blocks 48
  At 0x3ffaff10 len 240 free 0 allocated 196 min_free 0
    largest_free_block 0 alloc_blocks 3 free_blocks 0 total_blocks 3
  At 0x3ffd9150 len 28336 free 88 allocated 27880 min_free 0
    largest_free_block 84 alloc_blocks 82 free_blocks 2 total_blocks 84
  At 0x3ffe0440 len 15296 free 1188 allocated 14036 min_free 0
    largest_free_block 788 alloc_blocks 7 free_blocks 3 total_blocks 10
  At 0x3ffe4350 len 113840 free 53024 allocated 60508 min_free 38440
    largest_free_block 43212 alloc_blocks 55 free_blocks 14 total_blocks 69
  Totals:
    free 67492 allocated 160288 min_free 38440 largest_free_block 43212
```
I know i can try to optimize tasks stack a bit and recover about 5-10kB and i have 2 running `iot_thread` that does nothing (stack 5kB each), but difference is 100kB between those 2 branches.

alexa-noxon, 2019-06-26:
Hi chegewara,

The differences between ble-beta and the most recent release is primarily directory/folder restructuring, and the addition of cmake, along with minor bug fixes and code updates.   

I wouldn't expect any of these changes to affect the memory usage significantly.  Based on the magnitude of the heap usage difference, one possibility is that the new application could be creating a new connection where it wasn't before.  

I am assuming that you previously had a BLE connection on the application based on the ble-beta branch, and that you still have that connection on the new merged application?   If that is the case, can you also verify that you are not creating a TLS connection that did not exist before?  TLS handshakes are one of the most memory-intense actions that the devices undertake.

Best,
Alexa

chegewara, 2019-06-26:
Hi,
like i mentioned, i am killing BLE before i am checking memory usage, because i know that esp-idf is using more and more heap with every release, so its not that.
I also know that TLS initial connection needs about 70kB but most of it should be released after successful handshake, and i am trying to use only 1 mqtt connection over whole app, i think it was done in ble-beta in all examples, and in release-1.5 i moved establishing connection to main.c, but i am short of heap even before i reach IotMqtt_Connect, just right after init AWS_Sdk and network manager. 
Im not saying i fully understand AFR and i know i am still making mistakes using it. Thats why i am asking here for help, unless there is better place to do it. Thanks.

What i am doing is to call `DEMO_RUNNER_RunDemos()` and then this changed iot_demo_afr.c:
```
/**
 * @file iot_demo_afr.c
 * @brief Generic demo runner for C SDK libraries on Amazon FreeRTOS.
 */

/* The config header is always included first. */
#include "iot_config.h"

#include <string.h>
#include "aws_clientcredential.h"
#include "aws_clientcredential_keys.h"
#include "iot_demo_logging.h"
#include "iot_network_manager_private.h"
#include "platform/iot_threads.h"
#include "aws_demo.h"
#include "iot_init.h"

/* Remove dependency to MQTT */
#define MQTT_DEMO_TYPE_ENABLED (defined(CONFIG_MQTT_DEMO_ENABLED)||defined(CONFIG_SHADOW_DEMO_ENABLED)||defined(CONFIG_DEFENDER_DEMO_ENABLED)||defined(CONFIG_OTA_UPDATE_DEMO_ENABLED))

#if MQTT_DEMO_TYPE_ENABLED
#include "iot_mqtt.h"
#endif

static IotNetworkManagerSubscription_t subscription = IOT_NETWORK_MANAGER_SUBSCRIPTION_INITIALIZER;

/* Semaphore used to wait for a network to be available. */
static IotSemaphore_t demoNetworkSemaphore;

/* Variable used to indicate the connected network. */
static uint32_t demoConnectedNetwork = AWSIOT_NETWORK_TYPE_NONE;

#if MQTT_DEMO_TYPE_ENABLED
#if BLE_ENABLED 
extern const IotMqttSerializer_t IotBleMqttSerializer;
#endif

/*-----------------------------------------------------------*/

const IotMqttSerializer_t * demoGetMqttSerializer( void )
{
    const IotMqttSerializer_t * ret = NULL;

#if BLE_ENABLED
    if( demoConnectedNetwork == AWSIOT_NETWORK_TYPE_BLE )
    {
        ret = &IotBleMqttSerializer;
    }
#endif

    return ret;
}
#endif
/*-----------------------------------------------------------*/

void generateDeviceIdentifierAndMetrics( void );

static uint32_t _getConnectedNetworkForDemo( demoContext_t * pDemoContext )
{
    uint32_t ret = ( AwsIotNetworkManager_GetConnectedNetworks() & pDemoContext->networkTypes );

    if( ( ret & AWSIOT_NETWORK_TYPE_WIFI ) == AWSIOT_NETWORK_TYPE_WIFI )
    {
[... 497 lines cut ...]

/*
 * @brief Retrieves the unique identifier for the device.
 */
const char *getDeviceIdentifier( void )
{
    return deviceIdentifier;
}

/*
 * @brief Retrieves the device metrics to be sent to cloud.
 */
const char *getDeviceMetrics( void )
{
    return deviceMetrics;
}

/*
 * @brief Get the device metrics length.
 */
uint16_t getDeviceMetricsLength( void )
{
    return ( uint16_t )( strlen( deviceMetrics ) );
}
```

chegewara, 2019-06-26:
This is for example ble-test components size:
```
Total sizes:
 DRAM .data size:   37212 bytes
 DRAM .bss  size:   60888 bytes
Used static DRAM:   98100 bytes (  17100 available, 85.2% used)
Used static IRAM:  103010 bytes (  28062 available, 78.6% used)
      Flash code:  865586 bytes
    Flash rodata:  208552 bytes
Total image size:~1214360 bytes (.bin may be padded larger)
Per-archive contributions to ELF file:
            Archive File DRAM .data & .bss   IRAM Flash code & rodata   Total
                 libbt.a        522  26357    909     275494    82522  385804
           libnet80211.a        322   8999   3629     119148    14193  146291
libamazon-freertos-commo       2228  10002    212      81390    24590  118422
            libmbedtls.a         56     72      0      74741    14679   89548
                 libpp.a       1229   5286  13471      45803     4208   69997
           libbtdm_app.a        367   4745  20467      40199     3409   69187
                libphy.a       1284    915   5710      29779        0   37688
                libtft.a      12064     77   3554      11105    10309   37109
              libesp32.a       2387    313   7894      12271    10962   33827
             libdriver.a         35    190   4824      19200     7936   32185
             libstockd.a      10916    235     72      10172     4285   25680
        libsmartconfig.a         24    463      0      21984     2593   25064
                libwpa.a          0    682      0      21004     2312   23998
           libfreertos.a       2064    988  15231          0     1777   20060
             libc_nano.a          0      0     12      16689     2825   19526
              libfatfs.a          4     47      0      16029     1378   17458
     libwpa_supplicant.a          0      0      0      12957     2318   15275
          libnvs_flash.a          0     32      0      11483     2962   14477
              libsdmmc.a          0      0      0       6325     2839    9164
          libspi_flash.a         32    317   6291        929      588    8157
                libsoc.a        705      8   3783          0     3583    8079
          libspidriver.a          3     33   3118       4120      486    7760
               libheap.a        916      8   3729       1806     1072    7531
            libcoexist.a       1365    115   4368       1654        0    7502
                libgcc.a          4     20    104       5548      888    6564
         libstockd_ble.a        180     25      0       2915     2376    5496
                  libm.a          0      0     84       4920      472    5476
                libvfs.a        232    103      0       3626      494    4455
             libstdc++.a          8     20      0       2649     1253    3930
            libpthread.a          0     16    269       1692      772    2749
                librtc.a          0      4   2480          0        0    2484
             libnewlib.a        144    272    825        925      229    2395
                liblog.a          8    268    464        703      158    1601
               libcore.a          0      5      0        813      418    1236
         libapp_update.a          0      4      0        168      773     945
    libsmartconfig_ack.a          0      1      0        649      288     938
                libhal.a          0      0    515          0       32     547
      libtcpip_adapter.a          0     36      0        241       12     289
               libmesh.a          2      1      0         44        0      47
                libcxx.a          0      0      0         11        0      11
               libwpa2.a          0      1      0          0        0       1
                libwps.a          0      1      0          0        0       1
           libethernet.a          0      0      0          0        0       0
 libbootloader_support.a          0      0      0          0        0       0
```
and master components size:
```
Total sizes:
 DRAM .data size:   37308 bytes
 DRAM .bss  size:   65424 bytes
Used static DRAM:  102732 bytes (  12468 available, 89.2% used)
Used static IRAM:  103612 bytes (  27460 available, 79.0% used)
      Flash code:  957894 bytes
    Flash rodata:  243780 bytes
Total image size:~1342594 bytes (.bin may be padded larger)
Per-archive contributions to ELF file:
            Archive File DRAM .data & .bss   IRAM Flash code & rodata   Total
                 libbt.a        554  26347    805     256513    81962  366181
[... 21 lines cut ...]
         libstockd_ble.a          0     52      0       3928     3450    7430
               libheap.a        896      8   3623       1730     1032    7289
                libgcc.a          4     20    104       5552      888    6568
                  libm.a          0      0     84       4920      472    5476
            libcoexist.a        769      0   4181        139       31    5120
                libvfs.a        232    103      0       3742      454    4531
             libstdc++.a          8     20      0       2649     1253    3930
                librtc.a          0      4   2381          0        0    2385
             libnewlib.a        144    272    730        925      189    2260
                liblog.a          8    268    438        670      158    1542
            libpthread.a          0     12    269        501      704    1486
               libcore.a          0      5      0        813      418    1236
[... cut ...]
libxtensa-debug-module.a          0      0      8          0        0       8
               libwpa2.a          0      1      0          0        0       1
                libwps.a          0      1      0          0        0       1
           libethernet.a          0      0      0          0        0       0
 libbootloader_support.a          0      0      0          0        0       0
```
If you compare just few components size you can see difference, so its not just a folders restructuring changes.

alexa-noxon, 2019-06-28:
Hi chegewara,

Just wanting to confirm the symptom that you are seeing - are you seeing the code hitting the vApplicationMallocFailedHook, or pvPortMalloc calls are returning a NULL pointer, correct?  And it is not an issue that you aren't able to provide enough memory off the FreeRTOS heap for your task's stack, and are seeing a stack overflow?  

I can see that in the code you have pasted in the previous post, you have removed the invocation of the demo function, as well as the call to _cleanup() in runDemoTask(). In your previous demo, were allowing the demos to complete and the _cleanup() function to be called before measuring heap usage?  Some heap loss (not 100k, but still a significant amount) might not be freed due to this.

Can you summarize the call sequence that is being made from your project first in the ble-beta branch, and then secondly from your new project based on the new release, and exactly where in that sequence you are collecting the measurements of remaining heap from your first post.  I'm not fully sure that I am fully understanding the order that these calls are being made in, and it would help me to understand which initialization and deinitialization functions are being invoked to see if we can find where this memory is not being released.

Thank you,
Alexa

chegewara, 2019-06-28:
Hi,
i appreciate help and i will try to provide all info you need:
1. no, i dont have problems with malloc, at least i dont see logs that would suggest it (btw, in other project i had to move `vApplicationStackOverflowHook` to main.c and to use ESP_LOGX because app crashed before `vApplicationStackOverflowHook` could print any logs)
2. yes, what i want is that runDemoTask to initialize only IotSdk and network, basically all in _initialize() function
3. i dont call _cleanup, because i want to to IotSdk_Cleanup, i dont know yet i can to call it if i want to use mqtt after then, so i will check, and i want to reuse network connect and disconnect callbacks
4. here is main.c without calling client code, should be used with changes from my previous post

```
/*
 * Amazon FreeRTOS V1.4.7
 * Copyright (C) 2018 Amazon.com, Inc. or its affiliates.  All Rights Reserved.
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy of
 * this software and associated documentation files (the "Software"), to deal in
 * the Software without restriction, including without limitation the rights to
 * use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of
 * the Software, and to permit persons to whom the Software is furnished to do so,
 * subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 * http://aws.amazon.com/freertos
 * http://www.FreeRTOS.org
 */

#include "iot_config.h"

/* FreeRTOS includes. */

#include "FreeRTOS.h"
#include "task.h"

/* Demo includes */
#include "aws_demo.h"
#include "aws_dev_mode_key_provisioning.h"

/* AWS System includes. */
#include "bt_hal_manager.h"
#include "aws_system_init.h"
#include "aws_logging_task.h"

#include "nvs_flash.h"

#include "FreeRTOS_Sockets.h"

#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_interface.h"
#include "esp_gap_ble_api.h"
#include "esp_bt.h"
#include "esp_bt_main.h"
#include "esp_log.h"

#include "iot_mqtt.h"
#include "aws_iot_shadow.h"
#include "driver/uart.h"
#include "aws_application_version.h"
#include "iot_demo_logging.h"
#include "aws_ota_agent.h"
#include "platform/iot_clock.h"

#include "iot_network_manager_private.h"
[... 470 lines cut ...]
}

static void _tasksStatus(void* p)
{
    UBaseType_t uxArraySize = uxTaskGetNumberOfTasks();
    TaskStatus_t pxTaskStatusArray[uxArraySize];
    uint32_t pulTotalRunTime = 0;

    uxTaskGetSystemState(&pxTaskStatusArray[0], uxArraySize, &pulTotalRunTime);

    for(int i=0; i<uxArraySize; i++)
    {
        ESP_LOGI(__func__, "[%d]Task %s => stack remaining: %d", i, pxTaskStatusArray[i].pcTaskName, pxTaskStatusArray[i].usStackHighWaterMark);
    }
    heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
    ESP_LOGI(__func__, "stack remaining: %d", uxTaskGetStackHighWaterMark( NULL ));
    vTaskDelete(NULL);
}

void tasksStatus()
{
    xTaskCreate(_tasksStatus, "stackInfo", 5*612, NULL, 1, NULL);
}

```

chegewara, 2019-06-28:
Like i suspected, i dont want to call _cleanup and IotSdk_Cleanup:
```
9 354 [iot_thread] [INFO ][INIT][3540] SDK cleanup done.
10 1334 [conT] [ERROR][NET][13340] Failed to establish new connection.
11 1334 [conT] [ERROR][MQTT][13340] Failed to establish new MQTT connection, error NETWORK ERROR.
12 1334 [conT] [ERROR][DEMO][13340] MQTT CONNECT returned error NETWORK ERROR.
I (14713) PKCS11: Initializing NVS partition: "storage"
13 1921 [conT] [INFO ][MQTT][19200] Establishing new MQTT connection.
14 1921 [conT] [WARN ][MQTT][19210] (MQTT connection 0x3ffb1940, CONNECT operation 0x3ffbba60) Failed to schedule operation job, error SHUTDOWN IN PROGRESS.
15 1921 [conT] [ERROR][MQTT][19210] Failed to enqueue CONNECT for sending.
16 1921 [conT] [ERROR][MQTT][19210] Failed to establish new MQTT connection, error SCHEDULING ERROR.
17 1922 [conT] [INFO ][MQTT][19220] Network connection closed on error.
18 1926 [conT] [INFO ][MQTT][19260] (MQTT connection 0x3ffb1940) Network connection destroyed.
19 1926 [conT] [ERROR][DEMO][19260] MQTT CONNECT returned error SCHEDULING ERROR.
```
also this free not so much heap.

chegewara, 2019-07-01:
Ok, i did one more test and release-1.5 without client code and with BLE (no release BLE memory) this is the result:
```
  Totals:
    free 116020 allocated 117480 min_free 109280 largest_free_block 110740
```

Looks like i have to dig more. I will close this issue just for now.

chegewara, 2019-07-01:
Hi @alexa-noxon 
I am trying to understand, is it possible memory is not free from TLS after connecting to mqtt?

this is memory right before calling `IotMqtt_Connect`:
```
  Totals:
    free 124620 allocated 108976 min_free 111124 largest_free_block 113804
```
this is memory after connecting to mqtt:
```
  Totals:
    free 52348 allocated 180676 min_free 43332 largest_free_block 39692
```

sarenameas, 2019-07-02:
TLS retains some memory for the duration if the connection, until it is closed. 
MQTT has some connection context allocated as well post a successful call.

Notice that you declare your `xTaskCreate(connectTask, "connectT", 4*1024, NULL, 5, NULL);` to be double the size of the original demo one that we had. Confirm is that is correct for your case; if you are trying to save memory I just wanted to point that out.

Also noticed that you initialize the library AFTER you try to connect:
```
    status = _establishMqttConnection( clientcredentialIOT_THING_NAME, pConnectionParams, pCredentials, pNetworkInterface, &mqttConnection );

    if(status == EXIT_SUCCESS)
        status = _initializeMqtt();
    {
        ESP_LOGE(__func__, "Failed to establish connection");
    }
```
You need to call `IotMqtt_Init()` first before any other MQTT API.

chegewara, 2019-07-03:
> Notice that you declare your xTaskCreate(connectTask, "connectT", 4*1024, NULL, 5, NULL); to be double the size of the original demo one that we had. Confirm is that is correct for your case; if you are trying to save memory I just wanted to point that out.

Yes, because logging on esp32 requires higher stack, but i delete this task anyway after successful connection.

> Also noticed that you initialize the library AFTER you try to connect:
You need to call IotMqtt_Init() first before any other MQTT API.

I changed order, but result is similar.

I did 1 step back and did the same in ble-beta. Here is the result:
```
    heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
    xMqttStatus = AwsIotMqtt_Connect(
            &( xConnection.xMqttConnection ),
            &( xConnection.xNetworkInterface ),
            &xConnectInfo,
            NULL,
            echoDemoMQTT_OPERATION_TIMEOUT_MS );
    heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
```
```
Heap summary for capabilities 0x00001000:
  At 0x3ffbdb28 len 9432 free 20 allocated 9292 min_free 0
    largest_free_block 20 alloc_blocks 21 free_blocks 1 total_blocks 22
  At 0x3ffb8000 len 15144 free 252 allocated 14748 min_free 0
    largest_free_block 248 alloc_blocks 26 free_blocks 2 total_blocks 28
  At 0x3ffb0000 len 10032 free 4824 allocated 5072 min_free 0
    largest_free_block 3264 alloc_blocks 20 free_blocks 6 total_blocks 26
  At 0x3ffae6e0 len 6192 free 2704 allocated 3372 min_free 0
    largest_free_block 2676 alloc_blocks 18 free_blocks 3 total_blocks 21
  At 0x3ffbbb28 len 8192 free 48 allocated 8000 min_free 0
    largest_free_block 28 alloc_blocks 26 free_blocks 2 total_blocks 28
  At 0x3ffb2730 len 22736 free 1476 allocated 21000 min_free 4
    largest_free_block 648 alloc_blocks 43 free_blocks 14 total_blocks 57
  At 0x3ffaff10 len 240 free 0 allocated 196 min_free 0
    largest_free_block 0 alloc_blocks 3 free_blocks 0 total_blocks 3
  At 0x3ffd8190 len 32368 free 592 allocated 31356 min_free 0
    largest_free_block 384 alloc_blocks 93 free_blocks 4 total_blocks 97
  At 0x3ffe0440 len 15296 free 344 allocated 14884 min_free 0
    largest_free_block 344 alloc_blocks 8 free_blocks 1 total_blocks 9
  At 0x3ffe4350 len 113840 free 66728 allocated 46948 min_free 56720
    largest_free_block 58252 alloc_blocks 21 free_blocks 12 total_blocks 33
  Totals:
    free 76988 allocated 154868 min_free 56724 largest_free_block 58252
4 6833 [MQTTEcho] [INFO ][NET][6833] New network connection established.
5 6952 [MQTTEcho] [INFO ][MQTT][6952] Establishing new MQTT connection.
6 6954 [MQTTEcho] [INFO ][MQTT][6954] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
7 6954 [MQTTEcho] [INFO ][MQTT][6954] Waiting for operation CONNECT to complete.
Heap summary for capabilities 0x00001000:
  At 0x3ffbdb28 len 9432 free 4 allocated 9304 min_free 0
    largest_free_block 4 alloc_blocks 22 free_blocks 1 total_blocks 23
8 7310 [MQTTEcho] [INFO ][MQTT][7310] MQTT operation CONNECT complete with result SUCCESS.
9 7310 [MQTTEcho] [INFO ][MQTT][7310] New MQTT connection 0x3ffb2528 established.
  At 0x3ffb8000 len 15144 free 152 allocated 14840 min_free 0
    largest_free_block 128 alloc_blocks 27 free_blocks 3 total_blocks 30
  At 0x3ffb0000 len 10032 free 4364 allocated 5504 min_free 0
    largest_free_block 3264 alloc_blocks 26 free_blocks 7 total_blocks 33
  At 0x3ffae6e0 len 6192 free 2704 allocated 3372 min_free 0
    largest_free_block 2676 alloc_blocks 18 free_blocks 3 total_blocks 21
  At 0x3ffbbb28 len 8192 free 48 allocated 8000 min_free 0
    largest_free_block 28 alloc_blocks 26 free_blocks 2 total_blocks 28
  At 0x3ffb2730 len 22736 free 1476 allocated 21000 min_free 4
    largest_free_block 648 alloc_blocks 43 free_blocks 14 total_blocks 57
  At 0x3ffaff10 len 240 free 0 allocated 196 min_free 0
    largest_free_block 0 alloc_blocks 3 free_blocks 0 total_blocks 3
  At 0x3ffd8190 len 32368 free 592 allocated 31356 min_free 0
    largest_free_block 384 alloc_blocks 93 free_blocks 4 total_blocks 97
  At 0x3ffe0440 len 15296 free 344 allocated 14884 min_free 0
    largest_free_block 344 alloc_blocks 8 free_blocks 1 total_blocks 9
  At 0x3ffe4350 len 113840 free 66728 allocated 46948 min_free 56720
    largest_free_block 58252 alloc_blocks 21 free_blocks 12 total_blocks 33
  Totals:
    free 76412 allocated 155404 min_free 56724 largest_free_block 58252
10 7449 [MQTTEcho] [INFO ][MQTT][7444] MQTT SUBSCRIBE operation queued.
11 7449 [MQTTEcho] [INFO ][MQTT][7449] Waiting for operation SUBSCRIBE to complete.
12 7622 [MQTTEcho] [INFO ][MQTT][7622] MQTT operation SUBSCRIBE complete with result SUCCESS.
```
This is the same log from release-1.5.
```
            connectStatus = IotMqtt_Connect( &networkInfo,
                                         &connectInfo,
                                         TIMEOUT_MS,
                                         pMqttConnection );
```

 I think i am missing something obvious here. I also found issue in my thinking/tests. In ble-beta i didnt call mqtt code at all. Now it looks to me there is not very big difference in heap usage, about 10kB, which is consistent because its the stack for 2 `iot_thread` tasks.

I will close this issue now. Thanks for help and patients.

closed 2019-07-03 by chegewara