aws/amazon-freertos #853: OTA Agent does not handle invalid job data and the agent stuck

bbergquist0930, 2019-06-27:
**Describe the bug**
The OTA Agent does not handle invalid OTA job data correctly and the agent gets stuck.  This can occur when creating an OTA job using the AWS CLI.  In this case, a mistake in a script utilizing the AWS CLI created a OTA job that had an invalid job signature algorithm.

There is a condition missing in 

https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c

for the "if" at line 1464.  When the agent detects bad OTA job data, the context returned will be NULL, however inside this "if" only a test for the self test state is done.  A condition needs to be added that if not in the self test state and the context is NULL then abort the job with an error indicating that the job failed to parse.

**System information**
This issue was discovered on an older version of the code, version 1.0.0 however the relevant code is unchanged in latest version of this file.

**Expected behavior**
When the OTA Agent detects invalid job data when parsing the JSON, the OTA Agent should abort the OTA job with a parse error.

**To reproduce**
Create an OTA job with invalid JSON data.  We used the Amazon AWS CLI, to create a OTA job with the signature algorithm being "RSAX" which is invalid as a test case.  When the OTA job is received by the device, the parser correctly detects the invalid signature and the call to "prvProcessOTAJobMsg" return NULL.  The job is not aborted however and the OTA processes is stuck.

**Additional context**
A pull request will be created for this issue shortly.

Thank you!

bbergquist0930, 2019-06-27:
There is pull request for this issue

https://github.com/aws/amazon-freertos/pull/854

pvyawaha, 2019-06-27:
Hi Brett,

Thank you for pointing this out by providing details and submitting a pull request. I reviewed the PR and  it is a valid check to handle bad job document parsing. I left a small comment on the PR regarding the condition for aborting on job parsing failure. 

Instead of calling prvSetImageStateWithReason after checking for eOTA_ImageState_Unknown we can call it for any other state than eOTA_ImageState_Testing. For example previous abort set the image state to eOTA_ImageState_Aborted and after that we get a job document that fails parsing.

pvyawaha, 2019-06-28:
Hi Brett,

The https://github.com/aws/amazon-freertos/pull/854 is merged in mainline after running required checks. We are in process of updating the OTA tests for job parsing errors. 

Thank you again for providing details of this issue and creating pull request with the fix.

Closing this ticket as pull request is merged.

closed 2019-06-29 by pvyawaha