aws/amazon-freertos #870: About the Ninja build

sheldon123z, 2019-06-30:
Hi,
I am super confused about the documents. It has been changed many times in the past week.  I am using esp32 DevKit to do a small project with freeRTOS, and the last release was very smooth for me. However, this release migrated to Cmake build system, and then many problems have occurred. I got many suggestions from @tgsong in another issue, but my problem hasn't been solved. I noticed the newest docs deleted the part about using ninja to build the application, and actually, the ninja build was never effected for me. I used `cmake --build  build --target flash` to flash the demo to the board, but after everything was done, nothing happens to my IoT console. The program stopped at the last step. 
Here is the complete building-flashing process:

------------------------------------------record---------------------------------------------------------
Microsoft Windows [Version 10.0.17134.829]
(c) 2018 Microsoft Corporation. All rights reserved.

c:\Users\shzheng\Desktop>cd FreeRTOSWindows\amazon-freertos-master

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master>cmake --build build --target flash
[0/2] Re-checking globbed directories...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\Users\shzheng\Desktop\FreeRTOSW...indows/amazon-freertos-master/build" -P run_esptool.cmake"
Note: esptool.py will search for a serial port. To specify a port, set the ESPPORT environment variable.
Note: Using default baud rate 460800. To modify, set ESPBAUD environment variable.

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master>cmake --build build4 --target flash
[0/2] Re-checking globbed directories...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\Users\shzheng\Desktop\FreeRTOSW...ndows/amazon-freertos-master/build4" -P run_esptool.cmake"
Note: esptool.py will search for a serial port. To specify a port, set the ESPPORT environment variable.
Note: Using default baud rate 460800. To modify, set ESPBAUD environment variable.

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master>aws --version
aws-cli/1.16.184 Python/3.6.0 Windows/10 botocore/1.12.174

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master>cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -GNinja -S . -B build4
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Espressif/ESP-IDF Tools/toolchain/bin/xtensa-esp32-elf-gcc.exe
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- Checking Python dependencies...
Python requirements from C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/requirements.txt are satisfied.
-- Building for target esp32
-- Component names: soc log heap freertos vfs newlib driver tcpip_adapter xtensa-debug-module app_trace partition_table app_update spi_flash mbedtls micro-ecc bootloader_support ethernet nvs_flash pthread smartconfig_ack wpa_supplicant esp32 cxx amazon-freertos-common bootloader bt console esp_adc_cal esptool_py expat wear_levelling sdmmc fatfs nghttp openssl spiffs ulp
-- Component paths: C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/soc;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/log;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/heap;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/freertos;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/vfs;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/newlib;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/driver;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/tcpip_adapter;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/xtensa-debug-module;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/app_trace;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/partition_table;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/app_update;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/spi_flash;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/mbedtls;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/micro-ecc;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/bootloader_support;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/ethernet;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/nvs_flash;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/pthread;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/smartconfig_ack;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/wpa_supplicant;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esp32;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/cxx;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/amazon-freertos-common;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/bootloader;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/bt;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/console;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esp_adc_cal;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esptool_py;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/expat;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/wear_levelling;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/sdmmc;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/fatfs;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/nghttp;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/openssl;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/spiffs;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/ulp
Creating config file C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/build4/sdkconfig...
-- IDF_VER: v3.1.3-25-g36733db8d
fatal: not a git repository (or any of the parent directories): .git
-- Project version: HEAD-HASH-NOTFOUND
fatal: not a git repository (or any of the parent directories): .git
-- Adding linker script C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/build4/vendors/espressif/boards/esp32/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esp32/ld/esp32.common.ld
-- Adding linker script C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esp32/ld/esp32.rom.ld
[... cut ...]
c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master> 
------------------------------------end----------------------------------------------------------------

As you can see, it should show some steps of connecting to the IoT console, but it didn't happen.
I also tried it on MacOS, but for some unknown reasons, the board won't connect to my computer(the serial connection failed, but I installed the driver). 
Please help me and point out which step is wrong, I really need help with it.

Many thanks in advance!
Sheldon

alexa-noxon, 2019-07-01:
Hi Sheldon,

Apologies for the documentation changes-- we are working to update the docs and it looks like we have a little more work yet to do still.

Can you tell if the aws_demos.bin file is actually being generated?  This step might just have only generated the elf file (even though the print statement indicates otherwise).   If that is the case and there is no .bin, you will need to run the esptool to covert from elf to bin.

`python \path-to-repo\amazon-freertos\vendors\espressif\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 elf2image aws_demos
esptool.py v2.6`

and do the same for the bootloader
`cd bootloader`
`python \path-to-repo\amazon-freertos\vendors\espressif\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 elf2image bootloader.elf
esptool.py v2.6`

Then you would be able to pick back up with the instructions for using the IDF python tool idf.py to erase the flash, flash with a new image, and get the monitor output
`./vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p COMxy -B build`

Let me know if that resolves the issue, and if not, whether you can actually see the .bin file in the file system or not.   In the meantime we will work to update the documentation (again!) to capture all of these steps.

Thanks for reporting this,
Alexa

sheldon123z, 2019-07-02:
Hi Alexa,
Thank you for your reply. I have checked the files, and there are the .bin files in the folder. I just want to quit windows now and try MacOS. But I don't know why my Macbook pro cannot find the esp32 board. I have installed the driver, and the device info showed the driver was there. My mac has four thunderbolt type C port, I am wondering if that is the reason.

By the way, I can't run the idf.py on my windows, the system prompts the directory is invalid.

sarenameas, 2019-07-02:
Great that you can flash the board now. 

Try erasing the flash then starting over:
`python ./vendors/espressif/esp-idf/tools/idf.py erase_flash -B build`

Please make sure that your thing's certificate and private key are pasted into: 
https://github.com/aws/amazon-freertos/blob/master/demos/include/aws_clientcredential_keys.h

Please use try the ATS endpoint: `a1zwfs4sf2i9un-ats.iot.us-east-1.amazonaws.com`

Also will need an asterisk * at the end of your resource strings:
```
{
"Version": "2012-10-17",
"Statement": [
{
"Effect": "Allow",
"Action": "iot:Connect",
"Resource": "arn:aws:iot:us-east-1:330321474979:*"
},
{
"Effect": "Allow",
"Action": "iot:Publish",
"Resource": "arn:aws:iot:us-east-1:330321474979:*"
},
{
"Effect": "Allow",
"Action": "iot:Subscribe",
"Resource": "arn:aws:iot:us-east-1:330321474979:*"
},
{
"Effect": "Allow",
"Action": "iot:Receive",
"Resource": "arn:aws:iot:us-east-1:330321474979:*"
}
]
}
```

sheldon123z, 2019-07-03:
Hi Sarenameas，

Thank you very much for your reply. I have found the problem that my board cannot communicate to aws iot console. The reason is that my company blocked the AWS iot port. After I used my phone's wifi hotspot, everything went well.  

Issue summary: I have tried both Windows and MacOS operating system, and found out that the freeRTOS is so hard to setup on Windows, I struggled for a week but haven't succeeded yet. The ninja build system is pretty hard to use because the lack of instructions. Also, under MacOS, the esp32 driver needs to be removed from the blocking list of the security system, or the computer won't detect the board.

Thank you very much and this issue is closed.

closed 2019-07-03 by sheldon123z

sheldon123z, 2019-07-03:
Also want to mention that the error of the document about mqtt demo. The topic name is `iotdemo/topic/1` to` iotdemo/topic/4` rather than the `reertos/demos/echo` mentioned in the document