aws/amazon-freertos #875: OTA Agent hangs indefinitely if initial message to Amazon is dropped

salemlily, 2019-07-01:
**Describe the bug**
The OTA Agent can potentially fall into a hanging state if the initial message sent to AWS fails to send.  After initializing MQTT, `prvOTAPublishCallback` is called for each publish message the OTA Agent receives, including the initial callback to establish AWS connection.  If we are unsuccessful in queueing the initial packet (seen in [aws_iot_ota_agent.c:1330](https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c#L1330)), the response is to uptick our dropped packets and wait for the next event trigger.  If we do not successfully complete that initial connection to AWS, no new event comes and the system hangs indefinitely.

**System information**
This issue was discovered on an older version of the code, version 1.0.0 however the relevant code is unchanged in latest version of this file.

**Expected behavior**
If the initial message is unsuccessful, there should be a timeout that, if reached, either re-tries queueing the message or aborts the OTA job.

**To reproduce**
This bug is difficult to reproduce, and involves intentionally dropping the first message post-MQTT initialization.  The best way to reliably reproduce the issue is to add internal testing that blocks the first message from successfully queueing. 

**Additional context**
A pull request will be created for this issue shortly.  Attached below is a flow chart of the current OTA agent.
[image]

Thank you!

pvyawaha, 2019-07-16:
Hi salemlily,

Thank you for reporting this issue and creating a Pull Request.

> The OTA Agent can potentially fall into a hanging state if the initial message sent to AWS fails to send. After initializing MQTT, prvOTAPublishCallback is called for each publish message the OTA [...]

[aws_iot_ota_agent.c:1330](https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c#L1330) is where the job status is updated and not the initial message to job service which is called from `OTA_CheckForUpdate`.

The OTA agent after initialization publishes get next job message with QoS 1 and the `OTA_MAX_PUBLISH_RETRIES` is set to 3 with delay `OTA_RETRY_DELAY_MS` set to 1 second. These configurations can be changed if more retries and delay is required.

The changes that you have in the pull request extends this retries further using a timer and new configuration parameter `MAX_RETRIES` so effectively `MAX_RETRIES * OTA_MAX_PUBLISH_RETRIES`  retries. 

When publishing this initial get next job message we can have two cases - 

1. When the MQTT publish itself fails after `OTA_MAX_PUBLISH_RETRIES` and the changes  in your PR will retry it `MAX_RETRIES * OTA_MAX_PUBLISH_RETRIES` times with idle time in between till timer expires. So  similar to increasing OTA_MAX_PUBLISH_RETRIES. 

2. Second case is when the publish is successful for getting next job document and as you mentioned scenario where Job service does not send a job document. Have you seen this issue in your testing and can you share logs if you have them available?

I also have question with changes in PR regarding resetting of the device as you have added reset logic after `MAX_RETRIES ` and I think it will be abrupt for application to get a reset without callback.

bbergquist0930, 2019-07-17:
There is an issue without a timer for the first message which is the "Get Next Job Message".  If you look, the code in "OTA_CheckForUpdate" sends this message
`        eResult = prvPublishMessage(
            xOTA_Agent.pvPubSubClient,
            pcJobTopic,
            usTopicLen,
            pcMsg,
            ulMsgLen,
            IOT_MQTT_QOS_1 );
`

the QoS is 1 which indicates that it will retry the message until it is successfully received by the AWS cloud.  Note that there is no application level timeout here, just the MQTT timeout.  

Assume that the message is received by AWS cloud and now assume there is some processing error in the AWS cloud that prevents the AWS cloud from publishing the "next job response".   Although this is not expected, it can happen as it is software, the system could crash, whatever.  

The important point is that the MQTT message from the OTA process to the AWS cloud was delivered and acknowledged back to the OTA process as being delivered however the AWS cloud does not publish the "next job response".

At this point the OTA process is stuck.  Since there is no application level timeout waiting for the "next job response" message and this scenario the AWS cloud will not send that message, the OTA process will remain waiting for an event 

`            for( ; ; )
            {
                uxBits = xEventGroupWaitBits(
                    xOTA_Agent.xOTA_EventFlags, /* The event group being tested. */
                    OTA_EVT_MASK_ALL_EVENTS,    /* The bits within the event group to wait for. */
                    pdTRUE,                     /* Bits should be cleared before returning. */
                    pdFALSE,                    /* Any bit set will do. */
                    ( TickType_t ) ~( 0U ) );   /* Wait forever. */

 `

in "prvOTAUpdateTask" which is never going to come.   There is no timer started at this point, so there will be no OTA_EVT_MASK_REQ_TIMEOUT event.   There is no "next job" response message going to be received (in this scenario) so there will be no OTA_EVT_MASK_MSG_READY event.  

The code will wait forever in this call.  

This is the only request message in the OTA process that does not have an application level timer.  Although the scenario is not likely it can happen and this hole should be fixed.

The reset logic should be reexamined.  This could be updated to use the same "momentum" logic and fail the OTA when the maximum "momentum" is reached much like is done with the "get stream" message processing.

pvyawaha, 2019-07-23:
Hi Brett,

The OTA agent first subscribes to the two job notification topics `$aws/things/thingName/jobs/get/accepted` and `$aws/things/thingName/jobs/notify-next` and publishes on the topic - `$aws/things/%s/jobs/$next/get` for getting pending jobs. This publish has the retries that we discussed and I agree it can be extended by adding it to the momentum logic. This is the case when the publish itself fails.

Now other case is if the publish is successful but due to the scenario that you mentioned that service crashes and does not send response without device publishing on this topic again. Here if we use a timeout on device and if no pending jobs are available device will not get a response which is valid and does not necessarily indicate issues with service. If we add this to the momentum logic device will keep polling by publishing to get next job topic and within timeout if no jobs are pending shutdown the OTA agent. The application can restart the OTA agent and go thought this process again but this adds polling and increases number of MQTT messages defeating the purpose of notifications. 

I think we can address this scenario where service crashes by registering a MQTT disconnect callback as we will get a disconnect and in the callback reconnecting and resubscribing to the job notification topics and publishing to the get next topic. Let me know you thought on this.

pvyawaha, 2019-12-18:
Hello, 

Please let us know if you still have question(s) on this topic.

pvyawaha, 2019-12-23:
Hi,

I am closing this issue now, please reopen if you still have questions on the same issue. Please create a new issue for any other topic.

qiutongs, 2020-02-08:
Thanks for reporting the bug. We are still actively working on this and will merge the fix to master in coming releases.

closed 2020-02-08 by qiutongs