aws/amazon-freertos #877: iot_json_utils parser is not skipping opening and closing double quotes

rrivadeneira-rsc, 2019-07-01:
**Describe the bug**
Recently made a switch to use AFR's JSON parser to add simplicity. The parser finds the desired values, but I believe it is not skipping the opening and closing double quotes as it advertises. From what it looks like, it _should_ skip single and double quote characters. Or am I confused as to how this parser works?

I would expect that a string value found like this: `"www.example-website.com"` would be taken as this: `www.example-website.com` without the opening and closing characters `\"`. However, I can see it stored in memory as: `\"www.example-website.com\"`

**System information**
- **Which hardware board or part numbers?** - ESP32
- **IDE used** - Eclipse
- **Operating System** - MacOS
- **Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)** - 1.5 release
- **Project** - Custom Application
- **If your project is a Custom Application, please add the relevant code snippet in the section** -  I have no specific code for this, but a simple parse and retrieval of a JSON string value to re-use later on in the application. i.e. a JSON object sent as:
```
{
   "object": 
   {
      "url":"www.example-website.com"
   }
}
```

**Expected behavior**
I would expect that a string value found like this: `"www.example-website.com"` would be taken as this: `www.example-website.com` without the `\"` opening and closing characters.

**Screenshots or console output**
N/A

**To reproduce**
Run the **IoTJsonUtils_FindJsonValue** function with a valid document to parse, and the string values will be stored with opening and closing quotes.

**Code to reproduce the bug**
See above.

**Additional context**
N/A

Thank you!

gordonwang0, 2019-07-02:
Hi @rrivadeneira-rsc,

The inclusion of the quotes as part of the JSON value is intended.  This allows applications to distinguish JSON strings from other types.

For example, consider
`{"key": 0}` vs `{"key": "0"}`
Both of those are valid JSON, but the value type is different.  The quotes distinguish the second value as a string and not an integer.

rrivadeneira-rsc, 2019-07-02:
Upon reading through the function again and your explanation, that totally makes sense. That'd mean we need to change string values like these the way we'd like to have them stored afterwards. Thanks for the clarification @gordonwang0!

gordonwang0, 2019-07-02:
That's correct, your application should perform any needed conversion of string values.

rrivadeneira-rsc, 2019-07-08:
That makes sense. Thank you for clarifying!

closed 2019-07-08 by rrivadeneira-rsc