aws/amazon-freertos #897: OTA gets stuck if the "prvPublishGetStreamMessage" receives an error from "prvPublishMessage"

bbergquist0930, 2019-07-03:
**Describe the bug**
If the "prvPublishMessage" called within "prvPublishGetStreamMessage" returns an error, the OTA process becomes stuck.  

The “prvPublishGetStreamMessage” is called only on a timer event “prvOTAUpdateTask” which is set by “prvRequestTimer_Callback” which is the itself setup by “prvStartRequestTimer” which is called by both “prvProcessOTAJobMessage” and “prvInjectDataBlock”.

So if there is no timer setup, then “prvPublishGetStreamMessage” will never be called. When itself receives an error from the “prvPublishMessage”, then no timer is being restarted and it will never be called again.

The comment in the code indicates that the author wanted to just let the “momentum” mechanism to handle this issue and not return an error directly in case the publish error was intermittent, but the “momentum” mechanism is only incremented in this “prvPublishGetStreamMessage” function, so without it being called again this will never be updated.

The solution is to just start the timer again in this case. If the problem is intermittent, then the “prvPublishMessage” will not fail again. If it is persistent, then the “prvPublishGetStreamMessage” will be called exactly as if the there was a timeout on reception of the data blocks and after 32 retries, the OTA will be aborted.

**System information**
This issue was found on an earlier version of the software, however, looking at the latest code as of today 6/3/2019, the issue still exists.

**Expected behavior**
The OTA process should not become stuck and if the "prvPublishMessage" were to continue to fail and return an error, the OTA job would be aborted.

**Screenshots or console output**

**To reproduce**
So how cause the "prvPublishMessage" call to return an error.  What I did is to simulate this by having after 10 calls, to not call the "prvPublishMessage", but rather set "eResult" to something other than "eMQTTAgentSuccess".  This will show the symptoms and the OTA will become stuck.
[image]

Thank you!

sarenameas, 2019-07-03:
Hello,
Thanks for bringing this up. This intermittent issue we have seen and are actively working on internally. We will get back to you in a week or so after we recreate it internally and try your fix with a few days of soak as well.

pvyawaha, 2019-07-12:
Hi Brett, 

Thank you for reporting this. I have reviewed the PR and requested a change in the comment for calling prvStartRequestTimer.

pvyawaha, 2019-07-16:
The PR fixing this issue is merged in mainline so closing this issue.

Thank you.

closed 2019-07-16 by pvyawaha