aws/amazon-freertos #904: Old Version of FreeRtos code not working.

AagamMehta, 2019-07-05:
Hello,
     I was working on AWS IoT with ESP32 and RasPi. We were running FreeRtos on ESP32 and Greengrass on RasPi 3B+. The code was last tested on 7th June and was working absolutely fine but now when I am again starting to work on the project, I can see a major update on the FreeRtos page of GitHub. So, now the earlier version of FreeRtos code won't work? I am just using pub-sub service from FreeRtos to Cloud via Greengrass core. Need urgent help for project completion. Thanks for the help in advance!

Note - The code is getting compiled and even I can make new things via the setup file in Tools folder but not able to publish/subscribe any topic on the cloud.

sarenameas, 2019-07-06:
Hello AagemMehta,

Correct me if I misunderstand, so you are concerned that older versions of Amazon FreeRTOS are not working anymore since the release of 201906.00_Major? This release did not change the behavior of the cloud side, except in the Amazon FreeRTOS Device Software Download Configuration, which does not effect MQTT, so we expect the same behavior in older versions. 

What version of Amazon FreeRTOS are you using? Are you still able to discover your GGC from the ESP32? Could you run the Greengrass Discovery demo and post some logs?

AagamMehta, 2019-07-16:
Sorry, that was an issue from our side. Rechecking the complete set-up we have been able to get the error. Thanks for the support.

closed 2019-07-16 by AagamMehta