aws/amazon-freertos #905: Build fails for Cypress kit 43907 and 54907

K-Aadhi, 2019-07-05:
Build "aws_demo" fails for 43907 and 54907

**System information**
Hash tag 5a5fc6666c43ca4a571923fb1993e1d314b2434e

Fix for the same:

diff --git a/vendors/cypress/WICED_SDK/apps/demo/aws_demo/aws_demo.mk b/vendors/cypress/WICED_SDK/apps/demo/aws_demo/aws_demo.mk
index ab482457..419c3842 100644
--- a/vendors/cypress/WICED_SDK/apps/demo/aws_demo/aws_demo.mk
+++ b/vendors/cypress/WICED_SDK/apps/demo/aws_demo/aws_demo.mk
@@ -104,6 +104,7 @@ $(NAME)_SOURCES    := $(AMAZON_FREERTOS_PATH)vendors/cypress/boards/$(PLATFORM)/
                       $(AFR_ABSTRACTIONS_PATH)platform/freertos/iot_network_freertos.c \
                       $(AFR_ABSTRACTIONS_PATH)platform/freertos/iot_threads_freertos.c \
                       $(AFR_C_SDK_STANDARD_PATH)common/iot_init.c \
+                      $(AFR_C_SDK_STANDARD_PATH)common/iot_device_metrics.c \
                       $(AFR_C_SDK_STANDARD_PATH)serializer/src/iot_json_utils.c \
                       $(AFR_C_SDK_AWS_PATH)defender/src/aws_iot_defender_api.c \
                       $(AFR_C_SDK_AWS_PATH)defender/src/aws_iot_defender_collector.c \

initially this sounded like a third party code but this compilation error has been introduced because a new file has been added (iot_device_metrics.c) and not added to the compilation.

Thank you!
Aadhi Vairava Sundaram

sarenameas, 2019-07-06:
Thank you. Please see the comments on the PR.

sarenameas, 2019-07-07:
This fix has been merged. Go ahead and give the latest on master a try. Thanks.

closed 2019-07-07 by sarenameas