aws/amazon-freertos #910: Porting recent TLS Fix (MQTT Connections, etc.)

rrivadeneira-rsc, 2019-07-08:
**Describe the bug**
Slow connect time on any TLS related processes (MQTT connect calls, etc.). A fix has been submitted last week and merged today, but I wanted to open up an issue to communicate our testing with the team. The connection time seems to be slow still - what is the expected time for an MQTT connect call to run from [THIS](https://github.com/aws/amazon-freertos/pull/873/files#diff-6502580f090388c0771353b26e505e27) fix? How many seconds are you seeing on your end?

**System information**
- **Which hardware board or part numbers?** - ESP32
- **IDE used** - Eclipse
- **Operating System** - MacOS
- **Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)** - modified version from commit of AFR 1.5 (somewhere around commit 407790218 and 5a201aca6 from back in early May)
- **Project** - Custom Application
- **If your project is a Custom Application, please add the relevant code snippet in the section** - we run an iteration of the MQTT demo, so anything that runs a connection to the broker

**Expected behavior**
I'd expect this connection to run much smoother than before with the fix, but maybe I'm not applying the changes correctly to our code-base (see _**additional context**_ for more information)

**Screenshots or console output**
_**BEFORE** the TLS fix: note the ~15 second delay_
```
52 3277 [INFO ][DEMO][32770] MQTT demo client identifier is <IoT-ThingName> (length 12).
53 4722 [INFO ][MQTT][47220] Establishing new MQTT connection.
54 4722 [INFO ][MQTT][47220] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
55 4722 [INFO ][MQTT][47220] (MQTT connection 0x3ffb6770, CONNECT operation 0x3ffb6ea8) Waiting for operation completion.
56 4731 [INFO ][MQTT][47310] (MQTT connection 0x3ffb6770, CONNECT operation 0x3ffb6ea8) Wait complete with result SUCCESS.
57 4731 [INFO ][MQTT][47310] New MQTT connection 0x3ffc6cd0 established.
58 4731 [INFO ][DEMO][47310] MQTT Connection Returned: 0
59 4731 Connected to broker.
```

_**AFTER** the TLS fix: note the ~7 second delay_
```
52 2488 [INFO ][DEMO][24880] MQTT demo client identifier is <IoT-ThingName>  (length 12).
53 3254 [INFO ][MQTT][32540] Establishing new MQTT connection.
54 3254 [INFO ][MQTT][32540] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
55 3254 [INFO ][MQTT][32540] (MQTT connection 0x3ffb7a8c, CONNECT operation 0x3ffb78a0) Waiting for operation completion.
56 3268 [INFO ][MQTT][32680] (MQTT connection 0x3ffb7a8c, CONNECT operation 0x3ffb78a0) Wait complete with result SUCCESS.
57 3268 [INFO ][MQTT][32680] New MQTT connection 0x3ffc5e78 established.
58 3268 [INFO ][DEMO][32680] MQTT Connection Returned: 0
59 3268 Connected to broker.
```

**To reproduce**
**Steps to reproduce the behavior:**
Try any demo that requires an MQTT connection with the merged PR (probably from your master branch)

**Code to reproduce the bug**
See above

**Additional context**
I have not yet re-based on your master branch, as we were afraid of any breaking changes (are there any breaking changes?). So I took the snippet from your [recent PR](https://github.com/aws/amazon-freertos/pull/873/files#diff-6502580f090388c0771353b26e505e27) and implemented those on what we currently have (based on the commit posted above). Are the _CMakeLists.txt_ file changes under the different directories the only required edits to apply this fix? Is there any other change that I should be making in the AFR code itself? Let me know if the fix comes with additional changes other than the one I mentioned, so I can implement those as well - or if it is required for the fix to re-base our code-base to the latest commit on master of the SDK altogether.

Thank you!

rrivadeneira-rsc, 2019-07-09:
I took a few more samples this morning after soft reboots to the unit and these are some of the samples - note the 15+ second delays on these.
```
33 708 [INFO ][DEMO][7080] MQTT demo client identifier is <IoT-ThingName> (length 12).
34 2193 [INFO ][MQTT][21930] Establishing new MQTT connection.
35 2193 [INFO ][MQTT][21930] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
36 2193 [INFO ][MQTT][21930] (MQTT connection 0x3ffb75a8, CONNECT operation 0x3ffb79b8) Waiting for operation completion.
37 2284 [INFO ][MQTT][22840] (MQTT connection 0x3ffb75a8, CONNECT operation 0x3ffb79b8) Wait complete with result SUCCESS.
38 2284 [INFO ][MQTT][22840] New MQTT connection 0x3ffc5e78 established.
39 2284 [INFO ][DEMO][22840] MQTT Connection Returned: 0
40 2284 Connected to broker.
```
```
31 326 [INFO ][DEMO][3260] MQTT demo client identifier is <IoT-ThingName> (length 12).
32 1808 [INFO ][MQTT][18080] Establishing new MQTT connection.
33 1808 [INFO ][MQTT][18080] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
34 1808 [INFO ][MQTT][18080] (MQTT connection 0x3ffb7820, CONNECT operation 0x3ffb7160) Waiting for operation completion.
35 2008 [INFO ][MQTT][20080] (MQTT connection 0x3ffb7820, CONNECT operation 0x3ffb7160) Wait complete with result SUCCESS.
36 2008 [INFO ][MQTT][20080] New MQTT connection 0x3ffc5e78 established.
37 2008 [INFO ][DEMO][20080] MQTT Connection Returned: 0
38 2008 Connected to broker.
```

huguesBouvier, 2019-07-09:
Hey,

I ran a test on our latest master branch (ba6e43bd7141d1c80f6e98d18bddde989b032b5f).
Here are the results I obtained:
cmake:
7 411 [iot_thread] [INFO ][DEMO][4090] MQTT demo client identifier is IOT_BLE_HUGUES (length 14).
8 857 [iot_thread] [INFO ][MQTT][8570] Establishing new MQTT connection.

makefile:
7 396 [iot_thread] [INFO ][DEMO][3960] MQTT demo client identifier is IOT_BLE_HUGUES (length 14).
8 809 [iot_thread] [INFO ][MQTT][8090] Establishing new MQTT connection.

Aproximatly ~4s for both. That is the time you should expect on our master branch. Again that time will depend on you application and thread priorities.

Hugues

rrivadeneira-rsc, 2019-07-09:
Thank you for the answer Hugues.

I have ported it the way you suggested, copying the appropriate .txt files - thank you for the confirmation, that's helpful. I'll double check, run more tests with our application and also try to test your master branch to confirm we see the same timing and get back to you.

Again, thank you for the support.

huguesBouvier, 2019-07-10:
I create that branch https://github.com/aws/amazon-freertos/tree/fix/bootIn4s
Based on an older commit you might be using.
Please check this commit:
https://github.com/aws/amazon-freertos/commit/b6348a915714250d57240e902d815ac5c94f8381
It should connect in 4s

rrivadeneira-rsc, 2019-07-10:
Thanks Hugues, I'll try this as soon as possible. I am unable to run a CMake build on the master branch at the moment, so I can't test that yet. But I see that you are already working on a fix.

Let me know when that is available and I can test it. In the meantime, I'll try: https://github.com/aws/amazon-freertos/tree/fix/bootIn4s

huguesBouvier, 2019-07-10:
You are right, sorry about that. It should work now, the fix has been pushed.
Note that that fix was for tests, demos was working.

rrivadeneira-rsc, 2019-07-10:
Ah, interesting. I'll run CMake again from master once I'm done with this one. Not sure what could have gone wrong then. I'll get back to you soon, thanks!

ravibhagavandas, 2019-07-22:
Hi @rrivadeneira-rsc,

How did the test go ? Are you still having issues ?

rrivadeneira-rsc, 2019-07-30:
Hi @ravibhagavandas 

Tests are showing better results, thank you! We get ~6 seconds consistently trying the demos only on both branches with stable WiFi network, and ~7 seconds with our application on top. Though I definitely see it varies according to the network - so an unstable connection could take a lot longer.

pvyawaha, 2019-07-30:
Hi @rrivadeneira-rsc,

Thank you for sharing results , I am closing this issue now and please reopen if you have more questions.

closed 2019-07-30 by pvyawaha