aws/amazon-freertos #940: MQTT v4 Library Can Fail `Malloc` Silently and Fails to Flush Received Packet

tombresson, 2019-07-15:
**Describe the bug**
Incoming MQTT messages can **silently fail** and then cause an `Unknown packet type <xx> received.` message if buffer for packet can not be allocated. 

The issue of the Unknown Packet is caused by the function `IotMqtt_ReceiveCallback` calling `_getIncomingPacket`, but then not handling the `IOT_MQTT_NO_MEMORY` case. Because `IotMqtt_ReceiveCallback` does not handle the No Memory case, the receive of the packet is never actually done. This means, when the `_networkReceiveTask` task calls `SOCKETS_Recv` again (as it will once the failed allocation and packet read returns), the next byte (which in my case happens to be a `0x00` byte) is received **and the system expects it to be a new MQTT packet.** What it actually is, is the remainder of the packet that could not be allocated, which causes the MQTT packet type deserializer (`_IotMqtt_GetPacketType`) to determine that it is bad data.

I've included a snippet of the log. The relevant annotated lines are:
* `[DEBUG][NET][47305] Successfully received 1 bytes.` (First byte from real message received - Legitimate Packet Type)
* `[DEBUG][NET][47310] Successfully received 1 bytes.` (Second byte - Length byte 0)
* `[DEBUG][NET][47316] Successfully received 1 bytes.` (Third byte - Length byte 1)
* **_Malloc silently fails here, which returns all the way up the call stack to `_networkReceiveTask`._**
* **_`_networkReceiveTask` calls `SOCKETS_Recv` to receive the next byte (which is waiting there from the unflushed packet)_**
* `[DEBUG][NET][47321] Successfully received 1 bytes.` (Fourth byte of original packet - Treated as a **new** message)
* `[ERROR][MQTT][47326] (MQTT connection 20033b0c) Unknown packet type 00 received.` (`_getIncomingPacket` attempts to call `_IotMqtt_GetPacketType` on the received byte and fails because it's an invalid packet type.)

**System information**
- Version of Amazon FreeRTOS: **Version 201906.00 Major**
- Project [Custom Application|MQTT|Shadow]

**Expected behavior**
The expected behavior would be as follows:
* Debug Feedback from the MQTT library (probably best to be in `_getIncomingPacket`) that the `malloc` failed.
* `IotMqtt_ReceiveCallback` or `_getIncomingPacket` to _consume/flush_ the packet from the socket, so that the next byte that is waiting on the socket connection is not considered to be a **new MQTT message**.

**Screenshots or console output**

```
[DEBUG][MQTT][15502] (MQTT connection 20033b0c) Sending PUBACK for received PUBLISH 1.
[DEBUG][MQTT][15510] MQTT PUBACK packet:
40 02 00 01
[DEBUG][MQTT][15516] (MQTT connection 20033b0c) PUBACK for received PUBLISH 1 sent.
[DEBUG][MQTT][15524] (MQTT connection 20033b0c) Reference count changed from 2 to 3.
[DEBUG][Shadow][15532] Verifying client tokens for Shadow UPDATE.
[DEBUG][Shadow][15539] Received Shadow response on topic $aws/things/Cobalt/shadow/update/accepted
[INFO ][Shadow][15548] Shadow UPDATE of Cobalt was ACCEPTED.
[DEBUG][Shadow][15554] Shadow UPDATE for Cobalt has persistent subscriptions. Reference count will not be decremented.
[DEBUG][Shadow][15565] Destroying Shadow operation UPDATE.
[DEBUG][MQTT][15555] (MQTT connection 20033b0c) Reference count changed from 3 to 2.
1563210034808 0002 DEBUG ShadowClient: Successfully performed Device Shadow update.
[DEBUG][NET][47305] Successfully received 1 bytes.
[DEBUG][NET][47310] Successfully received 1 bytes.
[DEBUG][NET][47316] Successfully received 1 bytes.
[DEBUG][NET][47321] Successfully received 1 bytes.
[ERROR][MQTT][47326] (MQTT connection 20033b0c) Unknown packet type 00 received.
[ERROR][MQTT][47333] (MQTT connection 20033b0c) Error processing incoming data. Closing connection.
[DEBUG][MQTT][47342] (MQTT connection 20033b0c) Keep-alive job canceled and cleaned up.
[INFO ][MQTT][47351] (MQTT connection 20033b0c) Network connection closed.
[INFO ][MQTT][47358] (MQTT connection 20033b0c) Disconnecting connection.
[INFO ][MQTT][47364] (MQTT connection 20033b0c) Network connection closed.
[DEBUG][MQTT][47371] (MQTT connection 20033b0c) Reference count changed from 1 to 0.
[DEBUG][MQTT][47379] (MQTT connection 20033b0c) Connection will be destroyed now.
[DEBUG][NET][47358] Network receive task terminating.
[INFO ][MQTT][47393] (MQTT connection 20033b0c) Network connection destroyed.
[DEBUG][MQTT][47400] (MQTT connection 20033b0c) Connection destroyed.
[INFO ][Shadow][47406] Shadow library cleanup done.
[INFO ][Shadow][47411] Shadow library successfully initialized.
1563210066649 0002 DEBUG ShadowClient: Client Disconnect successful.
[INFO ][MQTT][48958] Establishing new MQTT connection.
```

**To reproduce**
Steps to reproduce the behavior:

1. Make an MQTT connection to AWS
1. Publish message(s) to the shadow **with persistent connection**. The message that is not able to be allocated for our application comes in on the `$aws/things/<thingName>/shadow/update/accepted` topic, which is a persistent topic.
1. Modify the shadow on the AWS platform.
1. When a **large** update is posted on the  `$aws/things/<thingName>/shadow/update/accepted`, the device will receive it (even though in the case where the Cloud has modified the device shadow, the Device doesn't need to know if it was accepted or not)
1. Device fails to `malloc` enough space for the incoming data on the subscribed topic and fails silently. 
1. MQTT library does not _flush_ the packet after failing to allocate the required space and attempts to receive the rest of the packet as a **new** packet.

**I think the addition of a `DEBUG` message when the `malloc` fails would be an EXCELLENT addition.**

I've included snippets below of the relevant parts of the functions.

If `malloc` fails, the function is cleaned up and exited.
https://github.com/aws/amazon-freertos/blob/24c0b1c078d60b20f73b24fe64128d814a70fc3e/libraries/c_sdk/standard/mqtt/src/iot_mqtt_network.c#L198-L209

`IOT_MQTT_NO_MEMORY` not handled at this level.
https://github.com/aws/amazon-freertos/blob/24c0b1c078d60b20f73b24fe64128d814a70fc3e/libraries/c_sdk/standard/mqtt/src/iot_mqtt_network.c#L862-L910

tombresson, 2019-07-15:
@lt72 Thanks for taking a look at this. 😄

qiutongs, 2019-07-15:
@tombresson  Appreciate for reporting this to us. We agree this is a bug in our MQTT library and it should handle IOT_MQTT_NO_MEMORY case.

We are in discussion of the fix and will post an update once it is ready.

gordonwang0, 2019-07-17:
Hi @tombresson,

The merged PR #943 resolves this.  Please give it a try and comment/reopen if you have any questions.

closed 2019-07-17 by gordonwang0