aws/amazon-freertos #953: getThingShadow Error

sheldon123z, 2019-07-18:
Hi,

I am using 'AwsIotShadow_TimedGet()' to get the thing shadow from AWS IoT, however, I met the error when my program runs for a while. I am using the shadow demo as the template to do my application.
Here is the error log, I think the error happens on the network layer, but I don't know how to handle it.

```
87 6586 [iot_thread] [INFO ][DEMO][65860] finished the shadow operation
88 6886 [iot_thread] [INFO ][MQTT][68860] (MQTT connection 0x3ffd5278) MQTT PUBLISH operation queued.
89 6924 [iot_thread] [INFO ][Shadow][69240] Shadow GET of esp32 was ACCEPTED.
90 6924 [iot_thread] [INFO ][Shadow][69240] pDocument here,payloadLength is 1420
91 6924 [iot_thread] [INFO ][DEMO][69240] state is "OFF","brightness":60,"value":{"value":0},"property1":{"default property1":0},"colorTemperatureInKelvin":3000},"Switch":{"Switch value":"OFF92 6924 [iot_thread] [INFO ][DEMO][69240] Shadow document transmitted to uart port! The shadow sent is "OFF"��?

93 6924 [iot_thread] [INFO ][DEMO][69240] finished the shadow operation
94 7224 [iot_thread] [INFO ][MQTT][72240] (MQTT connection 0x3ffd5278) MQTT PUBLISH operation queued.
95 7262 [iot_thread] [INFO ][Shadow][72620] Shadow GET of esp32 was ACCEPTED.
96 7262 [iot_thread] [INFO ][Shadow][72620] pDocument here,payloadLength is 1420
97 7262 [iot_thread] [INFO ][DEMO][72620] state is "OFF","brightness":60,"value":{"value":0},"property1":{"default property1":0},"colorTemperatureInKelvin":3000},"Switch":{"Switch value":"OFF98 7262 [iot_thread] [INFO ][DEMO][72620] Shadow document transmitted to uart port! The shadow sent is "OFF"��?

99 7262 [iot_thread] [INFO ][DEMO][72620] finished the shadow operation
100 7562 [iot_thread] [INFO ][MQTT][75620] (MQTT connection 0x3ffd5278) MQTT PUBLISH operation queued.
101 7600 [iot_thread] [INFO ][Shadow][76000] Shadow GET of esp32 was ACCEPTED.
102 7600 [iot_thread] [INFO ][Shadow][76000] pDocument here,payloadLength is 1420
103 7600 [iot_thread] [INFO ][DEMO][76000] state is "OFF","brightness":60,"value":{"value":0},"property1":{"default property1":0},"colorTemperatureInKelvin":3000},"Switch":{"Switch value":"OF104 7600 [iot_thread] [INFO ][DEMO][76000] Shadow document transmitted to uart port! The shadow sent is "OFF"��?

105 7600 [iot_thread] [INFO ][DEMO][76000] finished the shadow operation
106 7900 [iot_thread] [INFO ][MQTT][79000] (MQTT connection 0x3ffd5278) MQTT PUBLISH operation queued.
107 7938 [iot_thread] [INFO ][Shadow][79380] Shadow GET of esp32 was ACCEPTED.
108 7938 [iot_thread] [INFO ][Shadow][79380] pDocument here,payloadLength is 1420
109 7938 [iot_thread] [INFO ][DEMO][79380] state is "OFF","brightness":60,"value":{"value":0},"property1":{"default property1":0},"colorTemperatureInKelvin":3000},"Switch":{"Switch value":"OF110 7938 [iot_thread] [INFO ][DEMO][79380] Shadow document transmitted to uart port! The shadow sent is "OFF"��?

111 7938 [iot_thread] [INFO ][DEMO][79380] finished the shadow operation
112 8238 [iot_thread] [INFO ][MQTT][82380] (MQTT connection 0x3ffd5278) MQTT PUBLISH operation queued.
I (85685) wifi: bcn_timout,ap_probe_send_start
E (85695) NetInterface: Failed to get buffer descriptor
E (85695) NetInterface: Failed to get buffer descriptor
I (88185) wifi: ap_probe_send over, resett wifi status to disassoc
I (88195) wifi: state: run -> init (100)
I (88195) wifi: pm stop, total sleep time: 69954908 us / 84971977 us

I (88195) wifi: n:10 0, o:10 2, ap:255 255, sta:10 2, prof:1
I (88205) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 200
113 9238 [iot_thread] [ERROR][DEMO][92380] get thingshadow failed,the result is 7
114 9238 [iot_thread] [WARN ][DEMO][92380] Failed to find "state" in Shadow delta document.
�@��?238 [iot_thread] [WARN ][DEMO][92380] attribute didn't find, the bad value is D
116 9238 [iot_thread] [INFO ][DEMO][92380] finished the shadow operation
117 9538 [iot_thread] [INFO ][MQTT][95380] (MQTT connection 0x3ffd5278) MQTT PUBLISH operation queued.
118 10538 [iot_thread] [ERROR][DEMO][105380] get thingshadow failed,the result is 7
119 10538 [iot_thread] [WARN ][DEMO][105380] Failed to find "state" in Shadow delta document.
�@��?0538 [iot_thread] [WARN ][DEMO][105380] attribute didn't find, the bad value is D
121 10538 [iot_thread] [INFO ][DEMO][105380] finished the shadow operation
122 10838 [iot_thread] [INFO ][MQTT][108380] (MQTT connection 0x3ffd5278) MQTT PUBLISH operation queued.
```
As you can see above, from the part 
```I (85685) wifi: bcn_timout,ap_probe_send_start
E (85695) NetInterface: Failed to get buffer descriptor
E (85695) NetInterface: Failed to get buffer descriptor```
the application acts abnormally. I printed out the failure code `7` which is a "AWS_IOT_SHADOW_TIMEOUT". I set the timeout argument in the 'AwsIotShadow_TimedGet()' as 10000, and set the flag as `AWS_IOT_SHADOW_FLAG_KEEP_SUBSCRIPTIONS`. I noticed that there is a ` MQTT PUBLISH operation queued`  info log appears every time while I never set that. I don't understand how to handle this situation, could you please give me some suggestions?

Thanks!
Sheldon

sheldon123z, 2019-07-18:
Also a very urgent question is that how should I release the memory which was allocated to `mallocDocument`, the member function in the structure `AwsIotShadowDocumentInfo_t`? I used malloc to do the allocation, but it seems the function doesn't release the memory automatically. 
Here is the code how I use the 'AwsIotShadow_TimedGet'. The program prompts me an error the memory has ran out.
     
```
    AwsIotShadowError_t result = AWS_IOT_SHADOW_STATUS_PENDING;
    AwsIotShadowDocumentInfo_t getInfo = AWS_IOT_SHADOW_DOCUMENT_INFO_INITIALIZER;
    
    getInfo.pThingName = pThingName;
    getInfo.thingNameLength = thingNameLength;
    getInfo.u.get.mallocDocument = malloc;

 for (int i = 0; i< 100; i++)
    {
        const char* receivedDocument = NULL;
        size_t receivedDocumentLength = 0;
        result = AwsIotShadow_TimedGet( mqttConnection,
                                        &getInfo,
                                        AWS_IOT_SHADOW_FLAG_KEEP_SUBSCRIPTIONS,
                                        10000,
                                        &receivedDocument,
                                        &receivedDocumentLength);
                                
        if(result != AWS_IOT_SHADOW_SUCCESS)
        {
            IotLogError("get thingshadow failed,the result is %d",result);
        }
}
```

ravibhagavandas, 2019-07-18:
sheldon123z@,
Yes,  you should release the memory allocated for shadow document, from the application itself. `AwsIotShadow_TimedGet()` calls `AwsIotShadow_Get()` API which is an asynchronous operation. The get operation is queued for processing, and hence you see the log `117 9538 [iot_thread] [INFO ][MQTT][95380] (MQTT connection 0x3ffd5278) MQTT PUBLISH operation queued.`. 
`AwsIotShadow_TimedGet()` waits for the get callback to be completed within the timeout provided as part of the input parameters. It then cleans all internal resources before returning to user with the job document. At this point it's users responsibility to free the document.

sheldon123z, 2019-07-18:
Hi,
Thank you very much for the reply. 
So now I am using the function `AwsIotShadow_Wait` after `AwsIotShadow_Get` I wonder what kind of `AwsIotShadowOperation_t` --- one of the parameters in the Wait function should be set?

sheldon123z, 2019-07-22:
Thanks! that helps a lot!

closed 2019-07-22 by sheldon123z