aws/amazon-freertos #954: Porting existing project ESP32 to AWS Freertos

elieDaan, 2019-07-18:
**Describe the bug**
I've created a project using ESP-IDF. I want to port my project on AWS Freertos for OTA updates. With the [832](https://github.com/aws/amazon-freertos/issues/832) I succeed to write a Cmake with my own partition-table. The problem is that the bin created is only 66Ko and there is no code running. 
I've got the warning :
xtensa-esp32-elf/bin/ld: warning: cannot find entry symbol call_start_cpu0; not setting start address   

Architecture of the project is :
root folder : 
Cmakelists.txt
-directory components
   ->tft
   ->spiffs
   ...
 -directory main : 
   ->all sources files and header 
   ->menu_test.c contains the app_main 
- directory amazon-freertos 

**System information**
- Which hardware board or part numbers? ESP32 M5Stack
- IDE used ESP IDF 3.13
- Operating System [MacOS]
- Version of Amazon FreeRTOS 201906.00_Major-202-g24c0b1c0
- Project [Custom Application|MQTT]
- If your project is a Custom Application, please add the relevant code snippet in the section 

** CMakeLists  **
`# -------------------------------------------------------------------------------------------------
# Cmake to build project BOB and amazon freertos
# -------------------------------------------------------------------------------------------------
cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_COMPILER /Users/XX/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc)
set(CMAKE_CXX_COMPILER /Users/XX/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++)

#set(EXTRA_CFLAGS --save-temps -lm -l)
project(bob_v6)

file(GLOB SOURCES "/Users/XX/Documents/XXX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/main/*.c" ABSOLUTE)
file(GLOB HEADERS "/Users/XXDocuments/XXX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/main/*.h" ABSOLUTE)
add_executable(bob_v6 ${SOURCES} ${HEADERS})

# Bring in the include directories 
set(INCLUDE_DIRS ${INCLUDE_DIRS} 
    "/components/spiffs/"
    "/components/tft/"
    "/components/spidriver/"
    )
include_directories(${INCLUDE_DIRS})

# Tell IDF build to link against this target
set(IDF_PROJECT_EXECUTABLE bob_v6)

# define the components
get_filename_component(ESPRESSIF_CODE_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code" ABSOLUTE)
get_filename_component(ESP_SPIFFS_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/spiffs" ABSOLUTE)
get_filename_component(ESP_SPIFFS_IMG_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/spiffs_image" ABSOLUTE)
get_filename_component(ESP_MKSPIFFS_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/mkspiffs" ABSOLUTE)
get_filename_component(ESP_SPIDRIVER_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/spidriver" ABSOLUTE)
get_filename_component(ESP_TFT_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/tft" ABSOLUTE)

set(EXTRA_COMPONENT_DIRS    ${EXTRA_COMPONENT_DIRS} 
                                ${ESP_TFT_COMPONENT}
                                ${ESP_SPIFFS_COMPONENT}
                                ${ESP_SPIFFS_IMG_COMPONENT}
                                ${ESP_MKSPIFFS_COMPONENT}
                                ${ESP_MKSPIFFS_COMPONENT}
                           
)

#get_filename_component(EXTRA_COMPONENT_DIRS
#    "components/" ABSOLUTE
#)

# Add some external components to the project
#set(IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})

# Bring in the FreeRTOS code
# AFR_BOARD tells which board we need to target.
set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL "")
set(EXTRA_CFLAGS --save-temps -lm)
add_subdirectory(amazon-freertos)

message("++++++++++++++++++++++++++++++++++")
message("++++          MY APP          ++++")
message("++++++++++++++++++++++++++++++++++")

message("IDF_EXTRA_COMPONENT_DIRS: ${IDF_EXTRA_COMPONENT_DIRS}")

message("INCLUDE_DIRS: ${INCLUDE_DIRS}")

message("Glob Source : ${SOURCES}")

message("Glob headers : ${HEADERS}")
target_link_libraries(bob_v6 PRIVATE AFR::mqtt)`

**Expected behavior**
A clear and concise description of what you expected to happen.
Code running of menu_test.c app_main
**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

**To reproduce**
Steps to reproduce the behavior:

When executing monitor this the message always written : 
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
entry 0x00000000
user code done
ets Jun  8 2016 00:22:57
If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

Thank you!

ravibhagavandas, 2019-07-18:
Hi,

Re-posting the CMake for reference.

```cmake
# -------------------------------------------------------------------------------------------------
# Cmake to build project BOB and amazon freertos
# -------------------------------------------------------------------------------------------------
cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_COMPILER /Users/XX/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc)
set(CMAKE_CXX_COMPILER /Users/XX/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++)

#set(EXTRA_CFLAGS --save-temps -lm -l)
project(bob_v6)

file(GLOB SOURCES "/Users/XX/Documents/XXX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/main/*.c" ABSOLUTE)
file(GLOB HEADERS "/Users/XXDocuments/XXX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/main/*.h" ABSOLUTE)
add_executable(bob_v6 ${SOURCES} ${HEADERS})

# Bring in the include directories 
set(INCLUDE_DIRS ${INCLUDE_DIRS} 
    "/components/spiffs/"
    "/components/tft/"
    "/components/spidriver/"
    )
include_directories(${INCLUDE_DIRS})

# Tell IDF build to link against this target
set(IDF_PROJECT_EXECUTABLE bob_v6)

# define the components
get_filename_component(ESPRESSIF_CODE_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code" ABSOLUTE)
get_filename_component(ESP_SPIFFS_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/spiffs" ABSOLUTE)
get_filename_component(ESP_SPIFFS_IMG_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/spiffs_image" ABSOLUTE)
get_filename_component(ESP_MKSPIFFS_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/mkspiffs" ABSOLUTE)
get_filename_component(ESP_SPIDRIVER_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/spidriver" ABSOLUTE)
get_filename_component(ESP_TFT_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/tft" ABSOLUTE)

set(EXTRA_COMPONENT_DIRS    ${EXTRA_COMPONENT_DIRS} 
                                ${ESP_TFT_COMPONENT}
                                ${ESP_SPIFFS_COMPONENT}
                                ${ESP_SPIFFS_IMG_COMPONENT}
                                ${ESP_MKSPIFFS_COMPONENT}
                                ${ESP_MKSPIFFS_COMPONENT}
                           
)

#get_filename_component(EXTRA_COMPONENT_DIRS
#    "components/" ABSOLUTE
#)

# Add some external components to the project
#set(IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})

# Bring in the FreeRTOS code
# AFR_BOARD tells which board we need to target.
set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL "")
set(EXTRA_CFLAGS --save-temps -lm)
add_subdirectory(amazon-freertos)

message("++++++++++++++++++++++++++++++++++")
message("++++          MY APP          ++++")
message("++++++++++++++++++++++++++++++++++")

message("IDF_EXTRA_COMPONENT_DIRS: ${IDF_EXTRA_COMPONENT_DIRS}")

message("INCLUDE_DIRS: ${INCLUDE_DIRS}")

message("Glob Source : ${SOURCES}")

message("Glob headers : ${HEADERS}")
target_link_libraries(bob_v6 PRIVATE AFR::mqtt)
```
The CMake file looks correct to me. From the error `xtensa-esp32-elf/bin/ld: warning: cannot find entry symbol call_start_cpu0; not setting start address` it seems like bootloader code is not getting compiled correctly. `call_start_cpu0` is defined within the ESP-IDF bootloader component. Are you using a custom bootloader ? Also can you post the custom partition table?

elieDaan, 2019-07-19:
Hello,

Thanks for your reply.

First of all, my bootloader is not custom : 

void call_start_cpu0()
{
    // 1. Hardware initialization
    if (bootloader_init() != ESP_OK) {
        return;

    }

    // 2. Select the number of boot partition
    bootloader_state_t bs = { 0 };
    int boot_index = select_partition_number(&bs);
    if (boot_index == INVALID_INDEX) {
        return;
    }

    // 3. Load the app image for booting
    bootloader_utility_load_boot_image(&bs, boot_index);
}`

My custom partition table is : 
`boot: Partition Table:
 ## Label            Usage          Type ST Offset   Length
0 nvs              WiFi data        01 02 00009000 00004000
1 otadata          OTA data         01 00 0000d000 00002000
2 phy_init         RF data          01 01 0000f000 00001000
 3 factory          factory app      00 00 00010000 000e0000
 4 ota_0            OTA app          00 10 000f0000 000e0000
5 ota_1            OTA app          00 11 001d0000 000e0000
6 storage          Unknown data     01 82 002b0000 00100000`

chegewara, 2019-07-19:
Most likely its not the cause of your issue, but you have very small ota partitions. Are you sure 0xe0000 is enough to store your binary?

elieDaan, 2019-07-19:
Currently it's enough and I have a lot of jpeg to store on spiffs partition..

ravibhagavandas, 2019-07-20:
Partition table looks good to me except that its missing `0x` before the hexadecimal numbers.  

Since the linker cant find the symbol `call_start_cpu0` can you inspect the bootloader elf file and check for the symbol.  To inspect the bootloader, you can use the following command: `xtensa-esp32-elf-nm build/bootloader/bootloader.elf`

elieDaan, 2019-07-22:
The result of the inspection of the bootloader is : 

`
3fff0000 A __XT_EXCEPTION_DESCS_
3fff0000 A __XT_EXCEPTION_DESCS_END__
3fff0000 A __XT_EXCEPTION_TABLE_
3fff0000 A __init_array_end
3fff0000 A __init_array_start
3fff0000 A _bss_end
3fff0000 A _bss_start
3fff0000 A _data_end
3fff0000 A _data_start
40080000 ? _etext
3fff0000 A _heap_start
3fff0000 A _lit4_end
3fff0000 A _lit4_start
40078000 A _loader_text_end
40078000 A _loader_text_start
3fff0000 A _rodata_end
3fff0000 A _rodata_start
40080000 ? _stext
40080000 A _text_end
40080000 A _text_start
         U call_start_cpu0 
`

closed 2019-07-23 by elieDaan

akukenov, 2022-02-14:
Hello @elieDaan 
Tell me please, have managed to fix your issue?
It looks to me that I have the same problem

Regards,
Andrei Kukenov