aws/amazon-freertos #962: WIFI disconnected issue

sheldon123z, 2019-07-22:
Hi,

I used shadow demo as the template to develop a small application which works with the shadow document of one virtual device on my iot console. However, I encountered a problem is that after a period of time, the wifi connection will disconnect automatically. Because the error log is derived from lower layer,I don't understand what does that mean and how to fix it.

My board is ESP32 devkit, and the error log is shown as below:
```
......
I (2158684) wifi: state: run -> init (1c0)
I (2158694) wifi: pm stop, total sleep time: 1713523208 us / 2155186422 us

I (2158694) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (2158694) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 200
I (2164684) wifi: bcn_timout,ap_probe_send_start
I (2165184) wifi: ap_probe_send over, resett wifi status to disassoc
I (2165184) wifi: state: init -> init (100)
12208 218235 [iot_thread] [INFO ][MQTT][2182350] (MQTT connection 0x3fff8880) UNSUBSCRIBE operation scheduled.
12209 218235 [iot_thread] [INFO ][MQTT][2182350] (MQTT connection 0x3fff8880, UNSUBSCRIBE operation 0x3ffd435c) Waiting for operation completion.
12210 218735 [iot_thread] [INFO ][MQTT][2187350] (MQTT connection 0x3fff8880, UNSUBSCRIBE operation 0x3ffd435c) Wait complete with result TIMEOUT.
12211 218735 [iot_thread] [ERROR][Shadow][2187350] Failed to unsubscribe from $aws/things/esp32/shadow/get/accepted, error TIMEOUT.
12212 218735 [iot_thread] [INFO ][MQTT][2187350] (MQTT connection 0x3fff8880) UNSUBSCRIBE operation scheduled.
12213 218735 [iot_thread] [INFO ][MQTT][2187350] (MQTT connection 0x3fff8880, UNSUBSCRIBE operation 0x3fffa9a8) Waiting for operation completion.
12214 219235 [iot_thread] [INFO ][MQTT][2192350] (MQTT connection 0x3fff8880, UNSUBSCRIBE operation 0x3fffa9a8) Wait complete with result TIMEOUT.
12215 219235 [iot_thread] [ERROR][Shadow][2192350] Failed to unsubscribe from $aws/things/esp32/shadow/get/rejected, error TIMEOUT.
12216 219235 [iot_thread] [INFO ][DEMO][2192350] finished the shadow operation
12217 219285 [iot_thread] [INFO ][MQTT][2192850] (MQTT connection 0x3fff8880) SUBSCRIBE operation scheduled.
12218 219285 [iot_thread] [INFO ][MQTT][2192850] (MQTT connection 0x3fff8880, SUBSCRIBE operation 0x3ffd435c) Waiting for operation completion.
12219 219785 [iot_thread] [INFO ][MQTT][2197850] (MQTT connection 0x3fff8880, SUBSCRIBE operation 0x3ffd435c) Wait complete with result TIMEOUT.
12220 219785 [iot_thread] [ERROR][Shadow][2197850] Failed to subscribe to $aws/things/esp32/shadow/get/accepted, error TIMEOUT.
12221 219785 [iot_thread] [INFO ][DEMO][2197850] finished the shadow operation
12222 219819 [iot_thread] [ERROR][MQTT][2198190] (MQTT connection 0x3fff8880) Failed to receive PINGRESP within 10000 ms.
12223 219819 [iot_thread] [WARN ][NET][2198190] Failed to close connection.
12224 219819 [iot_thread] [INFO ][MQTT][2198190] (MQTT connection 0x3fff8880) Network connection closed.
12225 219835 [iot_thread] [WARN ][MQTT][2198350] (MQTT connection 0x3fff8880) Attempt to use closed connection.
12226 219835 [iot_thread] [ERROR][MQTT][2198350] (MQTT connection 0x3fff8880) New operation record cannot be created for a closed connection
12227 219835 [iot_thread] [ERROR][Shadow][2198350] Failed to subscribe to $aws/things/esp32/shadow/get/accepted, error NETWORK ERROR.
12228 219835 [iot_thread] [INFO ][DEMO][2198350] finished the shadow operation
12229 219885 [iot_thread] [WARN ][MQTT][2198850] (MQTT connection 0x3fff8880) Attempt to use closed connection.
12230 219885 [iot_thread] [ERROR][MQTT][2198850] (MQTT connection 0x3fff8880) New operation record cannot be created for a closed connection
12231 219885 [iot_thread] [ERROR][Shadow][2198850] Failed to subscribe to $aws/things/esp32/shadow/get/accepted, error NETWORK ERROR.
12232 219885 [iot_thread] [INFO ][DEMO][2198850] finished the shadow operation
.....
```

ravibhagavandas, 2019-07-22:
Hi,

`WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 200` is for beacon timeout from the ESP-IDF [programming guide](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#wi-fi-reason-code). This means the device lost continuous 60 beacons from access point.
This could be due to a jittery network or access point or from device itself such as due to a memory leak issue. 
Does the disconnect happens at random timing from the start or exactly after a constant period of time. If the issue occurs always at the same time, then it could be worthwhile to print the available heap size at the time of disconnection. You can use FreeRTOS API `xPortGetFreeHeapSize()` to print the free heap size and  `xPortGetMinimumEverFreeHeapSize()`  to know minimum free heap size ever from start. 
Also, if you can capture the packets over WIFI link, that will help to isolate the issue to either access point or the device.

sheldon123z, 2019-07-23:
Hi,
Thank you very much for your reply.
I tested the program using the functions you gave me, and it turns out the program itself is ok, the memory wasn't leaking. 
I think the internet connection is the main problem because the disconnecting time is not fixed, sometimes it lasts a very long period, but sometimes it just disconnects right after the program start running. 
By the way, is there any methods that I can use to change the timeout period of the mqtt subscribe operation? I am in China so sometimes I cannot connect to AWS server.

Thank you !

ravibhagavandas, 2019-07-23:
You can use [IotMqtt_TimedSubscribe() ](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/standard/mqtt/include/iot_mqtt.h#L511) to pass a timeout value in Milliseconds to the subscribe operation.

pvyawaha, 2019-07-30:
Hi sheldon123z,

Closing this issue now , please reopen it if you still have issues.

closed 2019-07-30 by pvyawaha