aws/amazon-freertos #974: Amazon FreeRTOS on STM32

Amijic, 2019-07-24:
Hello,

I just wanted to ask does Amazon FreeRTOS support STM32L476RG or STM32L053?

Best regards :)

dnoal, 2019-07-25:
Thanks for this inquiry.  Neither board has been qualified up to now.  
1) The NUCLEO-L476RG based upon the STM32L476RG is not on our list of qualified MCU based boards from ST, but it is a fairly close match to the STM32L4 Discovery Kit IoT Node which we do support.  Porting to this board should be possible and you can consult our porting guide for details.  See here:  (https://docs.aws.amazon.com/freertos/latest/portingguide/porting-guide.html)
2) The STM32L053 discovery kit is not on our list of qualified MCU based boards from ST.  Memory may be a concern in supporting Amazon FreeRTOS.  See our FAQ for details here: 
https://aws.amazon.com/freertos/faqs/

Please let us know how we can assist further.

Amijic, 2019-07-25:
Greetings,

Thank you for your answer. I will to connect to AWS with nucleo board possessing STM32L476RG and external wifi module ESP8266. If something problematic occurs I will let you know.

Best regards

dnoal, 2019-07-25:
Thank you.  I will close this issue for now.  Please open a separate issue if needed.

closed 2019-07-25 by dnoal

vishaal43, 2020-01-20:
Can Amzon freertos be ported to stm32L072CZ (cortex m0+) and stm32L152 ?
Does AWS freertos support(ported) to these ICs?