aws/amazon-freertos #979: Amazon FreeRTOS vDevModeKeyProvisioning

MariusYoung92, 2019-07-25:
Greetings,

So to my current setup: I have managed to port the Amazon FreeRTOS to the STM32 IoT node controller within Atollic. So far everything works, i can publish/subscribe to topics, i can use the shadow thing. I already prepared a communication interface to upload configuration files such as the certificates. Therefore I was thinking how i can save your provided certificates (after a thing/policy and a certificate is created) within my device. Then i saw the example where you were using vDevModeKeyProvisioning. This function writes the certificates into the ROM and is converting it prio into an "*.der" file. My question is now, do you recommend to use this mechanism to upload new certificates into the device? Where is the advantage? I mean if someone would have access to the ROM he could read out the *pem file or the *der file. Is there everything correctly implemented in the STM example? So far i saw that the true random generator is being triggered, but no matter how often i am restarting the device, the saved content. if i am calling vDevModeKeyProvisioning, is still the same.

Thanks for your help and kind regards

pvyawaha, 2019-07-26:
Hi MariusYoung92,

Thank you for sharing your experience in porting on STM32 platform and getting communications working. Are you using the STM32L4 Discovery Kit IoT node?

The `vDevModeKeyProvisioning` is a simple example of provisioning keys and certificates in flash using PKCS#11 interface. The PEM format certificates are converted into DER for storing for two primary reasons - 

1. The DER formatted certificates are smaller in size.
2. It saves in computation required for conversion before they are parsed.  

The `vDevModeKeyProvisioning` is a demonstration and proper threat modelling should be done for the design to protect assets like certificates and keys and it is recommended to store them in secure storage.

escanzanocloudcath, 2019-07-29:
Hi, 

I am putting a STM32 type system into production use. I am new to dealing with the problem discussed in this case. Do you have any examples of how this is done for production systems?

Also, where/how in the AWS RTOS code does it read the certificate once vDevModeKeyProvisioning() stores it away? I am not clear how it is consumed?

Thanks

MariusYoung92, 2019-07-31:
Hi Edward,

I made a custom application where I were using the pkcs#11 API in order to store the private key and certificate. You have to customize the PAL function where you can read and store those keys (pkcs11_pal_saveobject, pkcs11_pal_findobject). There you can decide where your key and certificate should be stored, either in your ROM, an external flash or on a secure crypto chip. Now everytime a socket is being opened by amazon freeRTOS (If you want to establish a MQTT connection), the socket will use the pkcs#11 api to read the key and certificate (except you are using the offload feature from the external WIFI module). After the key and certificate is loaded corretcly, the amazon freeRTOS should care about the rest. 

Thanks to Prasad I was able to understand the purpose of vDevModeKeyProvisioning and could customize it. If you will debug the example of vDevModeKeyProvisioning  you will see that the key and certificate is being stored every time which is not needed in a real product. You have to think about a concept how you make sure to upload those certificates once after production and how to make sure that no one else can read them. Then you should have a safe system.

Kind regards

abhidixi11, 2019-08-01:
Hello MariusYoung92 ,
If there is anything more you need to know regarding vDevModeKeyProvisioning please let us know. Otherwise I'll close the issue (or feel free to close it).
Thanks.

escanzanocloudcath, 2019-08-01:
Please close it. Thanks

closed 2019-08-01 by abhidixi11