aws/amazon-freertos #988: WiFi Init: Guru meditation error

ArturBa, 2019-07-29:
**Describe the bug**
A clear and concise description of what the bug is.

**System information**
- Which hardware board or part numbers?
- IDE used
- Operating System [Windows|Linux|MacOS]
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

**To reproduce**
Steps to reproduce the behavior:

1. ...
2. ...

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
Add any other context about the problem here.

Thank you!

abhidixi11, 2019-07-31:
Hello ArturBa ,

I'm assuming , as per your comment, amazon demo involving WIFI works OK, so most likely something going wrong with your init sequence
Looking at the trace:

```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)

0x4008ddc8: esp_rom_spiflash_read_data at /home/artur/esp/hello_world/amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_rom_patch.c:412
```

I think this code is somewhere in flash erase, so it is most likely happening during flash erase of flash init.   You can comment out  vApplicationIPInit(); to make sure it flash routines are the problem.

Also, I see in the code you are executing BLE related initialization 

```
#if BLE_ENABLED
        NumericComparisonInit();
        spp_uart_init();
#endif
```
If you do not need BLE, you should remove that portion so that we can only focus on getting wifi up first.

Thanks.

ArturBa, 2019-08-01:
Hi @abhidixi11. 
You are assuming the right thing. Amazon's demo is working fine. 
I've removed the section with `BLE_ENABLED`.

Also commented out `vApplicationInit()`, but still having the same problem with esp32 init.

mahavirj, 2019-08-01:
@ArturBa Seems bit strange but your `port.c` path is not pointing to `./lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c`, which means it is not being build as a part of `libfreertos.a` library and hence (all symbols) getting placed in flash (instead of RAM due to https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/esp-idf/components/esp32/ld/esp32.common.ld#L99). Once you fix this problem, this crash should get resolved.

ArturBa, 2019-08-01:
Hi @mahavirj.
Thanks a lot for your answer. 
I've tried linking this file multiple ways, but none of them worked. 
I've tried to link it by `ln -s …`, but unsuccessful. 
Then removed all files from src except `main.c` and `wifi.c` and added espressif code as a component.
Added it to `CMakeLists.txt` using this 
```cmake
get_filename_component(EXTRA_COMPONENT_DIRS
        "components/espressif_code" ABSOLUTE
        )

# Add some external components to the project
set(IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})
```

the structure of components directory looks like this
```sh
├── espressif_code
    ├── amazon-freertos-common
    │   ├── CMakeLists.txt
    │   └── component.mk
    ├── ethernet
    │   ├── CMakeLists.txt
    │   ├── component.mk
    │   ├── include
    │   │   └── esp_eth.h
    │   └── port.c
    ├── freertos
    │   ├── CMakeLists.txt
    │   ├── component.mk
    │   ├── extra.c
    │   ├── include
    │   │   ├── freertos
    │   │   │   ├── event_groups.h
    │   │   │   ├── FreeRTOSConfig.h
    │   │   │   ├── FreeRTOS.h
    │   │   │   ├── list.h
    │   │   │   ├── portable.h
    │   │   │   ├── portmacro.h
    │   │   │   ├── queue.h
    │   │   │   ├── ringbuf.h
    │   │   │   ├── semphr.h
    │   │   │   ├── task.h
    │   │   │   ├── timers.h
    │   │   │   ├── xtensa_api.h
    │   │   │   ├── xtensa_context.h
    │   │   │   └── xtensa_timer.h
    │   │   └── freertos_tasks_c_additions.h
    │   ├── Kconfig
    │   └── ringbuf.c
    ├── mbedtls
    │   ├── CMakeLists.txt
    │   ├── component.mk
    │   ├── Kconfig
    │   ├── Makefile.projbuild
    │   └── port
    │       ├── esp_bignum.c
    │       ├── esp_hardware.c
    │       ├── esp_mem.c
    │       ├── esp_sha1.c
    │       ├── esp_sha256.c
    │       ├── esp_sha512.c
    │       ├── include
    │       │   ├── aes_alt.h
    │       │   ├── esp_mem.h
    │       │   ├── mbedtls
    │       │   │   ├── bignum.h
    │       │   │   ├── config.h
    │       │   │   ├── esp_config.h
    │       │   │   └── esp_debug.h
    │       │   ├── sha1_alt.h
    │       │   ├── sha256_alt.h
    │       │   └── sha512_alt.h
    │       └── mbedtls_debug.c
    ├── smartconfig_ack
    │   ├── CMakeLists.txt
    │   ├── component.mk
    │   ├── include
    │   │   └── smartconfig_ack.h
    │   └── smartconfig_ack.c
    └── tcpip_adapter
        ├── CMakeLists.txt
        ├── component.mk
        ├── include
        │   └── tcpip_adapter.h
        └── tcpip_adapter.c
```

How should I add `port.c` file to my project? 
Thanks for your help

abhidixi11, 2019-08-01:
Thanks @mahavirj. 

ArturBa ,
So it confirms that your code is not crashing in WIFI init, but in Flash routines. 
If you use our latest release (release 1.5), port.c is located here:
[port.c](https://github.com/aws/amazon-freertos/blob/master/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c)

> Version of Amazon FreeRTOS: 201906.00_Major-211-gdb185e9b

Are you using latest release from github? or using some other version ?  I'm not sure where the above version number is coming from.  
Hopefully using correct port.c fixes your problem
Thanks.

ArturBa, 2019-08-02:
@abhidixi11 
Hi. I've added this file to my CMakeLIsts 
`"amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/*.c"` and removed it from src files. After flashing the problem is still there. Also tried to replace it by removing and downloading straight from the GitHub, which also was unsuccessful. 

I'm using the latest version of amazon freertos – 201906.00 Major.

abhidixi11, 2019-08-02:
Are using the latest mainline from github ?
Can you please post your crash dump after updated code? 
Especially I would like see  where the crash is reported,  like the one below:
> 0x400d643c: xPortSysTickHandler at /home/artur/esp/hello_world/src/port.c:258
Thanks.

ArturBa, 2019-08-04:
Here are the newest logs after flashing. 

```sh
I (101) cpu_start: Starting scheduler on PRO CPU.
[Artur's ESP][Init] Hello there
[Artur's ESP][MiscInit] NVS Init
[Artur's ESP][MiscInit] Logging Task Init
I (131) wifi: wifi driver task: 3ffd1a28, prio:23, stack:3584, core=0
I (131) wifi: wifi firmware version: dbb148a
I (141) wifi: config NVS flash: enabled
I (141) wifi: config nano formating: disabled
I (141) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (151) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (171) wifi: Init dynamic tx buffer num: 32
I (171) wifi: Init data frame dynamic rx buffer num: 32
I (171) wifi: Init management frame dynamic rx buffer num: 32
I (181) wifi: Init static rx buffer size: 1600
I (181) wifi: Init static rx buffer num: 10
I (181) wifi: Init dynamic rx buffer num: 32
W (191) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (361) phy: phy_version: 4008, c9ae59f, Jan 25 2019, 16:54:06, 0, 2
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400d62e4  PS      : 0x00050034  A0      : 0x4008159d  A1      : 0x3ffc05f0  
0x400d62e4: xPortSysTickHandler at /home/artur/esp/hello_world/src/port.c:258

0x4008159d: _xt_lowint1 at /home/artur/esp/hello_world/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1105

A2      : 0x00186a00  A3      : 0x035c5d6a  A4      : 0x4008abd1  A5      : 0x4008ab2e  
0x4008abd1: _frxt_timer_int at /home/artur/esp/hello_world/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/portasm.S:313

0x4008ab2e: _frxt_int_enter at /home/artur/esp/hello_world/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/portasm.S:119

A6      : 0x0000884a  A7      : 0x00009368  A8      : 0x40081545  A9      : 0x00000000  
0x40081545: _xt_lowint1 at /home/artur/esp/hello_world/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1098

A10     : 0x00000000  A11     : 0x00000770  A12     : 0x00000002  A13     : 0x3ffd1530  
A14     : 0x00000000  A15     : 0x3ffd1560  SAR     : 0x00000017  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x4008dead  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d62e4
0x4008dead: esp_rom_spiflash_program_page_internal at /home/artur/esp/hello_world/amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_rom_patch.c:412

0x400d62e4: xPortSysTickHandler at /home/artur/esp/hello_world/src/port.c:258

Backtrace: 0x400d62e4:0x3ffc05f0 0x4008159a:0x3ffc0600 0x4008deaa:0x00000000
0x400d62e4: xPortSysTickHandler at /home/artur/esp/hello_world/src/port.c:258

0x4008159a: _xt_lowint1 at /home/artur/esp/hello_world/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1105

0x4008deaa: esp_rom_spiflash_program_page_internal at /home/artur/esp/hello_world/amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_rom_patch.c:412

Rebooting...
ets Jun  8 2016 00:22:57
```
Thanks

abhidixi11, 2019-08-05:
Hello ArturBa 
It looks like port.c has the same problem that @mahavirj mentioned.  Are you downloading AFR from github and following getting started guide ? Just curious why are you using flash routines?  I'm assuming if you don't use flash routines, your code works.
We will  investigate more on Monday and get back to you.
Thanks.

ArturBa, 2019-08-05:
Hi @abhidixi11 
I've downloaded AFR by git clone straight from GitHub.
Default amazon demo works. After the first flash often it goes further. 
I'm not sure what do you mean by
> I'm assuming if you don't use flash routines, your code works.

abhidixi11, 2019-08-05:
Hello ArturBa,

Thanks for response. 
If the default demo works, we would like understand  the reason you are modifying the init sequence (app_main) provided by espressif and replacing with your own. You should able to let demo runner call your application directly by replacing one of the demos instead replacing init sequence.  
I don't know what exactly is happening in your initialization sequence.  Something may be missing either in your makefiles or init sequence, that may be causing some libraries not to get linked in libfreertos.a
If you have specific reason to have your own init sequence we need to see the entire code to determine what could be going wrong. 
If your concern is the size of the code, you should able to download only ESP specific custom code from AWS FreeRTOS console : https://docs.aws.amazon.com/freertos/latest/userguide/freertos-ocw.html. 
Thanks.

ArturBa, 2019-08-06:
Hello @abhidixi11 
Thanks for the response. 
Today I've tried to create this project using main.c file from the demo, but it also failed while init. 
[Here](https://github.com/ArturBa/aws_init_error) you can check this project ;) 

Thanks a lot

anurag-kar, 2019-08-07:
@ArturBa Is there a specific reason for having a separate `src/port.c`, which I checked is exactly the same as "amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c"?

Anyways, could you remove `src/port.c` and try compiling with this cmake file:

```
cmake_minimum_required(VERSION 3.13)

project(Arturs_app)

file(GLOB SOURCES
        "amazon-freertos/demos/demo_runner/aws*.c"
        "amazon-freertos/demos/ble/*.c"
        "amazon-freertos/demos/dev_mode_key_provisioning/src/*.c"
        "src/*.c"
)

file(GLOB HEADERS "include/*.h")

add_executable(Arturs_app ${SOURCES} ${HEADERS})

set(INCLUDE_DIRS ${INCLUDE_DIRS}
        "include"
        "amazon-freertos/demos/include"
        "amazon-freertos/demos/network_manager"
        "amazon-freertos/demos/dev_mode_key_provisioning/include"
)

include_directories(${INCLUDE_DIRS})

set(IDF_PROJECT_EXECUTABLE Arturs_app)

set(IDF_SDKCONFIG_DEFAULTS "sdkconfig.defaults")

set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL "")

add_subdirectory("amazon-freertos")

target_link_libraries(Arturs_app PRIVATE
        AFR::wifi
        AFR::utils
        AFR::ble
        AFR::pkcs11
        )

target_compile_options(Arturs_app PRIVATE -Wno-pointer-sign)
```

The extra compile option is there to ignore difference in signedness at line 830 in `amazon-freertos/demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c`.

Let me know if it works for you.

ArturBa, 2019-08-07:
Hi @anurag-kar 
There is no specific reason. I've tried to make my own project basing on [this](https://github.com/aws/amazon-freertos/issues/832), so it's probably my mistake. 

your solution works fine. Thanks a lot for solving my problem. 
*Closing issue cause to the problem solved*

closed 2019-08-07 by ArturBa