aws/amazon-freertos #989: Missing to free pThreadInfo leads to memory leak in iot_threads_freertos.c

Sherryzhang2, 2019-07-29:
**Describe the bug**
DEMO_RUNNER_RunDemos call Iot_CreateDetachedThread in which pThreadInfo is malloced. And it is freed if returns from threadRoutine. When executing the task, at the end of runDemoTask, vTaskDelete( NULL ) is called as a result of which a pendsv interrupt is triggered and task reschedules to another task, then it will not return from pThreadInfo->threadRoutine( pThreadInfo->pArgument ). So the malloced pThreadInfo will not be freed. This causes memory leak.

**System information**
- Which hardware board or part numbers?    All hardware as long as I run the demo.
- IDE used      DS5
- Operating System [Windows|Linux|MacOS]    All
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)   201906.00_Major-428-g026fc88e
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]     MQTT Demo
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
A clear and concise description of what you expected to happen.
No memory leak when I run the demo.
**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

**To reproduce**
Steps to reproduce the behavior:

Just run the MQTT demo on any target board.

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
Add any other context about the problem here.

Thank you!

aggarg, 2019-07-29:
Thank you taking time and reporting this issue. We have fixed this in this PR: https://github.com/aws/amazon-freertos/pull/991

Thanks.

Sherryzhang2, 2019-07-30:
Thanks for fixing it.

closed 2019-07-30 by Sherryzhang2