aws/amazon-freertos #995: Speed up the subscribe/unsubscribe process

sheldon123z, 2019-07-30:
Hi,
I am currently doing a project using esp32 to get thing shadow from IoT Core, my project now is encountering a problem is that the response time from IoT console is too slow. I used the `AwsIotShadow_Get` function and `AwsIotShadow_Wait` to acquire the shadow document, but it usually takes more than 3 or 4 seconds to finish this operation. I wonder what's caused this lag and if I can do anything to improve it?

Thank you very much & Regards !
Sheldon

abhidixi11, 2019-07-31:
Hello sheldon123z ,

Thanks for contacting us.  Can you please let us know what is the size of shadow document ?  Also, is it possible to get wire trace ?  It will be helpful to see packet trace from subscribe to response.
Thanks.

sheldon123z, 2019-08-02:
Hi,
The shadow document is not very large. The most time consuming part is the SUBSCRIBE and UNSUBSCRIBE operation. Also I am very confused why each GET operation will need to do two subscribe operations? Is that because one subscription is for `get` and one operation is for `get/accept` topic? Also, can I just use a persistent subscription to continuously get the shadow document without doing the UNSUBSCRIBE operation? 

Here is a part of the running log:
```
[2019-08-02 10:51:55.357] I (3349) WIFI: SYSTEM_EVENT_STA_CONNECTED
[2019-08-02 10:51:58.337] 3 602 [IP-task] vDHCPProcess: offer c0a80203ip
[2019-08-02 10:51:58.344] I (6349) event: sta ip: 192.168.2.3, mask: 255.255.255.0, gw: 192.168.2.1
[2019-08-02 10:51:58.353] I (6349) WIFI: SYSTEM_EVENT_STA_GOT_IP
[2019-08-02 10:51:58.357] 4 603 [IP-task] vDHCPProcess: offer c0a80203ip
[2019-08-02 10:51:58.360] 5 603 [iot_thread] [INFO ][DEMO][6030] Successfully initialized the demo. Network type for the demo: 1
[2019-08-02 10:51:58.369] 6 603 [iot_thread] [INFO ][DEMO][6030] thingNameLength is 5, Identifier esp32
[2019-08-02 10:51:58.378] 7 603 [iot_thread] [INFO ][MQTT][6030] MQTT library successfully initialized.
[2019-08-02 10:51:58.384] 8 603 [iot_thread] [INFO ][Shadow][6030] Shadow library successfully initialized.
[2019-08-02 10:51:58.393] 9 603 [iot_thread] [INFO ][DEMO][6030] Shadow Thing Name is esp32 (length 5).
[2019-08-02 10:52:08.873] 10 1656 [iot_thread] [INFO ][MQTT][16560] Establishing new MQTT connection.
[2019-08-02 10:52:08.881] 11 1656 [iot_thread] [INFO ][MQTT][16560] (MQTT connection 0x3ffd4ee8, CONNECT operation 0x3ffd4bec) Waiting for operation completion.
[2019-08-02 10:52:09.180] 12 1686 [iot_thread] [INFO ][MQTT][16860] (MQTT connection 0x3ffd4ee8, CONNECT operation 0x3ffd4bec) Wait complete with result SUCCESS.
[2019-08-02 10:52:09.193] 13 1686 [iot_thread] [INFO ][MQTT][16860] schedule keep alive job here
[2019-08-02 10:52:09.202] 14 1686 [iot_thread] [INFO ][MQTT][16860] New MQTT connection 0x3ffde2a0 established.
[2019-08-02 10:52:09.207] 15 1686 [iot_thread] [INFO ][DEMO][16860] free heap size is 32368 bytes 
[2019-08-02 10:52:09.213] 16 1686 [iot_thread] [INFO ][DEMO][16860] entered getThingshadow function
[2019-08-02 10:52:09.222] 17 1686 [iot_thread] [INFO ][MQTT][16860] (MQTT connection 0x3ffd4ee8) SUBSCRIBE operation scheduled.
[2019-08-02 10:52:09.233] 18 1686 [iot_thread] [INFO ][MQTT][16860] (MQTT connection 0x3ffd4ee8, SUBSCRIBE operation 0x3fff8504) Waiting for operation completion.
[2019-08-02 10:52:09.489] 19 1717 [iot_thread] [INFO ][MQTT][17170] (MQTT connection 0x3ffd4ee8, SUBSCRIBE operation 0x3fff8504) Wait complete with result SUCCESS.
[2019-08-02 10:52:09.511] 20 1717 [iot_thread] [INFO ][MQTT][17170] (MQTT connection 0x3ffd4ee8) SUBSCRIBE operation scheduled.
[2019-08-02 10:52:09.523] 21 1717 [iot_thread] [INFO ][MQTT][17170] (MQTT connection 0x3ffd4ee8, SUBSCRIBE operation 0x3ffd49c0) Waiting for operation completion.
[2019-08-02 10:52:09.791] 22 1747 [iot_thread] [INFO ][MQTT][17470] (MQTT connection 0x3ffd4ee8, SUBSCRIBE operation 0x3ffd49c0) Wait complete with result SUCCESS.
[2019-08-02 10:52:09.805] 23 1748 [iot_thread] [INFO ][MQTT][17480] (MQTT connection 0x3ffd4ee8) MQTT PUBLISH operation queued.
[2019-08-02 10:52:11.332] 24 1902 [iot_thread] [INFO ][Shadow][19010] Shadow GET of esp32 was ACCEPTED.
[2019-08-02 10:52:11.340] 25 1902 [iot_thread] [INFO ][Shadow][19020] pDocument here,payloadLength is 1723
[2019-08-02 10:52:11.347] 26 1902 [iot_thread] [INFO ][MQTT][19020] (MQTT connection 0x3ffd4ee8) UNSUBSCRIBE operation scheduled.
[2019-08-02 10:52:11.365] 27 1902 [iot_thread] [INFO ][MQTT][19020] (MQTT connection 0x3ffd4ee8, UNSUBSCRIBE operation 0x3ffd4bec) Waiting for operation completion.
[2019-08-02 10:52:11.644] 28 1932 [iot_thread] [INFO ][MQTT][19320] (MQTT connection 0x3ffd4ee8, UNSUBSCRIBE operation 0x3ffd4bec) Wait complete with result SUCCESS.
[2019-08-02 10:52:11.651] 29 1932 [iot_thread] [INFO ][MQTT][19320] (MQTT connection 0x3ffd4ee8) UNSUBSCRIBE operation scheduled.
[2019-08-02 10:52:11.666] 30 1932 [iot_thread] [INFO ][MQTT][19320] (MQTT connection 0x3ffd4ee8, UNSUBSCRIBE operation 0x3ffd4bec) Waiting for operation completion.
[2019-08-02 10:52:11.945] 31 1963 [iot_thread] [INFO ][MQTT][19620] (MQTT connection 0x3ffd4ee8, UNSUBSCRIBE operation 0x3ffd4bec) Wait complete with result SUCCESS.
[2019-08-02 10:52:11.978] 32 1963 [iot_thread] [INFO ][DEMO][19630] Get Shadow document, begin to analyze the retrived shadow document
[2019-08-02 10:52:11.979] 33 1963 [iot_thread] [INFO ][DEMO][19630] attributeFound
[2019-08-02 10:52:11.979] 34 1963 [iot_thread] [INFO ][DEMO][19630]  command is :"ON" 
[2019-08-02 10:52:11.979] 35 1963 [iot_thread] [INFO ][DEMO][19630] Write command to uart port successful! the data is :"ON"
[2019-08-02 10:52:11.987] 
[2019-08-02 10:52:11.987] 36 1963 [iot_thread] [INFO ][MQTT][19630] (MQTT connection 0x3ffd4ee8) SUBSCRIBE operation scheduled.
[2019-08-02 10:52:11.996] 37 1963 [iot_thread] [INFO ][MQTT][19630] (MQTT connection 0x3ffd4ee8, SUBSCRIBE operation 0x3fff8504) Waiting for operation completion.
[2019-08-02 10:52:12.250] 38 1993 [iot_thread] [INFO ][MQTT][19930] (MQTT connection 0x3ffd4ee8, SUBSCRIBE operation 0x3fff8504) Wait complete with result SUCCESS.
[2019-08-02 10:52:12.263] 39 1993 [iot_thread] [INFO ][MQTT][19930] (MQTT connection 0x3ffd4ee8) SUBSCRIBE operation scheduled.
[... cut ...]
[2019-08-02 10:52:15.017] 50 2270 [iot_thread] [INFO ][MQTT][22700] (MQTT connection 0x3ffd4ee8, UNSUBSCRIBE operation 0x3ffd495c) Wait complete with result SUCCESS.
[2019-08-02 10:52:15.031] 51 2270 [iot_thread] [INFO ][DEMO][22700] Get Shadow document, begin to analyze the retrived shadow document
[2019-08-02 10:52:15.040] 52 2270 [iot_thread] [INFO ][DEMO][22700] attributeFound
[2019-08-02 10:52:15.045] 53 2270 [iot_thread] [INFO ][DEMO][22700]  command is :"ON" 
[2019-08-02 10:52:15.050] 54 2270 [iot_thread] [INFO ][DEMO][22700] Write command to uart port successful! the data is :"ON"
[2019-08-02 10:52:15.062] 
```

abhidixi11, 2019-08-05:
Hello sheldon123z ,

I do see two subscribe and unsubscribe operations from your log.  Can you please share the application code that you are using to get the shadow document ?  

> Also, can I just use a persistent subscription to continuously get the shadow document without doing the UNSUBSCRIBE operation?

I think it should be possible , have you tried keeping persistent connection ?  I'll investigate more and let you know on Monday.
Thanks.

sheldon123z, 2019-08-05:
Hi abhidixi11,
Yes, here is the application code, I actually just modified the original code:
```

/* The config header is always included first. */
#include "iot_config.h"

/* Standard includes. */
#include <stdbool.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <stdarg.h>

/* Set up logging for this demo. */
#include "iot_demo_logging.h"

/* Platform layer includes. */
#include "platform/iot_clock.h"
#include "platform/iot_threads.h"

/* MQTT include. */
#include "iot_mqtt.h"

/* Shadow include. */
#include "aws_iot_shadow.h"

/* JSON utilities include. */
#include "iot_json_utils.h"

#include "FreeRTOSIPConfig.h"

/*Uart headers */
#include "FreeRTOS.h"
#include "task.h"
#include "driver/uart.h"
#include "aws_iot_shadow_blem.h"
/*
 * - Port: UART1
 * - Receive (Rx) buffer: on
 * - Transmit (Tx) buffer: off
 * - Flow control: off
 * - Event queue: off
 * - Pin assignment: see defines below
*/
#define ECHO_TEST_TXD (GPIO_NUM_16)
#define ECHO_TEST_RXD (GPIO_NUM_17)
#define ECHO_TEST_RTS (UART_PIN_NO_CHANGE)
#define ECHO_TEST_CTS (UART_PIN_NO_CHANGE)

#define BUF_SIZE (1024)

/**
 * Provide default values for undefined configuration settings.
 */

#define SHADOW_UPDATE_PERIOD_MS (2000)

#define SHADOW_GET_PERIOD_MS (2000)

/* Validate Shadow demo configuration settings. */

#if SHADOW_UPDATE_PERIOD_MS <= 0
#error "SHADOW_UPDATE_PERIOD_MS cannot be 0 or negative."
#endif

/**
 * @brief The keep-alive interval used for this demo.
 *
 * An MQTT ping request will be sent periodically at this interval.
[... 920 lines cut ...]
    }

    if(status == EXIT_SUCCESS)
    {
        IotLogInfo("free heap size is %d bytes ",xPortGetFreeHeapSize());
        status = _thingShadowOperation( mqttConnection,
                                        pIdentifier,
                                        thingNameLength);
    }
    
    /* Disconnect the MQTT connection if it was established. */

    if (connectionEstablished == true)
    {
        IotMqtt_Disconnect(mqttConnection, 0);
    }

    /* Clean up libraries if they were initialized. */
    if (librariesInitialized == true)
    {
        _cleanupDemo();
    }
    return status;
}
```

htibosch, 2019-08-05:
Hi @sheldon123z, would you mind attaching the FreeRTOSIPConfig that you are using?
The settings of FreeRTOS+TCP are also determining the communication speed. Hein

gordonwang0, 2019-08-05:
Hi @sheldon123z,

Some things to speed up Shadow operations:
- Use QoS 0 by setting [AwsIotShadowDocumentInfo.qos](https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/shadow/struct_aws_iot_shadow_document_info__t.html#a2b89b6a311b231123824b941d4fd890e) to IOT_MQTT_QOS_0.  QoS 1 for Shadows doesn't really provide any benefit because Shadow already has an acknowledgement mechanism.
- Pass the flag [AWS_IOT_SHADOW_FLAG_KEEP_SUBSCRIPTIONS](https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/shadow/aws__iot__shadow__types_8h.html#ae5bef45f32f283af7d8d0dcb1300f5b0) flag to the Shadow GET call.  This will cause the subscriptions to be maintained across calls to Shadow GET.  When you no longer need persistent subscriptions, you can call [AwsIotShadow_RemovePersistentSubscriptions](https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/shadow/shadow_function_removepersistentsubscriptions.html) or [AwsIotShadow_Cleanup](https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/shadow/shadow_function_cleanup.html) to remove them.

alexa-noxon, 2019-09-03:
Closing this issue.  Feel free to re-open if there is still an issue.

closed 2019-09-03 by alexa-noxon