aws/amazon-freertos #1006: Missing _VECT in #pragma interrupt (Renesas RX100 series)

Plornel, 2019-08-01:
Illegal #pragma interrupt in port.c l.299 for Renesas RX100 series

Should be : 
```c
#pragma interrupt ( prvTickISR( vect = _VECT( configTICK_VECTOR ), enable ) )
```
Instead of : 
```c
#pragma interrupt ( prvTickISR( vect = configTICK_VECTOR, enable ) )
```

abhidixi11, 2019-08-01:
Thanks Plornel, we will take a look. I'll post update soon.

abhidixi11, 2019-08-02:
Hello Plornel,
Did you experience problem with the pragma statement with the latest compiler?  Amazon FreeRTOS does not support RX100 port.c, so I would like make sure. 
Thanks.

Plornel, 2019-08-05:
Hello abhidixi11,
I encountered no problem with the syntax I suggest.

abhidixi11, 2019-08-05:
Hello Plornel,

Thanks for the response. I was wondering if you experience any problem with the existing pragma statement ?  (before addition of _VEC).
Thanks.

Plornel, 2019-08-07:
With the existing statement I have an error "Illegal #pragma interrupt declaration" (toolchain CCRX 3.01.00)

yuhui-zheng, 2019-08-07:
@Plornel Thank you. We are discussing internally for the other potential impact by merging this change. Shall be able to either merge directly from our side or get back to you with reasoning. Thanks again for bring it up.

RichardBarry, 2019-08-08:
The important thing here is the compiler version - the syntax needs to be correct for the latest compiler - which compiler version are you using?

alexa-noxon, 2019-08-30:
@Plornel This has been resolved by #1013.  I will resolve the ticket.  Thank you again for pointing this out!

closed 2019-08-30 by alexa-noxon