aws/amazon-freertos #1042: Integrating OTA with GreenGrassDiscovery code

pegasus-automation, 2019-08-08:
Hello,
    I am using esp32 DevKitC and raspi 3b+ running amazon freertos and GreenGrassCore respectively. I was able to successfully test the GreenGrassDiscovery demo code available and also the OTA code in separately but merging the OTA fragment (referring to this: [https://docs.aws.amazon.com/freertos/latest/userguide/integrate-ota-agent.html](https://docs.aws.amazon.com/freertos/latest/userguide/integrate-ota-agent.html)) with the GreenGrassDiscovery code I am facing issues.

How have I merged the fragment?
`if( prvMQTTConnect( &xHostAddressData ) == 
{
configPRINTF(  ( "Connection Successfully Done!\n" )  );
OTA_AgentInit( xMQTTClientHandle, ( const uint8_t * ) ( clientcredentialIOT_THING_NAME ), NULL, ( TickType_t ) ~0 );
printf("OTA Init DONE!\n");
}`

Probable issue what I suspect is, with GreenGrassDiscoveryCode we need to create an MQTT client which is used while publishing, subscribing, etc. and same goes for the OTA fragment available in the document. This can be the issue with the project. Hence what I tried doing instead is to use the same client of GreenGrassDiscovery task in the OTA fragment too. But this gives us an error.

Can you help us with how to integrate the code?

yuhui-zheng, 2019-08-08:
Hi @pegasus-automation,

The problem can be twofold -- things specifically on device, on device-server contract. 

When having two clients may want to check things are still within expectation from server's perspective. Server-client contract (really limitation) is mostly define here. 
https://docs.aws.amazon.com/general/latest/gr/aws_service_limits.html#limits_iot
May be relevant to this incident -- "Connect requests per second per client ID". 

On device side though, could you send us things logged to screen? This will help us understand your sequence. (Or if you could summarize your sequence that'll even be better. Also, things may help us -- which version of code/library you are using?)

As for NTP, unfortunately we do not NTP library publicly available. I will bring this it up to stakeholders, see if we can prioritize it. Though, would it be possible to send timestamp in MQTT message to device?

pegasus-automation, 2019-08-09:
Hello yuhui-zheng,
     Thanks for an early response. As far as I understand, you want me to explain our sequence and here goes it with the code.

The sequence of the code:
1) I want to publish and subscribe to a few topics on AWS IoT via GreenGrassCore (basically using the GreenGrassConnectivity code and customizing it!). Hence the routine consists of regular publish-subscribe to AWS IoT topics.
2) At the same time parallelly I want my esp32 to run a task (using xTaskCreate ??) to check if there is an OTA update available after every specified time interval (maybe every hour...).

Code Snippet:
1) Only GreenGrassDiscoveryTask (in the image1)
[image]

2) Only OTA code I am using is the default code provided in the link I previously referenced.

3) Tried Integrating both the codes (in the image2)
[image]

`/*-----------------------------------------------------------*/

static void prvDiscoverGreenGrassCore( void * pvParameters )
{
    GGD_HostAddressData_t xHostAddressData;

    ( void ) pvParameters;

    /* Create MQTT Client. */
    if( MQTT_AGENT_Create( &( xMQTTClientHandle ) ) == eMQTTAgentSuccess )
    {
        memset( &xHostAddressData, 0, sizeof( xHostAddressData ) );

        /* Demonstrate automated connection. */
        configPRINTF( ( "Attempting automated selection of Greengrass device\r\n" ) );

        if( GGD_GetGGCIPandCertificate( pcJSONFile,
                                        ggdDEMO_DISCOVERY_FILE_SIZE,
                                        &xHostAddressData )
            == pdPASS )
        {
            configPRINTF( ( "Greengrass device discovered.\r\n" ) );
            configPRINTF( ( "Establishing MQTT communication to Greengrass...\r\n" ) );
           
            configPRINTF(  ( "Connecting The Topics\n" )  );

            if( prvMQTTConnect( &xHostAddressData ) == pdPASS )
            {
                configPRINTF(  ( "Connection Successfully Done!\n" )  );
               
                // OTA Initialize
                OTA_AgentInit( xMQTTClientHandle, ( const uint8_t * ) ( clientcredentialIOT_THING_NAME ), NULL, ( TickType_t ) ~0 );
            }

            // Example of publish routine
            publishMessage( &xHostAddressData, ggdDEMO_MQTT_MSG_TOPIC, ggdDEMO_MQTT_MSG_DISCOVERY );
            publishMessage( &xHostAddressData, ggdDEMO_MQTT_MSG_TOPIC1, ggdDEMO_MQTT_MSG_DISCOVERY1 );

            // Actual Functions of GGD will work here
            /**
                xTaskCreatePinnedToCore(1..);
            **/
            ( void ) xTaskCreate( OTAupdate, "OTAupdate", 1000, &xHostAddressData , 10, NULL );

            /* Report on space efficiency of this demo task. */
            #if ( INCLUDE_uxTaskGetStackHighWaterMark == 1 )
                {
                    configPRINTF( ( "Heap low watermark: %u. Stack high watermark: %u.\r\n",
                                    xPortGetMinimumEverFreeHeapSize(),
                                    uxTaskGetStackHighWaterMark( NULL ) ) );
                }
            #endif
        }
        else
        {
            configPRINTF( ( "Auto-connect: Failed to retrieve Greengrass address and certificate.\r\n" ) );
        }
    }

    configPRINTF( ( "----Demo finished----\r\n" ) );
    vTaskDelete( NULL );
`

Questions:
1) What is the way I can make it out work? In my system, I am using one GreenGrassCore (on Raspi 3B+) and multiple amazon-freertos (esp32 DevKitC). I want the complete system to update over-the-air. For GreenGrassCore there is a direct deployment available wherein I can update it. Similarly, for OTA of amazon-freertos, I thought of using the code fragment available in the docs. Is there a better way to bulk update all the things of a certain group other than this (Greengrass group?)

2) Regarding the current timestamp, we need it on FreeRtos since when we are publishing the data on some topic we need to publish it with a time stamp. Is there a way I can obtain current date and time on aws infrastructure (since NTP isn't available) so that I can include it in our JSON msg.

yuhui-zheng, 2019-08-09:
While we are trying to setup GreenGrass + OTA on our end, a couple of notes... 

- Green Grass uses about ~8K memory, and ESP requires ~70K per MQTT connection. Memory could be a concern at some point... 

- Please confirm your two MQTT connections are using different client ID at least. Otherwise may subject to server limitation. "AWS IoT limits MQTT CONNECT requests from the same accountId and clientId to 1 MQTT CONNECT operation per second."

- As for NTP, a temporary solution could be using AWS Lambda to handle request, and send out timestamp. By courtesy of a colleague, a code snippet for AWS Lambda can look like -- 

```
# Pair with a trivial IoT Rule: 
# SELECT {} from 'iot/timestamp/request'
# - Success action, send to lambda 
# - Error action, republish to iot/timestamp/errors
 
# Lambda body
import json
import boto3
import datetime
 
throttle_seconds = 5
TOPIC = "iot/timestamp"
last_broadcast = None
 
def lambda_handler(event, context):
    # There is a little throttling here to prevent any lambda instance from spamming the timestamp channel
    global last_broadcast
    timestamp = datetime.datetime.now()
    
    if (last_broadcast is None or (timestamp - last_broadcast).total_seconds() > throttle_seconds ):
        last_broadcast = timestamp
        client = boto3.client('iot-data')
        client.publish(topic=TOPIC,qos=0,payload=timestamp.strftime("%s"))
        
    
    return {
        'statusCode': 200,
        'body': json.dumps(last_broadcast.strftime("%s"))
    }
```

In this case, device publishes to MQTT topic ```iot/timestamp/request```. AWS backend is setup in a way that, once IoT endpoint receives this MQTT topic an AWS Lambda function will be triggered to publish timestamp to ```iot/timestamp``` MQTT topic. A relevant tutorial can be found here -- https://docs.aws.amazon.com/iot/latest/developerguide/iot-lambda-rule.html. 

(As for how far you may want to go with this Lambda solution, I guess it depends on your timeline for NTP requirement... We have noticed that this is a popular request.)

pegasus-automation, 2019-08-12:
Hello yuhui-zheng,
     I have still not tested what you suggested to me. However I have a few questions,

1) So, you suggest that in order to work esp32 GreenGrassDiscovery Code and OTA update code, I need to make two separate MQTT client using `MQTT_AGENT_Create( &( xMQTTClientHandle ) )`. I use one of the clients for GreenGrassDiscovery and the other one for OTA updates? Is it possible to have 2 MQTT clients for one IoT thing (ESP32)?

2) Also another issue, specifically regarding the GreenGrassDiscovery Code, I need to run 4 parallel tasks so I created 4 parallel functions using xTaskCreate(). But the weirdy issue is that not all of them runs. This means that the only first 2 function which I create using xTaskCreate() runs seamlessly. The other two which are created just below the first two task doesn't get called too. I know it seems too weird to be true but I am facing this from past a week and haven't obtained a fix for this. Weirder part is that the same code was working seamlessly when I was working on an older version of freertos (December' 18). It will be very very helpful of you if you can help me of this. I can help you with the logs if you want.

yuhui-zheng, 2019-08-13:
Commenting on the second bullet point first, as that may solve some part of the problem. 

2, There are quite some amount of differences between an older version of amazon-freertos (in your case Dec' 18) and our recent major release (tagged with ```201906.00_Major```). In regards to "the weirdy issue is that not all of them runs", may want to check at which priority your tasks are running at. (In ```201906.00_Major```, we bumped task priority for multiple tasks, and introduced task pool which also creates tasks. In the case that your tasks are created with lower priority, I doubt they'll get a chance to run. Could also set break point at ```xTaskCreate``` and ```xTaskCreateStatic```, and see which priority other tasks are created with.)

1, It's a good question for us to consider as well -- which design pattern user should follow -- single client vs. multiple clients. I would agree with you that in memory constrained devices, which covers most cases, single client probably makes more sense with current MQTT implementation. And, our demo is not yet setup in a way that it's obvious to user how an MQTT client can be shared between top layer applications. As for two clients approach, besides the memory concern, it does provide some benefits (namely multiple destination forwarding for wild card topics). 

-------- implementation details --------
Some notes for each approach.
If you choose to go with two clients, could mostly copy over established demos with these edits recommended -- 
* Make sure to use different client ID for different MQTT connection. 
* Please stick with APIs defined in ```iot_mqtt_agent.h```. (OTA is using ```iot_mqtt_api.c```, which we need to clean up.)
* Memory is a concern. (esp32 DevKitC has 520KB SRAM, may fit in two agents.)

If you choose to go with single client --
* Initialize agent once, before OTA/GreenGrass functional demo starts. (May need to tweak our demo a little bit, to separate connection out of the functional code.)
* In case you application needs to response to server published messages, please refer to [IotMqtt_Subscribe() API](https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/mqtt/mqtt_function_subscribe.html) to see how to have multiple topic filters/callbacks for a single MQTT agent. (Typically [IotMqttSubscription_t]( https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/mqtt/struct_iot_mqtt_subscription__t.html).)
* If the application needs to forward a wildcard topic to more than one application (or broadcast), it may not work due to https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/standard/mqtt/src/iot_mqtt_subscription.c#L331.

pegasus-automation, 2019-08-13:
Hello yuhui-zheng,
     For the 2nd point of running task in parallel. They are as follows,

Task 1 - Subscribe Task (Subscribe to an MQTT Topic and does some computation over it.)
Task 2 - Peripherals related (This task reads GPIO input from esp32 and does a little bit of computation)
Task 3 - Publish Task (Based on computation results, it publishes value to a certain MQTT Topic.)
Task 4 - Frequent Publish Task (This task uploads sensor data every second to an MQTT Topic other than that used in Task 3)

All the four tasks are supposed to run for an infinite time period in parallel to each other.

Now at what priority have I been using them? All of these tasks are defined using xTaskCreate with exactly equal priority in all the four.

`( void ) xTaskCreate( task1, "Task1", democonfigGREENGRASS_DISCOVERY_TASK_STACK_SIZE, xMQTTsubHandle, democonfigDEMO_PRIORITY, NULL );
        
        ( void ) xTaskCreate( task2, "Task2", democonfigGREENGRASS_DISCOVERY_TASK_STACK_SIZE, xMQTTsubHandle, democonfigDEMO_PRIORITY, NULL );
        
        ( void ) xTaskCreate( task3, "Task3", democonfigGREENGRASS_DISCOVERY_TASK_STACK_SIZE, NULL, democonfigDEMO_PRIORITY, NULL );
        
        ( void ) xTaskCreate( task4, "Task4", democonfigGREENGRASS_DISCOVERY_TASK_STACK_SIZE, NULL, democonfigDEMO_PRIORITY, NULL );`

Is there a better way to define this function? Or some modification with which I can work it out? I am noobie at this, so help on this matter will ease down a lot for me. Thanks in advance!

Regarding merging with OTA, I will try to integrate it the way you specified and if found any issue will post it here.

yuhui-zheng, 2019-08-13:
Task creation itself looks good fine to me. [xTaskCreate() API](https://www.freertos.org/a00125.html). And using the same priority among these four should be fine. Could also refer to```user=rtel```'s explanation on ```configUSE_PREEMPTION``` and ```configUSETIMESLICING``` -- https://www.freertos.org/FreeRTOS_Support_Forum_Archive/February_2018/freertos_Preemption_by_an_equal_priority_task_c8cb93d0j.html

I'm assuming you are modifying a working demo project, which implies the interrupt configuration and scheduler are setup correctly. In this context, I'd check the value of ```democonfigDEMO_PRIORITY``` and ```democonfigGREENGRASS_DISCOVERY_TASK_STACK_SIZE``` in your current project. (```democonfigGREENGRASS_DISCOVERY_TASK_STACK_SIZE``` is quite large, may want to confirm the value still makes sense, and may also want to check the return value of ```xTaskCreate ```. Having tasks with priority higher than ```democonfigDEMO_PRIORITY``` _may_ result your tasks not getting run in time. )

pegasus-automation, 2019-08-19:
Hello yuhui-zheng,

Issue - 1: Resolved) The solution you suggested for handling multiple task creation worked for us. Thank you.
Issue - 2) Regarding integrating OTA with GreenGrassDiscovery. As you suggested we tried to implement using 2 different clients. Following is our implementation,

--> We wrote two parallel functions. 1st function publishes and subscribes to AWS IoT whereas 2nd function runs the OTA routine i.e. vRunOTAUpdateDemo() (the demo code available!). On executing the code, in function prxCreateNetworkConnection(), there is an undefined reference to xMqttDemoCreateNetworkConnection().

Log for the same is pasted below,
[image]

Issue - 3) There is one more issue in GreenGrassDiscovery. We have to restart GreenGrassCore Software on raspberry pi for establishing a new connection of esp32 (AWS IoT Thing!). So basically every time when we restart the esp32, we need to restart the GreenGrassCore too to have a successful MQTT connection. Though we have also noticed that the demo code for GreenGrassDiscovery doesn't require us to restart the software. Any fix for this?

huguesBouvier, 2019-08-20:
Hello,

The demo are not designed originally to work in tandem. So you may have to do a big overhaul.
Regarding issue 2: You can probably fix the compile issue by including  #include "aws_iot_demo_network.h" header in Greengrass demo file.

I am sorry I have never heard of issue 3. Could you provide more details? Why would you need a restart? Why greengrass and not the device?  The only thing I could see is the random generation for tcp sequence number is not working properly.
Could you do a Wireshark analysis?

pegasus-automation, 2019-08-22:
Hello, @yuhui-zheng and @huguesBouvier,

We have already included the "aws_iot_network.h" file in the header of the GreenGrassDemo file. Can you please lookout for any other issue? A small help on the matter will be highly appreciated. Thanks in advance.
**Regarding Issue - 3**,
Abstract of our system: In our system design, we power the GreenGrassCore and AWS IoT things (esp32, amazon freertos) in isolation. So in case if there is a power cut-off, any device can get turned off in any order. So, for a singular case when everything was well connected and now if ESP32 turns off (due to any reason) and then again get powered on. The code starts executing from scratch. So, it again attempts to connect to the GreenGrassCore which is already running on Raspberry Pi. In this scenario what I am observing is though the GreenGrassCore is running very well in the Pi the esp32 can't just connect to the GreenGrassCore. Now if I just restart the GreenGrassCore with the command "sudo ./greengrassssd restart", esp32 can successfully connect. Till then I didn't restart the GreenGrassCore it just keeps attempting and can't connect to the GGC. Can you please help on this too. Thanking you in advance.

huguesBouvier, 2019-08-23:
Hello,
For issue 3: Of course let's try to troubleshoot it! I'll need a lot more data to go forth.
One thing I could see is the random generation for tcp sequence number is not working properly. Wireshark could allow use to troubleshoot that but also to dive deeper into the problem.

For issue2: This is a compile issue that is specific to code you added, I don't have the source for it.
But again even if you added the header file, it mean that somehow that source code is not seeing the function declaration. One could be to manually add it as "extern thatfunction" in your source file directly and then slowly move it down in the header file until the issue starts appearing.

Best,
Hugues

pegasus-automation, 2019-08-30:
Hey,
I gave it a thought and decided to start that specific section of code from scratch for which I have tried to combine the demo code of Greengrass into the demo code of OTA. I am attaching the code here itself.

[aws_iot_ota_update_demo.c.txt](https://github.com/aws/amazon-freertos/files/3561184/aws_iot_ota_update_demo.c.txt)

Whenever I try to run this code something like this appears:
[image]

pegasus-automation, 2019-09-04:
@yuhui-zheng @huguesBouvier can you please have a look at this issue?

yuhui-zheng, 2019-11-26:
It seems quite some time since I ended my last oncall shift... And seems this is still open. Apologies.

Credit to a colleague, "OTA over Greengrass on ESP32" works with minor code change:

- Change MQTT Quality of Service from QoS1 to QoS0 in the OTA Agent.
- Use the same MQTT connection in GreenGrass and OTA. 

Please take a look at this AWS IoT forum post for more details -- https://forums.aws.amazon.com/thread.jspa?threadID=310051&tstart=0. 
(You'll need to login to AWS account first.)

Since this post has been in active, I'm closing it for now. If more questions, please open new support request.

closed 2019-11-26 by yuhui-zheng