aws/amazon-freertos #1052: Microchip Curiosity Demo Ethernet Not working

shanmugav, 2019-08-11:
**Describe the bug**
The demo can not work with the Ethernet adapter anymore.
The demo works fine with Wi-Fi.

**System information**
pic32mz_ef_curiosity with Ethernet
- MPLAB X 5.20
- Windows
- Version of Amazon FreeRTOS - 
201906.00_Major-493-g7721fda4
- Project [MQTT]. 

**Expected behavior**
MQTT Connectivity fails. The DNS GetHostByName fails returning IP address as zero. 
Note: DHCP is working fine, the same demo runs works fine with WIFI, if i removed the definition PIC32_USE_ETHERNET.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

**To reproduce**
Steps to reproduce the behavior:

1. Just clone amazon-freertos
2. update the aws client credential information, 
3. Define PIC32_USE_ETHERNET in MPLABX project configuration
4. Run the demo.

Thank you!

htibosch, 2019-08-11:
HI @shanmugav, your complaint looks very similar to my earlier issue [#831](https://github.com/aws/amazon-freertos/issues/831)

> The DNS GetHostByName fails returning IP address as zero

Please note that by default the demo will connect to "", an empty string, which can not be found by DNS.
I think that you have to define `clientcredentialMQTT_BROKER_ENDPOINT` in `aws_clientcredential.h`, could that be?

shanmugav, 2019-08-11:
I have set this string, that is why, it works with WI-FI!!!

htibosch, 2019-08-11:
> Note: DHCP is working fine, the same demo runs works fine with WIFI

Which means that you did define the broker end-point :-)

In that case, maybe it is useful to show a PCAP of the DNS session? Please filter the records with e.g. `ip.addr==192.168.x.x`

htibosch, 2019-08-11:
and if a creating a PCAP is difficult, you might want to enable +TCP logging, see `FreeRTOS_printf()` and `FreeRTOS_debug_printf()`.
Do you see any other network activity from the demo board? It should send so-called gratuitous ARP requests every 30 seconds or so.
Can you ping the board?

shanmugav, 2019-08-11:
Thanks, I used a different PHY and now the issue is closed.

closed 2019-08-11 by shanmugav