aws/amazon-freertos #1067: MQTT connection error

izaton, 2019-08-13:
I'm trying to connect to mqtt using my esp32 device and I get ``` Connect failure. Error 5 ``` [Network Error]

**System information**
- Hardwere: ESP32
- IDE used: Clion
- Operating System: Linux
- Version of Amazon FreeRTOS: 201906.00_Major-225-g523c08a3
- Project: Custom Application, MQTT
`Code to reproduce the bug`:

```c
IotMqttError_t _MqttConnect(void){
// Parameters to MQTT connect.
    IotMqttConnection_t mqttConnection = IOT_MQTT_CONNECTION_INITIALIZER;
    IotMqttNetworkInfo_t networkInfo = IOT_MQTT_NETWORK_INFO_INITIALIZER;
    IotMqttConnectInfo_t connectInfo = IOT_MQTT_CONNECT_INFO_INITIALIZER;
    IotMqttPublishInfo_t willInfo = IOT_MQTT_PUBLISH_INFO_INITIALIZER;
// Example network abstraction types.
    IotNetworkServerInfo_t serverInfo = AWS_IOT_NETWORK_SERVER_INFO_AFR_INITIALIZER;
    IotNetworkCredentials_t credentialInfo = AWS_IOT_NETWORK_CREDENTIALS_AFR_INITIALIZER;
    credentialInfo.pRootCa = tlsATS1_ROOT_CERTIFICATE_PEM;
    credentialInfo.rootCaSize = sizeof(tlsATS1_ROOT_CERTIFICATE_PEM);
// Example using a generic network implementation.
    networkInfo.createNetworkConnection = true;
    networkInfo.u.setup.pNetworkServerInfo = &serverInfo;
    networkInfo.u.setup.pNetworkCredentialInfo = &credentialInfo;
#ifdef BLE_ENABLED
    networkInfo.pNetworkInterface = IOT_NETWORK_INTERFACE_BLE;
#else
    networkInfo.pNetworkInterface = IOT_NETWORK_INTERFACE_AFR;
#endif
// Set the members of the connection info (password and username not used).
    connectInfo.cleanSession = true;
    connectInfo.keepAliveSeconds = 300;
    connectInfo.pClientIdentifier = clientcredentialIOT_THING_NAME;
    connectInfo.clientIdentifierLength = sizeof(clientcredentialIOT_THING_NAME);
// Set the members of the will info (retain and retry not used).
    willInfo.qos = IOT_MQTT_QOS_1;
    willInfo.pTopicName = "will/topic/name";
    willInfo.topicNameLength = 15;
    willInfo.pPayload = "MQTT client unexpectedly disconnected.";
    willInfo.payloadLength = 38;
// Set the pointer to the will info.
    connectInfo.pWillInfo = &willInfo;
// Call CONNECT with a 5 second block time. Should return
// IOT_MQTT_SUCCESS when successful.

    ESP_LOGI(MQTT_TAG, "Connecting...");
    IotMqttError_t result = IotMqtt_Connect( &networkInfo,
                                             &connectInfo,
                                             5000,
                                             &mqttConnection );
    if(result != IOT_MQTT_SUCCESS){
        ESP_LOGE(MQTT_TAG, "Connect failure. Error %d", result);
    }
    return result;
}
``` 

**Expected behavior**
Clear connection to amazon broker.

**Screenshots or console output**
```
I (2095) WIFI: SYSTEM_EVENT_STA_CONNECTED
2 303 [IP-task] vDHCPProcess: offer c0a86485ip
I (3365) event: sta ip: 192.168.100.133, mask: 255.255.254.0, gw: 192.168.100.1
I (3365) WIFI: SYSTEM_EVENT_STA_GOT_IP
3 306 [IP-task] vDHCPProcess: offer c0a86485ip
I (3375) [ESP32_13_08_2019][WiFi]: Connected to AP success
4 307 [main] [INFO ][MQTT][3070] MQTT library successfully initialized.
I (3385) [ESP32_13_08_2019][MQTT]: Connecting...
5 1313 [main] [ERROR][NET][13130] Failed to establish new connection.
6 1313 [main] [ERROR][MQTT][13130] Failed to establish new MQTT connection, error NETWORK ERROR.
E (13435) [ESP32_13_08_2019][MQTT]: Connect failure. Error 5
W (13445) [ESP32_13_08_2019][MQTT]: Connect failure

```
Steps to reproduce the behavior:

1. IotMqtt_Init()
2. Run Mqtt_Connect()

Thank you!

dcgaws, 2019-08-16:
Hello @izaton, can you please take a look at [this possibly-related issue](https://github.com/aws/amazon-freertos/issues/988)? In summary, I recommend that you start with the [latest version of our MQTT Hello World demo project](https://github.com/aws/amazon-freertos/tree/master/vendors/espressif/boards/esp32/aws_demos) as-is, ensure that it works, and then change the app task as required for your needs.

izaton, 2019-08-19:
> Hello @izaton, can you please take a look at [this possibly-related issue](https://github.com/aws/amazon-freertos/issues/988)? In summary, I recommend that you start with the [latest version of our [...]

Demo works fine. I'm trying to execute mqtt-esp32 connection based on the code from the demo and I keep getting that error. I also had a problem with wifi connection but the brave-heart Indian guy helped me figured it out (there was a problem with CmakeLists.txt file). Mqtt is just another step that I'm not able to perform.

dcgaws, 2019-08-19:
To confirm, the Hello World demo works on ESP32 just fine for you as-is? If so, can you please list the changes that you've made to the demo, step-by-step? After each change, please re-test your project and indicate which change has caused it to stop connecting.

gordonwang0, 2019-10-02:
Hello @izaton,

Please let us know if you need any more help with this.  Closing issue.

closed 2019-10-02 by gordonwang0