aws/amazon-freertos #1103: HTTP PUT example on esp32

chegewara, 2019-08-19:
Hi,
i am trying to run HTTP PUT example from feat/http_put_post branch on esp32. To configure pre-signed URL ia m using this blog and it works with GET:
https://medium.com/@aidan.hallett/securing-aws-s3-uploads-using-presigned-urls-aa821c13ae8d

Im having some issue with PUT, because i can use PUT if and only if body is empty, but i will find the cause later.
The problem with HTTP PUT example (or feat/http_put_post branch) is that credentials cant be used properly(Loading credentials into TLS context failed with error -8576.):
```
I (8859) PKCS11: Initializing NVS partition: "storage"
4 874 [iot_thread] ERROR: Loading credentials into TLS context failed with error -8576.
5 874 [iot_thread] [ERROR][NET][8740] Failed to establish new connection.
6 874 [iot_thread] [ERROR][HTTPS Client][8740] Failed to connect to the server at xxxxxx.s3.amazonaws.com on port 443 with error: 4
7 874 [iot_thread] [ERROR][HTTPS Client][8740] Error in IotHttpsClient_Connect(). Error code 203.
8 874 [iot_thread] [ERROR][DEMO][8740] Failed to connect to the S3 server, retrying after 3000 ms.
9 1209 [iot_thread] ERROR: Loading credentials into TLS context failed with error -8576.
10 1209 [iot_thread] [ERROR][NET][12090] Failed to establish new connection.
11 1209 [iot_thread] [ERROR][HTTPS Client][12090] Failed to connect to the server at xxxxxx.s3.amazonaws.com on port 443 with error: 4
12 1209 [iot_thread] [ERROR][HTTPS Client][12090] Error in IotHttpsClient_Connect(). Error code 203.
13 1209 [iot_thread] [ERROR][DEMO][12090] Failed to connect to the S3 server, retrying after 3000 ms.
14 1536 [iot_thread] ERROR: Loading credentials into TLS context failed with error -8576.
15 1537 [iot_thread] [ERROR][NET][15370] Failed to establish new connection.
16 1537 [iot_thread] [ERROR][HTTPS Client][15370] Failed to connect to the server at xxxxxx.s3.amazonaws.com on port 443 with error: 4
17 1537 [iot_thread] [ERROR][HTTPS Client][15370] Error in IotHttpsClient_Connect(). Error code 203.
18 1537 [iot_thread] [ERROR][DEMO][15370] Failed to connect to the S3 server. Error code: 203.
19 1537 [iot_thread] [ERROR][DEMO][15370] Error running demo.
20 1539 [iot_thread] [INFO ][INIT][15390] SDK cleanup done.
```

I tested shadow example from this branch with the same result:
```
I (8646) PKCS11: Initializing NVS partition: "storage"
7 852 [iot_thread] ERROR: Loading credentials into TLS context failed with error -8576.
8 852 [iot_thread] [ERROR][NET][8520] Failed to establish new connection.
9 853 [iot_thread] [ERROR][MQTT][8530] Failed to establish new MQTT connection, error NETWORK ERROR.
10 853 [iot_thread] [ERROR][DEMO][8530] MQTT CONNECT returned error NETWORK ERROR.
11 853 [iot_thread] [INFO ][Shadow][8530] Shadow library cleanup done.
12 853 [iot_thread] [INFO ][MQTT][8530] MQTT library cleanup done.
13 853 [iot_thread] [ERROR][DEMO][8530] Error running demo.
14 855 [iot_thread] [INFO ][INIT][8550] SDK cleanup done.
```
 
With master and/or release-1.5 branch my certificates work.

Any ideas what can be wrong? Thanks.

dcgaws, 2019-08-19:
Hi @chegewara that branch is still a work in progress. Our more stable HTTP code can be found in our https://github.com/aws/amazon-freertos/tree/release-candidate branch for the time being. However, that version doesn't have PUT yet, and the HTTP library as a whole should not be considered production-ready until it's been merged into master. Thank you for your patience.

dcgaws, 2019-08-19:
@chegewara update on the above, can you please clarify how you're testing that your client private key and certificate are configured correctly?

Second, the https://github.com/aws/amazon-freertos/tree/feat/http_put_post branch was updated on Friday; can you please retry with the latest, if you haven't? I recommend starting with the HTTP Hello World demo first on any branch; it's even simpler than the Shadow demo.

Thank you.

chegewara, 2019-08-19:
Hi @dcgaws thanks for response.
Its nothing very important, i just thought i will see how this API works and learn it since my client is interested in using it.

I cloned it just today, so i am on latest version. My certificates should be ok, because just to make sure its not certificate, i am using the one that is already registered and tested with master/release-1.5 branch.
I dont complain about it, i just thought i will report, but you said its better to test with release-candidate branch so i will do it. Thanks.

alexa-noxon, 2019-09-03:
Hi,

The error code you are getting is from mbedTLS's X.509 parsing:
`#define MBEDTLS_ERR_X509_INVALID_FORMAT                   -0x2180  /**< The CRT/CRL/CSR format is invalid, e.g. different type expected. */`

Can you verify that your certificate is formatted as expected by the Amazon FreeRTOS library?
This can be done by providing PEM files to: amazon-freertos/tools/certificate_configuration/CertificateConfigurator.html 

Thanks,
Alexa

chegewara, 2019-09-04:
Hi @alexa-noxon 
my certificates are completely fine, ive been using the same certificates with regular mqtt and shadow code, maybe the problem is that my certificates are in DER format and this demo requires PEM.

sarenameas, 2019-09-09:
Hello @chegewara,

I just wanted to check in. The HTTP PUT/POST has been merged to master, although unreleased. Please also keep in mind that master is bleeding edge code. 
22 days ago HTTP PUT/POST was on-top of to-be-released code which was a bit different, but since release 201809.00 that code has since been merged to master.
Could you try your certificates again on master with the HTTP demo and SHADOW or MQTT demos?
All of these demos use the same keys: keyCLIENT_CERTIFICATE_PEM and the keyCLIENT_PRIVATE_KEY_PEM.

chegewara, 2019-09-10:
Thanks for update, i will close this issue now and report back when i find time to test it with master and have problem with it.

closed 2019-09-10 by chegewara