aws/amazon-freertos #1104: What is the use of Bootloader_tests project?

shanmugav, 2019-08-19:
**Describe the bug**
We have two projects

https://github.com/aws/amazon-freertos/tree/master/projects/microchip/curiosity_pic32mzef/mplab/bootloader_tests

https://github.com/aws/amazon-freertos/tree/master/projects/microchip/curiosity_pic32mzef/mplab/bootloader

Looks like bootloader_tests is a duplicate, Is it needed and what is the use?

**System information**
PIC32MZ
- IDE used
- Operating System [Windows]
- Version of Amazon FreeRTOS  - 

201906.00_Major-515-g879158c2
--

- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
- OTA

**Expected behavior**
This bootloader_tests project is redundant and needs to be removed?

Thank you!

dcgaws, 2019-08-19:
Hello @shanmugav, I confirmed that the project in question builds successfully when loaded as a dependency to https://github.com/aws/amazon-freertos/tree/master/projects/microchip/curiosity_pic32mzef/mplab/aws_tests. However, I'm conferring with a colleague as to whether we need to continue to maintain that dependency as separate from the other bootloader project, as you say.

pvyawaha, 2019-08-19:
Hi Shanmugav,

The bootloader test project can be removed and is not required now. The aws_demos and aws_tests both can use the same bootloader project as dependency. I am creating a pull request with the fix.

pvyawaha, 2019-08-23:
Hi @shanmugav,

The fix is merged to master in pull request https://github.com/aws/amazon-freertos/pull/1138.