aws/amazon-freertos #1131: ESP32: linking fails due to both Nimble and Bluedroid being built

matteoscordino, 2019-08-22:
**Describe the bug**
I have a CMake project that used to build before commit 3949bfa. 
Since then, the NimBLE stack has been introduced in the build and now linking fails due to multiple definitions of functions that are present in both NimBLE and Bluedroid. It seems for some reason both stack are built, despite only NimBLE being enabled in sdkconfig.

The problem does not occur if I build the demos, only if I build my own project, which consumes amazon-freertos externally as described in issue #832 

**System information**
- Which hardware board or part numbers? ESP32
- IDE used: none, CMake CLI build
- Operating System: Linux
- Version of Amazon FreeRTOS: master branch since commit 3949bfa
- Project: Custom Application 

**Expected behavior**
Linking succeeds

**Screenshots or console output**
Build output:
https://gist.github.com/matteoscordino/b0fbc27184f08a17a35bb3afc4b7db21

**To reproduce**
Build with the project below

**Code to reproduce the bug**
This is the CMakeList.txt:
https://gist.github.com/matteoscordino/5a1b0f984305f480f25ec88f8e187c17

Full project structure tree:
https://gist.github.com/matteoscordino/3bc62c94d6c93cfbbc836275a6a947e5

huguesBouvier, 2019-08-23:
Hello,

I looked at the cmake file but I am not seeing where "set(afr_ports_dir "${CMAKE_CURRENT_LIST_DIR}/ports")" gets included. 
I think the issue stems from this:
vendors\espressif\boards\esp32\ports\ble\nimble\iot_ble_hal_common_gap.c
Has the following code:
#if CONFIG_NIMBLE_ENABLED == 1
    #include "nimble/iot_ble_hal_common_gap.c"
#else
    #include "bluedroid/iot_ble_hal_common_gap.c"
#endif

So you need to include the nimble/iot_ble_hal_common_gap.c" and bluedroid/iot_ble_hal_common_gap.c as include files, not source files.

matteoscordino, 2019-08-24:
Hello @huguesBouvier ,

I added `set(afr_ports_dir "${CMAKE_CURRENT_LIST_DIR}/amazon-freertos/ports")` at the beginning of my `CMakeLists.txt` but it didn't make a difference. I was under the impression that it would be brought in anyway by dependencies of the root amazon-freertos `CMakeLists.txt`?

Sorry, but I didn't understand the meaning of your comment about including the .c files as includes instead of sources. I am not directly referencing them in my own `CMakeLists.txt` either as source or include files, so I guess they are built due to amazon-freertos dependencies? I don't have control over that, do I?

By the way, user @kevinresol has uploaded a repo at [https://github.com/kevinresol/amazon-freertos-hello-world](url) that shows the same problem .
Their CMakeLists is very similar to mine, since they both originate from the discussion in issue #832

tgsong, 2019-08-29:
@matteoscordino Basically there are 2 problems,
1. We're doing `#include` for some C files in the BLE hal porting layer in ESP32, see this file for example, `vendors/espressif/boards/esp32/ports/ble/iot_ble_hal_common_gap.c`. This essentially requires us to tell compiler to treat those C files as header files and do not compile them as regular C source files. There's a way to do this in CMake, however, I just found out that this settings is only visible to CMake targets that are defined under same directory level and cannot be transitively passed to other dependents.
2. The BLE hal layer is not defined as static library, but as an "interface library" in CMake terms. In other words, the source files are compiled directly into your application code, which is defined outside of our directory. But according to the first problem, your application did not get our settings to treat those C files as header files, thus both nimble and bluedroid stack are getting compiled.

The solution is to not expose this settings to users by defining BLE hal as a static library, and fix the scope problem. I'll raise a PR to fix this.

dan4thewin, 2019-09-12:
Thanks, @matteoscordino , for reviewing and approving the PR.  I'll see that it gets merged.

closed 2019-09-12 by dan4thewin