aws/amazon-freertos #1157: How aws_demo.h can #include "iot_demo_runner.h" ?

SangDooLee, 2019-08-28:
Hi, First of all, this is not a bug. It is a question about amazon freertos code.

I'm just studying with amazon_freertos Demo

in main.c 

line 229   DEMO_RUNNER_Rundemos();  

this line execute 'mqtt_demo play' successfully. So I just tried to find how main.c call this function. 

main.c  includes 'aws_demo.h' in line 44.

yes, now it is aws_demo.h 

in line 56 void DEMO_RUNNER_RunDemos( void );

there is a declaration. 

this function located in 'iot_demo_runner.c'

here is my question. how aws_demo.h includes iot_demo_runner.c very well?

even if I erase line 26 , line 27 (#include "platform/iot_network.h", #include "platform/iot_ ethreads.h") , the function (DEMO_RUNNER_RunDemos()) link to iot_demo_runner.c very well

I would be very appreciate if you help me. thank  u.

alexa-noxon, 2019-08-28:
Hello,

In order to make the Amazon FreeRTOS demos configurable and provide a uniform demo entry point even when not all demos are supported by all platforms, demo function control is split between a few files.  Below is an overview of those files and some important items inside of them:

- As you've pointed out in your post, the starting point is `main.c`, which calls **DEMO_RUNNER_RunDemos**
- **DEMO_RUNNER_RunDemos** is implemented in `iot_demo_runner.c`, where the demo task is set by setting a function pointer to[ **DEMO_entryFUNCTION**](https://github.com/aws/amazon-freertos/blob/master/demos/demo_runner/iot_demo_runner.c#L85) and creating a task [**runDemoTask**](https://github.com/aws/amazon-freertos/blob/master/demos/demo_runner/iot_demo_runner.c#L90)

```
void DEMO_RUNNER_RunDemos( void )
{
    /* These demos are shared with the C SDK and perform their own initialization and cleanup. */
    static demoContext_t mqttDemoContext =
    {
        .networkTypes                = democonfigNETWORK_TYPES,
        .demoFunction                = DEMO_entryFUNCTION,
        .networkConnectedCallback    = DEMO_networkConnectedCallback,
        .networkDisconnectedCallback = DEMO_networkDisconnectedCallback
    };

    Iot_CreateDetachedThread( runDemoTask,
                              &mqttDemoContext,
                              democonfigDEMO_PRIORITY,
                              democonfigDEMO_STACKSIZE );
```
- **DEMO_entryFUNCTION** is defined in `iot_demo_runner.h`.
```
#if defined( CONFIG_MQTT_DEMO_ENABLED )
    #define DEMO_entryFUNCTION              RunMqttDemo
```
- The control for which **DEMO_entryFUNCTION** is used is located in `aws_demo_config.h`
```
#define CONFIG_MQTT_DEMO_ENABLED
```
- **RunMqttDemo** lives in `iot_demo_mqtt.c`
- The demoContext.demoFunction is invoked by [**runDemoTask**](https://github.com/aws/amazon-freertos/blob/master/demos/demo_runner/iot_demo_freertos.c#L309) in `iot_demo_freertos.c` 
```
        /* Run the demo. */
        status = pContext->demoFunction( true,
                                         clientcredentialIOT_THING_NAME,
                                         pConnectionParams,
                                         pCredentials,
                                         pNetworkInterface );
```

Does this answer your question about how we get from main.c to the demo code?

Alexa

SangDooLee, 2019-08-29:
Thank you for your detailed explanation!

I was confused for basic C skill

closed 2019-08-29 by SangDooLee