aws/amazon-freertos #1176: ESP32 can't find greengrass core

ahmedwahdan, 2019-08-31:
I'm trying to have my ESP32 connect to a greengrass core installed on an EC2 instance.
I always get this output:
> 13 2213 [iot_thread] GGD - Can't connect to greengrass Core
> 14 2213 [iot_thread] Auto-connect: Failed to retrieve Greengrass address and certificate.

System information

ESP32
Notepad++
Operating System : Windows
Version of Amazon FreeRTOS : 201908.00 release
Project : Greengrass

**Screenshots or console output**

> > I (2823) WIFI: SYSTEM_EVENT_STA_CONNECTED
> 3 882 [IP-task] vDHCPProcess: offer c0a80102ip
> I (9163) event: sta ip: 192.168.1.2, mask: 255.255.255.0, gw: 192.168.1.1
> I (9163) WIFI: SYSTEM_EVENT_STA_GOT_IP
> 4 888 [IP-task] vDHCPProcess: offer c0a80102ip
> 5 889 [iot_thread] [INFO ][DEMO][8890] Successfully initialized the demo. Network type for the demo: 1
> 6 889 [iot_thread] Attempting automated selection of Greengrass device
> 7 1682 [iot_thread] About to close socket.
> 8 1713 [iot_thread] Socket closed.
> 9 1713 [iot_thread] Stack high watermark for discovery helper task: 5296.
> 10 2213 [iot_thread] About to close socket.
> 11 2213 [iot_thread] Socket closed.
> 12 2213 [iot_thread] Stack high watermark for discovery helper task: 5056.
> 13 2213 [iot_thread] GGD - Can't connect to greengrass Core
> 14 2213 [iot_thread] Auto-connect: Failed to retrieve Greengrass address and certificate.
> 15 2213 [iot_thread] ----Demo finished----

Thank you!

alexa-noxon, 2019-09-03:
Hi Ahmed,

When you device connects to AWS for a list of Greengrass devices it can connect to, it might be receiving an IP address that is only locally accessible.  

If that's the case, then you would need to update the Greengrass device IP address to use a publicly available EC2 IP address.

- In the IoT Console, in the side bar, you can go to Greengrass > Groups, and then click on your Greengrass group name.  
- Next, in the sidebar, select Settings.  For "Core connectivity information", make sure that "Manually manage connection information" is chosen, then click on "View cores for specific endpoint information"
- Click on your core, and select "Connectivity" in the sidebar.  Here, you can edit the core endpoint to be a publicly accessible endpoint.

Does this take care of the issue?

Thanks,
Alexa

ahmedwahdan, 2019-09-03:
Hello Alexa,
Thanks for your response. 
The group end point is the same as EC2 instance.
The problem is the connection timeout.
Does the region is that important? I'm in Egypt and using US-West2 region.

danhorner, 2019-09-06:
Hello, 
Were you able to resolve this issue?  I don't think it's related to the initial discovery, otherwise you would see a message from [aws_greengrass_discovery.c](https://github.com/aws/amazon-freertos/blob/37c62009b4c9e696483ab7bd62e18193a97f8b77/libraries/freertos_plus/aws/greengrass/src/aws_greengrass_discovery.c#L454) such as:  "JSON request failed"

After the JSON is pulled and parsed, the discovery library will attempt to connect to all nonlocal hosts/ports returned in the json descriptor.  Here is a troubleshooting sequence if you have access to curl and openssl. 

1.  If you have access to the credential files you used to provision, you can query the descriptor directly: 
```
curl --cert  <THING_CERT.PEM> --key <THING_PRIVATE_KEY> https://<YOUR_ENDPOINT>/greengrass/discover/thing/<THING_NAME>
```

2. You can add a print statement into `GGD_GetIPandCertificateFromJSON` to log these attempts: 
` configPRINTF(("Try to connect to host %s on port %d\n", pxHostAddressData->pcHostAddress, pxHostAddressData->usPort));`

3. If that list looks funny, simplify using @alexa-noxon's instructions above. 

4. Can you verify connectivity to the GG core on EC2 from your local network using openssl or the mosquitto client? You will need the CA certificate returned in Step 1 above: 

```
openssl s_client -connect <GG_HOST:GG_PORT> -cert <THING_CERT.PEM> -key <THING_PRIVATE_KEY.PEM> -CAfile <CA_FILE_RETURNED_FROM_GG_DISCOVERY>
#or 
```

```
mosquitto_sub  -t "#" -h <GG_HOST> -p <GG_PORT> --cert <THING_CERT.PEM> --key <THING_PRIVATE_KEY.PEM> --cafile <CA_FILE_RETURNED_FROM_GG_DISCOVERY> -i <IOT_THING_NAME>
```

ahmedwahdan, 2019-09-15:
I think it's a timeout connection which was not clear from the ESP32 logs.
I also added the EC2 DNS to the greengrass connections.

danhorner, 2019-09-20:
@ahmedwahdan, does that mean that you're now able to connect?

ahmedwahdan, 2019-09-22:
Yes, we can close this issue

closed 2019-09-22 by ahmedwahdan