aws/amazon-freertos #1177: Cannot find FreeRTOS+{CLI,UDP,Trace,IO} modules

free-easy, 2019-09-01:
**Describe the bug**
Hi,
it seems that the FreeRTOS Plus modules (CLI, UPD, Trace, IO) are not present in this repository.
It there any reason for that?

free-easy, 2019-09-01:
[image]

Ok, it seems that the lack of some modules is due to licensing issues, but still, I wasn't able to find the CLI module.

0x3333, 2019-09-01:
Exactly what I'm looking for. I could not find it. Will search a little more.

htibosch, 2019-09-01:
Did you have a look on [freertos.org/plus](https://freertos.org/FreeRTOS-Plus/index.html).
Yes I think that you have, because there it shows the picture.
Some of the products of FreeRTOS have not ( yet ) been adopted in the AWS/FreeRTOS repository. But on the mentioned page on freertos.org, you will find the downloads that you are looking for,

free-easy, 2019-09-01:
The reason for which I prefer to use this repository is that it integrates better with our git workflow (i.e. having dependencies as submodules).
I'd like to avoid resorting to downloading a zip from freertos.org/plus or maintaining a git mirror of the SVN repository that is on SourceForge. I'm aware of some unofficial mirror of FreeRTOS (https://github.com/coldnew/FreeRTOS-mirror), but too bad that it's unofficial.

0x3333, 2019-09-01:
I found the sources here:

https://sourceforge.net/p/freertos/code/HEAD/tree/trunk/FreeRTOS-Plus/Source/

htibosch, 2019-09-02:
> I'm aware of some unofficial mirror of [FreeRTOS](https://github.com/coldnew/FreeRTOS-mirror), but too bad that it's unofficial.

I suppose that you have seen the new **official** [GIT repository of AWS/FreeRTOS](https://github.com/aws/amazon-freertos)? You will find the kernel, FreeRTOS+TCP and a lot more, like e.g. secure sockets/TLS. For the some other FreeRTOS products (CLI, UPD, Trace, IO, FreeRTOS+FAT), you will have to consult the [SVN on sourceforge](https://sourceforge.net/p/freertos/code/HEAD/tree/trunk/FreeRTOS-Plus/Source).

dan4thewin, 2019-09-10:
Hi, @free-easy and @0x3333 .  We value your feedback.  Although I don't have any timelines to offer, I'd like to assure you we are working to improve the way we distribute FreeRTOS and related software.  Thank you.

closed 2019-09-10 by dan4thewin