aws/amazon-freertos #1178: OTA without ThingName?

dongouyang, 2019-09-02:
I need to make OTA updates to multiple devices.   However, the examples are all building "Thing Name" as part of the binary file.   Is there a way to build for a group of things and make OTA updates to all of them?

RichardBarry, 2019-09-03:
While the demos use a string for the thing name you can use anything unique identifier you like.  For example you could append the serial number of the device, or a chip specific identifier, etc.  To do that you can have the thing name returned by a function, and implement the function to make use of a unique identifier.  Please re-open this post if you have further questions.

dongouyang, 2019-09-03:
I want to clarify what I am asking.   After setting "Thing Name" in confgure.json, I believe "thing name" is compiled as part of binary .bin file.  It does not work well with OTA because OTA is attempting to either match "thing name" with the existing esp32 device or override "thing name" on the device.    I am looking for ways to use the same .bin file for multiple esp32 devices.    How can I ask OTA to ignore the "thing name" in the .bin file?

alexa-noxon, 2019-09-03:
Hi,

Yes, if you are using the AWS Config Quick Start template to get started, the Thing Name is copied into **clientcredentialIOT_THING_NAME** located in `demos\include\aws_clientcredential.h` .

As you mentioned, this does get built into the binary, so in order to update multiple things over OTA, with one binary, this needs to be replaced with something that is thing agnostic, for example

`#define clientcredentialIOT_THING_NAME   getThingName()`

where getThingName() either pulls the thing name from one of the locations suggested by Richard above, or from a previously provisioned location in NVM. 

One more thing to keep in mind before doing multi-device OTA is that other credentials (device private key, device certificate) should similarly not be part of the binary, so make sure that `vDevModeKeyProvisioning` is not called in subsequent updates (that way devices will continue to use the previously provisioned credentials).

Does that answer the question?

dongouyang, 2019-09-04:
Thank you for the prompt answer.  If I have multiple iot devices share the same device private key and device certificate, that should be ok, right?

What I want to do is the following:  
1. flash the same code to multiple devices.  
2. The device can read something unique like the mac address as its "thing name".   
3. When the device comes online, it will send MQTT messages to tell the server about its name.  
4. Then I will build a central location on AWS to catalog them and tag their properties.    
5. The mac address "thing name" and corresponding tags will be stored on AWS RDS for future reference.

Is this system feasible?

danhorner, 2019-09-04:
Hi, 

Mac-based thing names will allow you to distinguish between devices, but the shared certificates are not secure. 

If you hard code the same private key and certificate in all your devices (for example using `vDevModeKeyProvisioning`), you will be giving up important security properties: 
 - Any device can masquerade as another
 - A compromised private key can be used to clone an unlimited number of devices
 - You will be unable to recover your network if the key in any one of your devices is compromised.

Best practice would be to deploy an individual certificate to each device during provisioning and store it non-volatile memory or a trusted store.

There's some more detailed discussion of the risks in [this forum post](https://forums.aws.amazon.com/thread.jspa?threadID=234102)

dongouyang, 2019-09-05:
Thank you.  Security is less of a concern.  My system consists of many devices in a controlled environment, all connecting to single wifi, then to AWS.   Is there a way to simply the setup and ota update process?

danhorner, 2019-09-09:
@dongouyang, 

If you want to deploy the same OTA image to all of your devices, you have the following options:

1) As you suggested, you could replace references to `clientcredentialIOT_THING_NAME` with calls to your own `getIOT_THING_NAME()` that returns a pointer to a static and device-unique thing name generated based on the mac address. You would use this together with a single hardcoded certificate shared among your devices.  *We discourage this for the reasons above*.

2) However, you might consider programming a unique key, certificate, and thing name to the device flash on a separate partition. You can write it to an encrypted flash partition during device setup using espressif's tools `espsecure.py` and `esptool.py --write-flash` [documented here](https://docs.espressif.com/projects/esp-idf/en/latest/security/flash-encryption.html). You can read it back during operation using the memory-mapped api: `spi_flash_mmap()` ([documentation](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/spi_flash.html#flash-partition-apis))

3) You can use Amazon FreeRTOS PKCS11 support to manage this information, together with a custom provisioning image that allows you to securely write these credentials before accepting an OTA of your application binary.

Unfortunately, we don't currently have a reference implementation of #2 or #3. Please feel free to reply or reopen this isssue if you have further questions.

closed 2019-09-09 by danhorner