aws/amazon-freertos #1191: Using OTA and MQTT together

wayne-david, 2019-09-05:
Iam trying to integrate esp-idf blink example with the aws-freertos OTA demo on esp32.I have build the aws-OTA demo and it works fine.How do I add my code to the OTA demo so that I have blink code running and also have OTA agent running on my esp32.

chegewara, 2019-09-07:
Because AWSFreeRTOS does not support shared connection you will need to create separate connections from OTA Job and mqtt and take care of connection cleaning for each of them in case disconnection is longer than keep alive.

wayne-david, 2019-09-09:
Thanks Chegewara ,my real issue is how do I edit the OTA code to add my custom code eg.blink in it.I tried formating the OTA code to be a task and then added the blink code compartible with esp-idf as another task.I then called the tasks with xTaskCreate.AWS have not given an example of how to integrate custom code with aws-freertos.If I could have any example of aws-freertos integrated with other esp-idf code,I will be good to go.

chegewara, 2019-09-09:
Dont touch OTA code, try to add your own code here:
https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/application_code/main.c#L132
or here (depends on your needs):
https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/application_code/main.c#L129

wayne-david, 2019-09-09:
@chegewara you are a savior,all along I wondered where app-main was.I was doing it wrong.Thanks alot.Do you have a link to the source of your information.Thanks so much

chegewara, 2019-09-09:
Yes, my brain.

wayne-david, 2019-09-09:
@chegewara alright,if am allowed to ask,was the code in the main essenstial or I can create my own clean main file.

chegewara, 2019-09-09:
I think you can create new main file, but to me it is much easier to start with existing file and delete code i dont need or add my own code.

wayne-david, 2019-09-09:
Thanks,I appreciate alot

closed 2019-09-09 by wayne-david

sameerdasarwad, 2020-01-10:
> Because AWSFreeRTOS does not support shared connection you will need to create separate connections from OTA Job and mqtt and take care of connection cleaning for each of them in case disconnection is longer than keep alive.

@chegewara  as  i you say if MQTT and OTA cannot run same time what if i have sensor which update shadow for every second and i also want keep running the OTA job for check new updates ...

gordonwang0, 2020-01-23:
Hi,

It is now possible to share the MQTT connection used with OTA.  The MQTT connection is given to the OTA agent as `pvControlClient` in the `OTA_ConnectionContext_t`.  See this [line in the demo](https://github.com/aws/amazon-freertos/blob/557307c44ed828cd598692cc82c53869a86973b3/demos/ota/aws_iot_ota_update_demo.c#L251).  The application may use this connection for other purposes too.

sameerdasarwad, 2020-01-25:
so it is possible to share connection and run both ota and shadow together

pedroramireslima, 2020-09-12:
@sameerdasarwad were you able to isolate the mqtt connection from the aws ota example?

pedroramireslima, 2020-09-12:
> Yes, my brain and 2 commercial projects (closed source).

@chegewara do you have any code example using the ota example from aws and mqtt connection in other topics?