aws/amazon-freertos #1200: Guru Mediation Error In xTaskIncrementTick

coxkade, 2019-09-09:
**Describe the bug**
On startup a guru mediation error occurs in xTaskIncrementTick.

**System information**
- Hardware
    - ESP32
- IDE used
   - None
- Operating System [Windows|Linux|MacOS]
   - MacOS
- Version of Amazon FreeRTOS
    - 201908.00-2-g8c459865e of this [fork](https://github.com/coxkade/amazon-freertos/tree/kade_fix) (has a fix for the BT multiple definition issue)
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
    - [Custom Application](https://github.com/coxkade/aws_debug)

**Expected behavior**
I expect the application to launch without crashing.  Note that this project is to set up an external project that uses freertos.  To keep things as simple as possible.  The main is a copy of the demo app_main but with the demo launch commented out.

**Screenshots or console output**
```
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6300
load:0x40078000,len:11276
ho 0 tail 12 room 4
load:0x40080000,len:6084
0x40080000: _WindowOverflow4 at /Users/kade.cox/workspace/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

entry 0x4008032c
0x4008032c: _KernelExceptionVector at ??:?

I (31) boot: ESP-IDF v3.1.3-25-g36733db8d 2nd stage bootloader
I (31) boot: compile time 07:32:39
I (40) boot: Enabling RNG early entropy source...
I (40) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (67) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (75) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (82) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (90) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (97) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (105) boot: End of partition table
I (109) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x29700 (169728) map
I (177) esp_image: segment 1: paddr=0x00049728 vaddr=0x3ffc0000 size=0x02da8 ( 11688) load
I (182) esp_image: segment 2: paddr=0x0004c4d8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/kade.cox/workspace/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

I (184) esp_image: segment 3: paddr=0x0004c8e0 vaddr=0x40080400 size=0x03730 ( 14128) load
I (199) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xc1ea0 (794272) map
0x400d0018: _flash_cache_start at ??:?

I (480) esp_image: segment 5: paddr=0x00111ec0 vaddr=0x40083b30 size=0x10834 ( 67636) load
0x40083b30: pll_correct_dcap at ??:?

I (520) boot: Loaded app from partition at offset 0x20000
I (520) boot: ota rollback check done
I (520) boot: Disabling RNG early entropy source...
I (525) cpu_start: Pro cpu up.
I (528) cpu_start: Single core mode
I (533) heap_init: Initializing. RAM available for dynamic allocation:
I (540) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (546) heap_init: At 3FFD1C08 len 0000E3F8 (56 KiB): DRAM
I (552) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (558) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (565) heap_init: At 40094364 len 0000BC9C (47 KiB): IRAM
I (571) cpu_start: Pro cpu start user code
I (253) cpu_start: Starting scheduler on PRO CPU.
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400d3be0  PS      : 0x00060034  A0      : 0x400900ac  A1      : 0x3ffc05d0  
0x400d3be0: xTaskIncrementTick at /Users/kade.cox/workspace/aws_debug/amazon-freertos/freertos_kernel/tasks.c:5068

0x400900ac: _frxt_timer_int at /Users/kade.cox/workspace/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/portasm.S:313

A2      : 0x00000000  A3      : 0x00050021  A4      : 0x40081565  A5      : 0x00000000  
0x40081565: _xt_lowint1 at /Users/kade.cox/workspace/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1098

A6      : 0x3ffc5268  A7      : 0x00000001  A8      : 0x8008ff40  A9      : 0x3ffd43b0  
A10     : 0x00000000  A11     : 0x3ffd43e0  A12     : 0x0000003e  A13     : 0x00000000  
A14     : 0x3ffd4c9c  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x40093c6d  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d3be0
0x40093c6d: esp_rom_spiflash_read_data at /Users/kade.cox/workspace/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_rom_patch.c:412

0x400d3be0: xTaskIncrementTick at /Users/kade.cox/workspace/aws_debug/amazon-freertos/freertos_kernel/tasks.c:5068

Backtrace: 0x400d3be0:0x3ffc05d0 0x400900a9:0x3ffc05f0 0x400815ba:0x3ffc0600 0x40093c6a:0x00000000
0x400d3be0: xTaskIncrementTick at /Users/kade.cox/workspace/aws_debug/amazon-freertos/freertos_kernel/tasks.c:5068

0x400900a9: _frxt_timer_int at /Users/kade.cox/workspace/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/portasm.S:303

0x400815ba: _xt_lowint1 at /Users/kade.cox/workspace/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1105

0x40093c6a: esp_rom_spiflash_read_data at /Users/kade.cox/workspace/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_rom_patch.c:412

Rebooting...
```

**To reproduce**
Steps to reproduce the behavior:

1. Clone [this](https://github.com/coxkade/aws_debug) project
    * ```git clone git@github.com:coxkade/aws_debug.git```
2. Update the submodules to get amazon freertos
    * ``` cd aws_debug && git submodule update --init --recursive```
3. Build the project:
    * ``` make config && make ```
4. Flash the build
    * ``` make flash ```
5. Monitor the build
   * ``` make monitor ```

**Code to reproduce the bug**

[This](https://github.com/coxkade/aws_debug) project.

**Additional context**
As mentioned above this project is just pulling the source out of the demo launcher from Amazon Freertos source.  The only changes made to the Amazon freertos is to make the BT not build multiple sources for the same API.

It looks similar to some other issues.  However seems useful as I believe I have encapsulated the problem well.

dan4thewin, 2019-09-10:
Hello, @coxkade .  Thank you for the report.  The project is based on one of the demos, yes?  Does the original demo exhibit the same error?

coxkade, 2019-09-11:
No the Demo Does not.  This project is essentially building the demo out of source.  I created this project just as a quick demonstration of some of the problems I have been dealing with on a production project.  Note that the production project will not just build the demos source.  The goal was to quickly create something that demonstrated the problem.

dan4thewin, 2019-09-13:
Hi, @coxkade .  Thank you for the code to reproduce the issue.  Using it, I also see the panic.  I'll try to get to the bottom of this.

dan4thewin, 2019-09-18:
Hi, @coxkade .  I've been comparing the build results of your project with that of the demo at the same commit as your submodule.  Using VERBOSE=1 to show the commands make is running, I see that the top level include directory is not present in the compiler arguments for amazon-freertos objects.  Also quite disconcerting, the linker commands have different sets of objects files, and the binaries differ by ~240 KiB.

I think a hand-crafted makefile could readily circumvent these problems, but it seems clear that a cmake add_subdirectory line is not as useful as one would hope.

coxkade, 2019-09-18:
Ok thanks for looking into that.  That is good to know that the include is not being handled by the Amazon Components. I have updated the project such that the project include paths are present when building the lower level components.  For example building port.c

```
[ 42%] Building C object amazon-freertos/esp-idf/freertos/CMakeFiles/idf_component_freertos.dir/__/__/__/__/__/__/__/__/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c.obj
cd /Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/build/amazon-freertos/esp-idf/freertos && /Users/kade.cox/workspace/AWS_Wrapper_Project/AWS_Compilers/mac/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.1.3-25-g36733db8d\" -DPROJECT_NAME=\"esp-idf\" -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/demos/network_manager -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/freertos/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/config_files -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/freertos_kernel/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/build/config -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/esp32/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/driver/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/tcpip_adapter/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/source/portable/Compiler/GCC -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/newlib/platform_include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/newlib/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/vfs/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/heap/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/log/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/soc/esp32/include -I/Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/soc/include  -mlongcalls -g   -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -o CMakeFiles/idf_component_freertos.dir/__/__/__/__/__/__/__/__/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c.obj   -c /Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c
```

However even with this update.  The error persists as:

```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400d2c74  PS      : 0x00060034  A0      : 0x400900a4  A1      : 0x3ffc05d0  
0x400d2c74: xTaskIncrementTick at /Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/freertos_kernel/tasks.c:5068

0x400900a4: _frxt_timer_int at /Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/portasm.S:313

A2      : 0x00000000  A3      : 0x00050021  A4      : 0x4008155d  A5      : 0x00000000  
0x4008155d: _xt_lowint1 at /Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1098

A6      : 0xf0000040  A7      : 0x00000770  A8      : 0x8008ff38  A9      : 0x3ffd2290  
A10     : 0x00000000  A11     : 0x3ffd22c0  A12     : 0x00000001  A13     : 0x3ffbcdb8  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000018  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x40093c5b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d2c74
0x40093c5b: esp_rom_spiflash_read_data at /Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_rom_patch.c:412

0x400d2c74: xTaskIncrementTick at /Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/freertos_kernel/tasks.c:5068

Backtrace: 0x400d2c74:0x3ffc05d0 0x400900a1:0x3ffc05f0 0x400815b2:0x3ffc0600 0x40093c58:0x00000000
[... cut ...]
0x400900a1: _frxt_timer_int at /Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/portasm.S:303

0x400815b2: _xt_lowint1 at /Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1105

0x40093c58: esp_rom_spiflash_read_data at /Users/kade.cox/workspace/AWS_Wrapper_Project/aws_debug/amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/spi_flash_rom_patch.c:412

Rebooting...
```

dan4thewin, 2019-09-18:
@coxkade , the linker step is the culprit.  I used the linker command from the demo, supplying the missing objects, and tweaking the paths.  The new program worked as desired, with no panic.
```
O=~/work/amazon-freertos/build
~/bin/xtensa-esp32-elf-g++ \
-mlongcalls \
-nostdlib \
-Wl,--gc-sections \
-Wl,--cref \
-Wl,--Map=output.map \
-Wl,--undefined=uxTopUsedPriority \
CMakeFiles/kade_app.dir/main.c.obj \
$O/CMakeFiles/aws_demos.dir/demos/wifi_provisioning/aws_wifi_connect_task.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/libraries/freertos_plus/standard/utils/src/iot_system_init.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/libraries/freertos_plus/standard/utils/src/iot_pki_utils.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/mbedtls/port/esp_bignum.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/mbedtls/port/esp_mem.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/mbedtls/port/esp_sha256.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/mbedtls/port/esp_hardware.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/mbedtls/port/esp_sha1.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/mbedtls/port/esp_sha512.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/mbedtls/port/mbedtls_debug.c.obj \
$O/CMakeFiles/aws_demos.dir/demos/ble/aws_ble_gatt_server_demo.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/demos/demo_runner/aws_demo_network_addr.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/demos/demo_runner/aws_demo_version.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/demos/demo_runner/aws_demo.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/demos/network_manager/aws_iot_network_manager.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/demos/network_manager/aws_iot_demo_network.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/demos/demo_runner/iot_demo_freertos.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/demos/demo_runner/iot_demo_runner.c.obj \
$O/CMakeFiles/aws_demos.dir/demos/ble/iot_ble_numericComparison.c.obj \
$O/CMakeFiles/aws_demos.dir/demos/defender/aws_iot_demo_defender.c.obj \
$O/CMakeFiles/aws_demos.dir/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c.obj \
CMakeFiles/kade_app.dir/amazon-freertos/demos/mqtt/iot_demo_mqtt.c.obj \
$O/CMakeFiles/aws_demos.dir/demos/ota/aws_iot_ota_update_demo.c.obj \
$O/CMakeFiles/aws_demos.dir/demos/shadow/aws_iot_demo_shadow.c.obj \
$O/CMakeFiles/aws_demos.dir/demos/tcp/aws_tcp_echo_client_single_task.c.obj \
-o kade_app \
amazon-freertos/afr_wifi.a \
amazon-freertos/afr_ble.a \
-Wl,--gc-sections \
-Wl,--start-group \
amazon-freertos/esp-idf/soc/libsoc.a \
amazon-freertos/esp-idf/log/liblog.a \
amazon-freertos/esp-idf/heap/libheap.a \
amazon-freertos/esp-idf/freertos/libfreertos.a \
amazon-freertos/esp-idf/vfs/libvfs.a \
amazon-freertos/esp-idf/newlib/libnewlib.a \
amazon-freertos/esp-idf/driver/libdriver.a \
amazon-freertos/esp-idf/tcpip_adapter/libtcpip_adapter.a \
amazon-freertos/esp-idf/xtensa-debug-module/libxtensa-debug-module.a \
amazon-freertos/esp-idf/app_trace/libapp_trace.a \
amazon-freertos/esp-idf/app_update/libapp_update.a \
amazon-freertos/esp-idf/spi_flash/libspi_flash.a \
amazon-freertos/esp-idf/micro-ecc/libmicro-ecc.a \
amazon-freertos/esp-idf/bootloader_support/libbootloader_support.a \
amazon-freertos/esp-idf/ethernet/libethernet.a \
amazon-freertos/esp-idf/nvs_flash/libnvs_flash.a \
amazon-freertos/esp-idf/pthread/libpthread.a \
amazon-freertos/esp-idf/smartconfig_ack/libsmartconfig_ack.a \
amazon-freertos/esp-idf/wpa_supplicant/libwpa_supplicant.a \
amazon-freertos/esp-idf/esp32/libesp32.a \
amazon-freertos/esp-idf/cxx/libcxx.a \
amazon-freertos/esp-idf/bt/libbt.a \
amazon-freertos/esp-idf/console/libconsole.a \
amazon-freertos/esp-idf/esp_adc_cal/libesp_adc_cal.a \
amazon-freertos/esp-idf/expat/libexpat.a \
amazon-freertos/esp-idf/wear_levelling/libwear_levelling.a \
amazon-freertos/esp-idf/sdmmc/libsdmmc.a \
amazon-freertos/esp-idf/fatfs/libfatfs.a \
amazon-freertos/esp-idf/nghttp/libnghttp.a \
[... 45 lines cut ...]
amazon-freertos/afr_ble_wifi_provisioning.a \
amazon-freertos/afr_wifi.a \
amazon-freertos/afr_ble.a \
amazon-freertos/afr_ble_hal.a \
amazon-freertos/afr_ble.a \
amazon-freertos/afr_ble_hal.a \
amazon-freertos/afr_serializer.a \
amazon-freertos/afr_common.a \
amazon-freertos/afr_platform.a \
amazon-freertos/afr_common.a \
amazon-freertos/afr_platform.a \
amazon-freertos/afr_secure_sockets.a \
amazon-freertos/afr_pkcs11.a \
amazon-freertos/afr_tls.a \
amazon-freertos/afr_pkcs11_implementation.a \
amazon-freertos/afr_secure_sockets.a \
amazon-freertos/afr_pkcs11.a \
amazon-freertos/afr_tls.a \
amazon-freertos/afr_pkcs11_implementation.a \
amazon-freertos/afr_freertos_plus_tcp.a \
amazon-freertos/afr_crypto.a \
amazon-freertos/libraries/3rdparty/libafr_3rdparty_mbedtls.a \
amazon-freertos/afr_kernel.a \
amazon-freertos/libraries/3rdparty/libafr_3rdparty_tinycbor.a
```

coxkade, 2019-09-18:
@dan4thewin Sweet.  Thanks for the heads up.  I will look at pulling this into the project look into creating a fix.

coxkade, 2019-09-19:
Ok I think that I have it.  I had to add a way to add the source at the main esp cmake file.  In addition I had to make it so I could override the demo config file.  The changes I made to the amazon freertos can be found on [this](https://github.com/coxkade/amazon-freertos/tree/esp_override) branch.  A working example project using that branch can be found [here](https://github.com/coxkade/aws_debug).  @dan4thewin I am not sure if this is the best way to handle the out of order build going forward.  Can you create an issue for the Out of source build for the esp32 not functioning?  You can reference my demonstration project if you would like.  Otherwise I will go ahead and close this issue.

closed 2019-09-19 by coxkade