aws/amazon-freertos #1249: Support Cortex-v8M chip without TrustZone

cyliangtw, 2019-09-20:
For Cortex-v8M chip **with TrustZone**, it's fine to utilize amazon-freertos/freertos_kernel/portable/GCC/ARM_CM23/secure & non_secure folder.

For Cortex-v8M chip **without TrustZone**, will create amazon-freertos/freertos_kernel/portable/RVDS/ARM_CM23/port.c for this kind chip ?

aggarg, 2019-09-20:
>> For Cortex-v8M chip with TrustZone, it's fine to utilize amazon-freertos/freertos_kernel/portable/GCC/ARM_CM23/secure & non_secure folder.

Assuming that the MCU is Cortex-M23, yes you can use the above mentioned files:
* amazon-freertos/freertos_kernel/portable/GCC/ARM_CM23/secure - These are the files to be compiled in secure project. 
* amazon-freertos/freertos_kernel/portable/GCC/ARM_CM23/non_secure - These are the files to be compiled in non-secure project. 

If you can have Cortex-M33, you can use the respective files in amazon-freertos/freertos_kernel/portable/GCC/ARM_CM33/. 

>> For Cortex-v8M chip without TrustZone, will create amazon-freertos/freertos_kernel/portable/RVDS/ARM_CM23/port.c for this kind chip ?

If you have a chip without TrustZone, you can use the files in amazon-freertos/freertos_kernel/portable/GCC/ARM_CM23_NTZ or amazon-freertos/freertos_kernel/portable/GCC/ARM_CM33_NTZ. You need to define the following in your FreeRTOSConfig.h:

```
#define configENABLE_TRUSTZONE 0
#define configENABLE_MPU 1 /* Define as 0 if you do not want to use MPU. */
#define configENABLE_FPU 1 /* Define as 0 if you do not want to use FPU. */
```

Let us know if you face any issue and we are more than happy to help.

Thanks.

cyliangtw, 2019-09-23:
@aggarg , Thanks a lot, it's clear to control by `FreeRTOSConfig.h` and I will utilize xxx_NTZ.

closed 2019-09-23 by cyliangtw