aws/amazon-freertos #1256: Remove hardcoded defines for clientcredentialIOT_THING_NAME and other configuration items

tjirsch, 2019-09-21:
OS Code depends on preprocessor defines for clientcredentialIOT_THING_NAME and other definitons in aws_clientcredential.h and aws_clientcredential_keys.h.
The files are located in the demos folder and included in the os code in many files independent from the demo code.
For example the aws_greengrass_discovery.c file uses the defines directly in the code, not as parameters.

This makes the client thingname, broker endpoint etc. hardcoded and results in the need to compile a version of the device firmware for every device.

I would like to be able to pass most of the configuration info defined within the header files as parameters or have one function to configure them in a variable block. This would make it possible to create one firmware that dynamically creates a device name and connects to runtime configurable endpoints.   

[fgrep_clientcredential_usage_in_c_files.txt](https://github.com/aws/amazon-freertos/files/3638890/fgrep_clientcredential_usage_in_c_files.txt)

aggarg, 2019-09-23:
Thank you for bringing this to our notice. We will fix this. 

Thanks.

matteoscordino, 2019-09-28:
Agree with @tjirsch, I have implemented my own provisioning code to replace dev_mode_provisioning (which has the same kind of build-time dependency), but still `clientcredentialIOT_THING_NAME` is an issue (and it goes hand-in-hand with device key and certificate)

chegewara, 2019-09-28:
Thats true, but it is not very hard to fix. I am replacing:
`#define clientcredentialIOT_THING_NAME`
with
`char clientcredentialIOT_THING_NAME[64]` or `extern char clientcredentialIOT_THING_NAME[64]`
in aws_clientcredential.h.
This way i can set thing name in any place of my app i want.

tjirsch, 2019-09-28:
The aws_clientcredential.h and other files are included (from the demo folder !) also in many other files (mbedtls.c etc.) if you compile freeRTOS in a separate folder. You have to make really sure that those includes stay in sync. Put a #warning directive in the .h files to also see all indirect includes. 
To do this I am changing code within the freeRTOS folder and this has to be redone every time freeRTOS gets updated. Actually I replace the .h files in the demo folders with versions that just include my ones, but thats kind of arkward. 
I would like to be in a situation where I can just clone freeRTOS in a folder separate from my app code an do the configs in a file I keep in my repo. 
Same holds for the sdkconfig(.defaults) file, here the nessecary changes have been done in the meantime.

Of course it is easy to replace the constant with a variable, that was how it was done in earlier releases, but I don't get why this should have to be done every time I update to a newer release.

matteoscordino, 2019-09-28:
@tjirsch: getting to a state where Amazon Freertos can be consumed as an out-of-tree component is a common request, see issue #832 
Things have started moving that way, but the repo is not there just yet. I feel your pain =)

qiutongs, 2019-10-16:
https://github.com/aws/amazon-freertos/pull/1419

ravibhagavandas, 2019-11-04:
PR #1419 fixes the hard coded defines for clientcredentialIOT_THING_NAME in greengrass library. PR has been merged.

I am closing this issue. Please reopen this issue if you have any other queries.

closed 2019-11-04 by ravibhagavandas