aws/amazon-freertos #1257: Create multiple MQTT TLS error

ahmedwahdan, 2019-09-22:
**Describe the bug**
I'm trying to have another MQTT client to the same thing but I got TLS error.
I duplicated the _establishMqttConnection function call with another mqttConnection2  handler,
the second call always fails with TLS error
`ERROR: Handshake failed with error code -16`

`
status = _establishMqttConnection( awsIotMqttMode,
                                           NULL,
                                           pNetworkServerInfo,
                                           pNetworkCredentialInfo,
                                           pNetworkInterface,
                                           &mqttConnection );
         if( status == EXIT_SUCCESS )
         {
             IotLogInfo("Sucess 1 \r\n");
                 status = _establishMqttConnection( awsIotMqttMode,
                                           NULL,
                                           pNetworkServerInfo,
                                           pNetworkCredentialInfo,
                                           pNetworkInterface,
                                           &mqttConnection2 );
 
            if( status == EXIT_SUCCESS )
            {
                IotLogInfo("Sucess 2 \r\n");                               
            } 
            else
            {
                    IotLogInfo("Fail 2 \r\n");
            }
         }
         else
         {
             IotLogInfo("Fail 1 \r\n");
         }
`
Thank you!

aggarg, 2019-09-22:
Which hardware platform are you running on? TLS handshake requires a significant amount of RAM and there is high chance that you are running out of RAM (the assumption here is that you are running on a platform which does TLS in software). 

You can setup the Malloc Failed Hook function to determine if a call to ```pvPortMalloc``` is failing: https://www.freertos.org/a00016.html

I want to understand why do you want to create a separate MQTT client and not use the same MQTT client created previously? 

Thanks.

ahmedwahdan, 2019-09-22:
Hi,
Thanks for your reply. 
I'm working on ESP32. 
I'm trying to have a client connected to greengrass core and the second client will be used for the OTA agent through aws IOT core.

aggarg, 2019-09-22:
* Can you check if you are getting malloc failures?
* Can you try to make sure to start the second handshake after the first connection is established so that two TLS handshakes are not done in parallel. 

Thanks.

sarenameas, 2019-10-16:
Hello, 
Thsi in addition to comments from my colleague @aggarg .

There is indeed not enough heap space in your application to support two MQTT connections. -16 is 0x0010 and a search of this reveals the error code MBEDTLS_ERR_MPI_ALLOC_FAILED in mbedTLS. The heap in ESP32 is the space leftover in the dram after all static memory (.bss). You can increase this looking for places where you can reduce static memory in your application and temporarily allocate on the heap.

I saw you had a different issue where you were able to merge get OTA information through your greengrass core: https://github.com/aws/amazon-freertos/issues/1316. So is this issue still a problem for you?

ahmedwahdan, 2019-10-16:
We can close this issue for now.

closed 2019-10-16 by ahmedwahdan