aws/amazon-freertos #1272: OTA Demo - ESP32: Fails after initial boot into upgraded firmware

mnelson920, 2019-09-25:
I am trying to run the OTA Demo on espressif esp32-devkit, I was able to download the firmware and do the initial boot. But after the initial boot it fails and reboots back to org firmware. 

Here is the output of the uC:

`7 303 [iot_thread] OTA demo version 0.9.2
8 303 [iot_thread] Creating MQTT Client...
9 854 [iot_thread] Connecting to broker...
10 854 [iot_thread] [INFO ][MQTT][8540] Establishing new MQTT connection.
11 855 [iot_thread] [INFO ][MQTT][8550] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
12 855 [iot_thread] [INFO ][MQTT][8550] (MQTT connection 0x3ffe9418, CONNECT operation 0x3fff8eb0) Waiting for operation completion.
13 865 [iot_thread] [INFO ][MQTT][8650] (MQTT connection 0x3ffe9418, CONNECT operation 0x3fff8eb0) Wait complete with result SUCCESS.
14 865 [iot_thread] [INFO ][MQTT][8650] New MQTT connection 0x3ffc0af4 established.
15 865 [iot_thread] Connected to broker.
16 865 [OTA Task] [INFO ][MQTT][8650] (MQTT connection 0x3ffe9418) SUBSCRIBE operation scheduled.
17 865 [OTA Task] [INFO ][MQTT][8650] (MQTT connection 0x3ffe9418, SUBSCRIBE operation 0x3fffa948) Waiting for operation completion.
18 871 [OTA Task] [INFO ][MQTT][8710] (MQTT connection 0x3ffe9418, SUBSCRIBE operation 0x3fffa948) Wait complete with result SUCCESS.
19 871 [OTA Task] [prvSubscribeToJobNotificationTopics] OK: $aws/things/ESP32_AWS_FreeRTOS_Test/jobs/$next/get/accepted
20 872 [OTA Task] [INFO ][MQTT][8720] (MQTT connection 0x3ffe9418) SUBSCRIBE operation scheduled.
21 872 [OTA Task] [INFO ][MQTT][8720] (MQTT connection 0x3ffe9418, SUBSCRIBE operation 0x3fff8f74) Waiting for operation completion.
22 880 [OTA Task] [INFO ][MQTT][8800] (MQTT connection 0x3ffe9418, SUBSCRIBE operation 0x3fff8f74) Wait complete with result SUCCESS.
I (8945) ota_pal: prvPAL_GetPlatformImageState
I (8955) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (8955) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
I (8965) ota_pal: prvPAL_GetPlatformImageState
I (8975) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (8975) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
23 880 [OTA Task] [prvSubscribeToJobNotificationTopics] OK: $aws/things/ESP32_AWS_FreeRTOS_Test/jobs/notify-next
24 884 [OTA Task] [OTA_CheckForUpdate] Request #0
25 885 [OTA Task] [INFO ][MQTT][8850] (MQTT connection 0x3ffe9418) MQTT PUBLISH operation queued.
26 885 [OTA Task] [INFO ][MQTT][8850] (MQTT connection 0x3ffe9418, PUBLISH operation 0x3fff7544) Waiting for operation completion.
27 893 [OTA Task] [INFO ][MQTT][8920] (MQTT connection 0x3ffe9418, PUBLISH operation 0x3fff7544) Wait complete with result SUCCESS.
I (9075) ota_pal: prvPAL_SetPlatformImageState, 3
W (9075) ota_pal: Set image as invalid!
I (9085) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (9085) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
W (9095) ota_pal: Image not in self test mode 2
I (9095) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (9105) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
28 893 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ clientToken: 0:ESP32_AWS_FreeRTOS_Test ]
29 893 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ jobId: AFR_OTA-ota_test-4 ]
30 893 [OTA Task] [prvParseJSONbyModel] Identified parameter [ self_test ]
31 893 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ updatedBy: 589826 ]
32 893 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ streamname: AFR_OTA-a888c8b6-7c00-4e11-a4d6-0672d263ad61 ]
33 893 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ filepath: / ]
34 893 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ filesize: 1240960 ]
35 893 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ fileid: 0 ]
36 893 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ certfile: aws_ota_codesigner_certificate.h ]
37 893 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ sig-sha256-ecdsa: MEQCIEPXYrbcVcvNcBFnqNzh4Ruu5JkT... ]
38 893 [OTA Task] [prvParseJobDoc] In self test mode.
39 893 [OTA Task] [prvParseJobDoc] Failing job. We rebooted and the version is still the same.
40 894 [iot_thread] [OTA_AgentInit_internal] Ready.
41 898 [OTA Task] [prvUpdateJobStatus] Msg: {"status":"FAILED","statusDetails":{"reason":"rejected: 0x23000000"}}
42 898 [OTA Task] [INFO ][MQTT][8980] (MQTT connection 0x3ffe9418) MQTT PUBLISH operation queued.
43 898 [OTA Task] [INFO ][MQTT][8980] (MQTT connection 0x3ffe9418, PUBLISH operation 0x3fff6f10) Waiting for operation completion.
44 910 [OTA Task] [INFO ][MQTT][9100] (MQTT connection 0x3ffe9418, PUBLISH operation 0x3fff6f10) Wait complete with result SUCCESS.
45 910 [OTA Task] [prvUpdateJobStatus] 'FAILED' to $aws/things/ESP32_AWS_FreeRTOS_Test/jobs/AFR_OTA-ota_test-4/update
46 910 [OTA Task] [prvResetDevice] Attempting forced reset of device...`

Just trying to understand if I am doing something wrong.

Thanks

chegewara, 2019-09-25:
```
[OTA Task] [prvParseJobDoc] Failing job. We rebooted and the version is still the same
```
You have to increase version in this file:
https://github.com/aws/amazon-freertos/blob/master/demos/include/aws_application_version.h

mnelson920, 2019-09-25:
Great, thanks.

closed 2019-09-27 by pvyawaha