aws/amazon-freertos #1281: demo runner: demos try to use BLE as MQTT transport even if IOT_BLE_ENABLE_MQTT == 0

matteoscordino, 2019-09-26:
**Describe the bug**
It seems that if the define `BLE_ENABLED` is true, demos will try to use a BLE connection as an MQTT proxy regardless of whether `IOT_BLE_ENABLE_MQTT` is enabled.
In other words, there is the assumption that a BLE connection will always have on the other end an app that uses the AWS Mobile SDK to provide an MQTT proxy.
This is not true if we want to run the BLE Wifi Provisioning demo without the MQTT proxy.
It is also a problem if in your device you only want to use BLE for WiFI provisioning, and then have NetworkManager handle the wifi connection (which is my use case)
Of course, I might have misunderstood the meaning of the defines in question, but in my use case this seems to be an issue.

**System information**
- Which hardware board or part numbers? ESP32
- IDE used: CLion, CMake
- Operating System Linux host
- Version of Amazon FreeRTOS: master
- Project: Custom Application, MQTT

**Expected behavior**

- Enabling BLE as a network 
- but disabling MQTT over BLE 
- and enabling BLE Wifi Provisioning
should result in the application initialising the BLE services, but then not starting the demo task until a wifi network is available

**To reproduce**
Steps to reproduce the behavior:

1. set `#define configENABLED_NETWORKS      ( AWSIOT_NETWORK_TYPE_WIFI | AWSIOT_NETWORK_TYPE_BLE )` in `aws_iot_network_config.h`
2. set `#define IOT_BLE_ENABLE_MQTT  (0) ` in `iot_ble_config.h`
3. set `#define IOT_BLE_ENABLE_WIFI_PROVISIONING    ( 1 )` in `iot_ble_config.h`
4. run the MQTT demo
5. connect to the device with a BLE Central (no need to use an AWS demo, Lightblue is enough)
6. the demo will start immediately and try to use BLE as a network

**Additional context**
As a fix I have wrapped certain parts of `iot_demo_freertos.c` in `#if BLE_ENABLED && (IOT_BLE_ENABLE_MQTT == 1)` checks.
I can create a PR if this is considered valuable and I have not misunderstood the meaning of the defines in question
Thank you!

tgsong, 2019-09-27:
Hi,

Thanks for reporting the issue. Our doxygen comment was not correct about `IOT_BLE_ENABLE_MQTT`, it was for BLE data transfer service, not for setting which connection type to use in MQTT. So to clarify,

- `configENABLED_NETWORKS` in `aws_iot_network_config.h` controls which module is being built, and initialized in demos. So if you set it to `( AWSIOT_NETWORK_TYPE_WIFI | AWSIOT_NETWORK_TYPE_BLE )`, then both WiFi and BLE will be built and initialized.
  - For BLE, the device will start advertisement and wait for a connection.
  - For WiFi, if `IOT_BLE_ENABLE_WIFI_PROVISIONING` is 0, then WiFi credentials in `aws_clientcredential.h` will be used. If `IOT_BLE_ENABLE_WIFI_PROVISIONING` is 1, then WiFi will be provisioned through BLE and the mobile app.
- `democonfigNETWORK_TYPES` in `aws_demo_config.h` controls which transportation layer is used for MQTT. If `democonfigNETWORK_TYPES` is set to `AWSIOT_NETWORK_TYPE_WIFI | AWSIOT_NETWORK_TYPE_BLE`, then the first connection available will used for MQTT. Based on the first point, if you want to provision WiFi through BLE, BLE will be connected first, and WiFi will be waiting until it's provisioned through the app, so in this case BLE will be used in MQTT.

So, if you intention is to do WiFi provisioning through BLE, then use WiFi for MQTT,

1. set `#define configENABLED_NETWORKS ( AWSIOT_NETWORK_TYPE_WIFI | AWSIOT_NETWORK_TYPE_BLE )` in `aws_iot_network_config.h`
1. set `#define democonfigNETWORK_TYPES    ( AWSIOT_NETWORK_TYPE_WIFI )` in aws_demo_config.h (Make sure it's set in the `#if defined( CONFIG_MQTT_DEMO_ENABLED )` section)

matteoscordino, 2019-09-28:
Thank you Tiangang,

that is exactly what I needed to achieve, and indeed it was all about understanding the intended usage of those configuration macros. I guess the related PR #1283 is not needed, I'll go ahead and close it.

gordonwang0, 2019-10-02:
Hi @matteoscordino,

Please let us know if you have any further questions.  Closing issue.

closed 2019-10-02 by gordonwang0