aws/amazon-freertos #1302: OTA_MAX_FILES Support increase

WiktorBAbove, 2019-09-30:
**Is your feature request related to a problem? Please describe.**  
We are planning to use the OTA method over MQTT for more than one target.
(Three targets: A powerful MCU that has two other connected). 
In https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c we see that OTA_MAX_FILES is limited to just 1.
The typical OTA job document gives a "files" array.
Will AFR support for more than one file?

**Describe the solution you would like.**  
handle OTA_MAX_FILES to more than one.
update the JSON parsing for object array handling. 

**Additional context**  
use cJSON instead?

Thank you!

WiktorBAbove, 2019-10-02:
Saw this from last year: https://github.com/aws/amazon-freertos/issues/51
Is there a roadmap for when this is available?
Obviously it requires some changes in AFR code; but is this feature in place in the "AWS IoT Core"?

pvyawaha, 2019-10-02:
Hi,

The OTA agent at this time supports only one file per OTA job and the `OTA_MAX_FILES` should always be set to 1. When we are creating OTA job using console as described in [OTA console Doc](https://docs.aws.amazon.com/freertos/latest/userguide/ota-console-workflow.html) the `fileid` used for the file is always 0. 

We can still do OTA for more than one target by using this `fileid` to identify different targets in separate OTA jobs. The `fileid` can be set different for each target while creating OTA jobs using CLI as described in [OTA CLI Doc](https://docs.aws.amazon.com/freertos/latest/userguide/ota-cli-workflow.html). We can then register and use the application specific callbacks for processing a custom job(if required) and OTA pal callbacks for handling specific target. The application specific callbacks can be registered for almost all of the OTA pal functions see `OTA_PAL_Callbacks_t;`  Alternatively we can build a custom OTA pal for the specific platform. 

The OTA agent after parsing the Job document stores this `fileid` in the file context which is passed to OTA pal functions.For example  `xOTA_Agent.xPALCallbacks.xWriteBlock` can determine depending on the `fileid`  in file context to handle the received block for specific target.

Can you please share which platform you are using and how the micro-controllers are interfaced with each other? Also if you can share some information about where the images will be stored, update logic etc. 

Please note that we continue to looking into adding more complete support for multi part OTA updates and your input is valuable in helping us with prioritization.

WiktorBAbove, 2019-10-08:
Hi,
Thank you for your long answer describing options and limitations. I see now that I cant make it do as I planned. 

The HW and the platform we are talking about is an ESP32(esp-idf) connected to a STM32 via UART for OTA. Its planned that all images and storage will be on AWS S3.

I'll close this issue.

BR /W

closed 2019-10-08 by WiktorBAbove