aws/amazon-freertos #1316: OTA job error in packet received size

ahmedwahdan, 2019-10-02:
**Describe the bug**
I merged the GreenGrass demo with the OTA demo and succeed to have the OTA job information through the GreenGrass core.
The problem is I always get an error of wrong received packet size which cause the connection to be closed.
`884 429154 [OTA Task] [OTA] Published file request to $aws/bin/things/ray-stm32/streams/b01b91c0-0cbe-404c-9278-f610bc406339/ge885 429957 [OTA Task] [OTA] Received file block 40, size 1024
886 429973 [OTA Task] [OTA] Remaining: 3
887 429974 [OTA Task] [OTA] Returned buffer to MQTT Client.
888 429984 [OTA Task] [OTA] Received file block 82, size 1024
889 429997 [OTA Task] [OTA] Remaining: 2
890 429998 [OTA Task] [OTA] Returned buffer to MQTT Client.
891 430007 [OTA Task] [OTA] Received file block 107, size 1024
892 430022 [OTA Task] [OTA] Remaining: 1
893 430022 [OTA Task] [OTA] Returned buffer to MQTT Client.
894 430032 [OTA Task] [OTA] Received file block 109, size 1024
895 430046 [OTA Task] [OTA] Stopping the file request timer.
896 430046 [OTA Task] [OTA] Authenticating and closing file.
897 430047 [OTA Task] [OTA] Error (17:16766875) closing OTA file.
898 430047 [OTA Task] [OTA] Aborting due to prvIngestDataBlock error -2
899 430048 [OTA Task] Sending command to MQTT task.
900 430048 [MQTT] Received message b0000 from queue.
901 430075 [OTA] [OTA] Queued: 128   Processed: 127   Dropped: 90
902 430249 [MQTT] MQTT Publish was successful.
903 430250 [MQTT] Notifying task.
904 430250 [OTA Task] Command sent to MQTT task passed.
905 430251 [OTA Task] [OTA] Published 'FAILED' status to $aws/things/ray-stm32/jobs/f18ef3fc_54cb_4a2f_8fd3_1537b76bc795/update906 430251 [OTA Task] Sending command to MQTT task.
907 430251 [MQTT] Received message c0000 from queue.`

**System information**
- ESP32 Dev-Kit
- VSCode
- Windows
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
- Project Greengrass+ OTA

**Expected behavior**
To have a successful OTA update through GreenGrass core

[GIST](https://gist.github.com/ahmedwahdan/606bc99e9b2a26b3f07f8ff1ee2bef6e)

Thank you!

pvyawaha, 2019-10-04:
Hi, 

Thank you for providing logs for the issue you are facing. From the logs it seems the file download is complete 

> 892 430022 [OTA Task] [OTA] Remaining: 1
> 893 430022 [OTA Task] [OTA] Returned buffer to MQTT Client.
> 894 430032 [OTA Task] [OTA] Received file block 109, size 1024
> 895 430046 [OTA Task] [OTA] Stopping the file request timer.
> 896 430046 [OTA Task] [OTA] Authenticating and closing file.

and during closing the file the signature verification failed and job status is updated as failed
> 898 430047 [OTA Task] [OTA] Aborting due to prvIngestDataBlock error -2

Can you please ensure that the code signing profile is created as per instructions and code signing certificate is updated?  Also can you please provide logs when you get packet size error and connection gets closed ?

ahmedwahdan, 2019-10-08:
Hi,
I seems I posted inaccurate log
This is the issue I'm facing.
`1976 24699 [NetRecv] [ERROR][NET][246980] Successfully received 1140 bytes.
1977 24699 [OTA Task] [prvIngestDataBlock] Received file block 255, size 1024
1978 24699 [OTA Task] [prvIngestDataBlock] Remaining: 617
1979 24699 [NetRecv] [ERROR][NET][246990] Successfully received 1 bytes.
1980 24700 [NetRecv] [ERROR][NET][247000] Successfully received 1 bytes.
1981 24701 [NetRecv] [ERROR][NET][247010] Successfully received 1 bytes.
1982 24740 [NetRecv] [ERROR][NET][247400] Successfully received 1140 bytes.
1983 24740 [OTA Task] [prvIngestDataBlock] Received file block 222, size 1024
1984 24740 [OTA Task] [prvIngestDataBlock] Remaining: 616
1985 24741 [NetRecv] [ERROR][NET][247410] Successfully received 1 bytes.
1986 24741 [NetRecv] [ERROR][NET][247410] Successfully received 1 bytes.
1987 24742 [NetRecv] [ERROR][NET][247420] Successfully received 1 bytes.
1988 24796 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
1989 24796 [iot_thread] State: Active  Received: 256   Queued: 256   Processed: 256   Dropped: 0
1990 24801 [NetRecv] [ERROR][NET][248010] Successfully received 1140 bytes.
1991 24801 [OTA Task] [prvIngestDataBlock] Received file block 233, size 1024
1992 24801 [OTA Task] [prvIngestDataBlock] Remaining: 615
1993 24802 [NetRecv] [ERROR][NET][248020] Successfully received 1 bytes.
1994 24802 [NetRecv] [ERROR][NET][248020] Successfully received 1 bytes.
1995 24803 [NetRecv] [ERROR][NET][248030] Successfully received 1 bytes.
1996 24916 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
1997 24916 [iot_thread] State: Active  Received: 257   Queued: 257   Processed: 257   Dropped: 0
1998 25004 [NetRecv] [ERROR][NET][250040] Error 0 while receiving data.
1999 25004 [NetRecv] [WARN ][NET][250040] Receive requested 1140 bytes, but 97 bytes received instead.
2000 25004 [NetRecv] [ERROR][MQTT][250040] (MQTT connection 0x3fff14f0) Error processing incoming data. Closing connection.
2001 25006 [NetRecv] [INFO ][MQTT][250060] (MQTT connection 0x3fff14f0) Network connection closed.
2002 25036 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2003 25036 [iot_thread] State: Active  Received: 257   Queued: 257   Processed: 257   Dropped: 0
2004 25051 [OTA Task] [WARN ][MQTT][250510] (MQTT connection 0x3fff14f0) Attempt to use closed connection.
2005 25051 [OTA Task] [ERROR][MQTT][250510] (MQTT connection 0x3fff14f0) New operation record cannot be created for a closed connec
tion
2006 25051 [OTA Task] [prvPublishGetStreamMessage] Failed: $aws/things/HelloWorld_Publisher/streams/AFR_OTA-8280976a-6ef5-4e94-a98b
-e4404d177970/get/cbor
2007 25156 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2008 25156 [iot_thread] State: Active  Received: 257   Queued: 257   Processed: 257   Dropped: 0
2009 25276 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2010 25276 [iot_thread] State: Active  Received: 257   Queued: 257   Processed: 257   Dropped: 0
2011 25301 [OTA Task] [WARN ][MQTT][253010] (MQTT connection 0x3fff14f0) Attempt to use closed connection.
2012 25301 [OTA Task] [ERROR][MQTT][253010] (MQTT connection 0x3fff14f0) New operation record cannot be created for a closed connec
tion
2013 25301 [OTA Task] [prvPublishGetStreamMessage] Failed: $aws/things/HelloWorld_Publisher/streams/AFR_OTA-8280976a-6ef5-4e94-a98b
-e4404d177970/get/cbor
`
The OTA can't recover from corrupted massage.

sarenameas, 2019-10-10:
When a "Get Stream" message fails we set a timer that will try the request again.
But if MQTT does not receive on the network the number of bytes it requests, then it will shut down. Therefore we keep trying again on a closed connection. 
Eventually the OTA should stop after the default of 32 times.

Thank you for showing this to us. We are working on a fix.

ahmedwahdan, 2019-10-10:
So could we retry the last request for number of times before closing the connection?
And what could be the reason to receive number of bytes less than the requested number ?

sarenameas, 2019-10-11:
Which version of Amazon FreeRTOS are you using? 

This log:
```
1998 25004 [NetRecv] [ERROR][NET][250040] Error 0 while receiving data.
```
was fixed with release version 201908.00. So we will try to keep receiving data even when a zero is returned.

Please update to 201908.00 and let me know how it goes.

ahmedwahdan, 2019-10-12:
I updated to 201908.00 and had this behavior 
I didn't get the closed connection error, but I got the WIFI disconnected 
`2870 43731 [iot_thread] [INFO ][MQTT][437310] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2871 43731 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2872 43731 [iot_thread] State: Active  Received: 845   Queued: 845   Processed: 845   Dropped: 0
2873 43751 [OTA Task] [prvIngestDataBlock] Received file block 810, size 1024
2874 43752 [OTA Task] [prvIngestDataBlock] Remaining: 25
2875 43792 [IP-task] SACK[25012,8883]: optlen 12 sending 1054679 - 1055373
I (438227) wifi: state: run -> init (2a0)
I (438227) wifi: pm stop, total sleep time: 350264486 us / 435505385 us

I (438227) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (438237) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 204
2876 43851 [iot_thread] [INFO ][MQTT][438510] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2877 43851 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2878 43851 [iot_thread] State: Active  Received: 846   Queued: 846   Processed: 846   Dropped: 0
2879 43971 [iot_thread] [INFO ][MQTT][439710] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2880 43971 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2881 43971 [iot_thread] State: Active  Received: 846   Queued: 846   Processed: 846   Dropped: 0
2882 44001 [OTA Task] [INFO ][MQTT][440010] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2883 44001 [OTA Task] [prvPublishGetStreamMessage] OK: $aws/things/HelloWorld_Publisher/streams/AFR_OTA-0792ade6-bfa2-41b1-b7a6-f079cb1cef50/get/cbor
2884 44091 [iot_thread] [INFO ][MQTT][440910] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2885 44091 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2886 44091 [iot_thread] State: Active  Received: 846   Queued: 846   Processed: 846   Dropped: 0
2887 44211 [iot_thread] [INFO ][MQTT][442110] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2888 44211 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2889 44211 [iot_thread] State: Active  Received: 846   Queued: 846   Processed: 846   Dropped: 0
2890 44251 [OTA Task] [INFO ][MQTT][442510] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2891 44251 [OTA Task] [prvPublishGetStreamMessage] OK: $aws/things/HelloWorld_Publisher/streams/AFR_OTA-0792ade6-bfa2-41b1-b7a6-f079cb1cef50/get/cbor
2892 44331 [iot_thread] [INFO ][MQTT][443310] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2893 44331 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2894 44331 [iot_thread] State: Active  Received: 846   Queued: 846   Processed: 846   Dropped: 0
2895 44451 [iot_thread] [INFO ][MQTT][444510] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2896 44451 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2897 44451 [iot_thread] State: Active  Received: 846   Queued: 846   Processed: 846   Dropped: 0
2898 44501 [OTA Task] [INFO ][MQTT][445010] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2899 44501 [OTA Task] [prvPublishGetStreamMessage] OK: $aws/things/HelloWorld_Publisher/streams/AFR_OTA-0792ade6-bfa2-41b1-b7a6-f079cb1cef50/get/cbor
2900 44571 [iot_thread] [INFO ][MQTT][445710] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2901 44571 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2902 44571 [iot_thread] State: Active  Received: 846   Queued: 846   Processed: 846   Dropped: 0
2903 44691 [iot_thread] [INFO ][MQTT][446910] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2904 44691 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2905 44691 [iot_thread] State: Active  Received: 846   Queued: 846   Processed: 846   Dropped: 0
2906 44751 [OTA Task] [INFO ][MQTT][447510] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2907 44751 [OTA Task] [prvPublishGetStreamMessage] OK: $aws/things/HelloWorld_Publisher/streams/AFR_OTA-0792ade6-bfa2-41b1-b7a6-f079cb1cef50/get/cbor
2908 44811 [iot_thread] [INFO ][MQTT][448110] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
2909 44811 [iot_thread] WAHDAN : vRunOTAUpdateDemo : Inside the while loop
2910 44811 [iot_thread] State: Active  Received: 846   Queued: 846   Processed: 846   Dropped: 0
2911 44931 [iot_thread] [INFO ][MQTT][449310] (MQTT connection 0x3fff8088) MQTT PUBLISH operation queued.
`

sarenameas, 2019-10-14:
Hello ahmedwahdan,

That Wi-Fi disconnect is very unfortunate. I would recommend moving closer to the access point or creating a strong access point.

Re-connection of the Wi-Fi network is not automatically enabled by Amazon FreeRTOS. This re-connection in ESP32 is only enabled if you provision Wi-Fi using BLE.

dan4thewin, 2019-12-08:
There haven't been any updates to this issue in almost 8 weeks.  I'll close it now.  If there's still a problem needing attention, please re-open the issue.

closed 2019-12-08 by dan4thewin