aws/amazon-freertos #1333: Support for Shared MQTT connection

d1g2w3, 2019-10-03:
Currently the default shadow enablement assumes the MQTT connection is handled by the Shadow. There are lots of circumstances, including other AWS services, that use MQTT. The eSharedMQTT option needs to be enabled.

To rectify this utilize the existing stubbed shared MQTT structures to allow an MQTT instance to be passed to the shadow. This will also mean the MQTT callback is handled elsewhere and a function with the data needs to be passed.

gordonwang0, 2019-10-03:
Hello,

This has been done in the new v2 Shadow API.  Please take a look here:
https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/shadow/shadow_functions.html

All the functions in the new API take an MQTT connection handle, allowing you to share the MQTT connection.

The legacy version of the API (the one that uses eSharedMQTT) is deprecated.  We will not be adding new features to that library, so you will need to switch to the new v2 API.

closed 2019-10-04 by d1g2w3

chegewara, 2019-12-03:
I would like to reopen this issue with question:
is it possible to share mqtt connection between shadow and OTA update? I have marged OTA job and shadow and initial connections are successful, but then my esp32 does not receive delta updates and OTA seems to works. After few seconds (timeout) i see such logs:
```
136 7420 [iot_thread] [ERROR][NET][74200] Error -27648 while sending data.
137 7420 [iot_thread] [ERROR][MQTT][74200] (MQTT connection 0x3ffb5324) Failed to send PINGREQ.
138 7421 [iot_thread] [INFO ][MQTT][74210] (MQTT connection 0x3ffb5324) Network connection closed.
139 7459 [iot_thread] State: Ready  Received: 1   Queued: 1   Processed: 1   Dropped: 0
140 7559 [iot_thread] State: Ready  Received: 1   Queued: 1   Processed: 1   Dropped: 0
141 7659 [iot_thread] State: Ready  Received: 1   Queued: 1   Processed: 1   Dropped: 0
```
As we can see OTA still works, but shadow connection is closed due to PING timeout.

esp32 on AFR master.

chegewara, 2019-12-03:
Ok, i found correct settings and it is possible now.

egoltzman, 2019-12-14:
Can you share the updated setting you used to overcome this problem?
I'm facing the same issue

tandelprashant, 2020-04-30:
Hi @chegewara 
could you please share the settings that you used to fix this problem?
I am seeing the same error.
> Error -27648 while sending data

Thank you!

chegewara, 2020-04-30:
Hi,
its not easy to share solution i am using, because i completely reworked IoT_mqtt init, shadow and OTA tasks creation etc.
Most important difference IIRC is in this code:
https://github.com/aws/amazon-freertos/blob/master/demos/ota/aws_iot_ota_update_demo.c#L209-L255

```
    static bool inOTAnow = false;
    OTA_State_t eState;
    OTA_ConnectionContext_t xOTAConnectionCtx = { 0 };
    const IotNetworkInterface_t* pNetworkInterface = AwsIotNetworkManager_GetNetworkInterface(AWSIOT_NETWORK_TYPE_WIFI);
    void* pCredentials = AwsIotNetworkManager_GetCredentials(AWSIOT_NETWORK_TYPE_WIFI);
    uint16_t packets;

    ESP_LOGI(TAG, "vRunOTAUpdateTask: OTA version %u.%u.%u",
             xAppFirmwareVersion.u.x.ucMajor,
             xAppFirmwareVersion.u.x.ucMinor,
             xAppFirmwareVersion.u.x.usBuild);

    for (;;)
    {
        xNetworkConnected = prxCreateNetworkConnection();
        if (xNetworkConnected)
        {
            xOTAConnectionCtx.pvControlClient = xConnection.xMqttConnection;
            xOTAConnectionCtx.pxNetworkInterface = ( void * ) pNetworkInterface;
            xOTAConnectionCtx.pvNetworkCredentials = pCredentials;
            if ( ( eState = OTA_GetAgentState() ) == eOTA_AgentState_Stopped )
            {
                ESP_LOGI(TAG, "vRunOTAUpdateTask: Running OTA_AgentInit");
                eState = OTA_AgentInit( ( void * ) ( &xOTAConnectionCtx ), ( const uint8_t * ) ( my_thing_name ), App_OTACompleteCallback, ( TickType_t ) ~0 );
            }
            else
            {
                ESP_LOGI(TAG, "vRunOTAUpdateTask:  OTA_GetAgentState(%s)", pcStateStr[eState]);
            }
```
In my code i am passing (reusing) the same connection i did initialize for shadow.

This is how i start OTA task and pass values:
```
int vStartOTAUpdate(IotMqttNetworkInfo_t networkInfo, IotMqttConnection_t mqttConnection) <--- initialized for shadow
{
    int xRet = EXIT_SUCCESS;
    BaseType_t ret;

    ESP_LOGI(TAG, "vStartOTAUpdate: starting");
    xConnection.xNetworkInfo = networkInfo;
    xConnection.xMqttConnection = mqttConnection;
```