aws/amazon-freertos #1337: How to  enable ESP32 Dual core ?

elieDaan, 2019-10-04:
**Describe the bug**
Hello, I've seen that the ESP32 under AWS FreeRTOS is booting in single core mode. Is there a way to use the second core ? 

**System information**
- Which hardware board or part numbers? M5STack
- AWS FreeRtos 
- Operating System MacOS
- Project Custom ApplicationEcho|Sub pub Demo|Simple TCP Server]
- If your project is a Custom Application, please add the relevant code snippet in the section 

`Code to reproduce the bug`. 
I (256) boot: Loaded app from partition at offset 0x10000
I (256) boot: Disabling RNG early entropy source...
I (256) cpu_start: Pro cpu up.
I (260) cpu_start: Single core mode
I (265) heap_init: Initializing. RAM available for dynamic allocation:
I (271) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (277) heap_init: At 3FFC7B50 len 000184B0 (97 KiB): DRAM
I (284) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (290) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (296) heap_init: At 4008C168 len 00013E98 (79 KiB): IRAM
I (303) cpu_start: Pro cpu start user code
I (320) cpu_start: Starting scheduler on PRO CPU.
Init done!
I (321) gpio: GPIO[38]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 
I (321) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 
I (331) gpio: GPIO[37]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 

Thank you!

dnoal, 2019-10-04:
Thanks for this.  At present, FreeRTOS does not formally support SMP.  Our examples support only a single core as you have stated.  We know Espressif supports SMP in their distribution of FreeRTOS, so using their ESP-IDF, along with the AWS CSDK (github.com/espressif/esp-aws-iot), is an option for you to connect your dual core ESP32 based device to AWS IoT leveraging both cores.  

We have worked on an internal prototype of a current Amazon FreeRTOS release using the same Espressif supplied SMP capable kernel that they distribute.  We aren’t quite ready to post that as a functional example but hope to do so soon.  

We are in touch with Espressif on their SMP approach and are in discussions on how to best support SMP generally in FreeRTOS but this feature is not expected to be released in the near term.  This is a request we have heard before, and appreciate the input from you that helps us with prioritization.

elieDaan, 2019-10-07:
Thanks i will try it. I need AWS FreeRTOS because there is all OTA stuffs with security already implemented..

sarenameas, 2019-10-11:
Hello elieDaan how did trying out the new branch go?

elieDaan, 2019-10-14:
Helle Sarena, currently I'm working on the mqtt side of my project but i'm gonna test it at the end of the week or the begin of the next week. I will let you know about my tests ;)

abhidixi11, 2019-11-30:
Hello  elieDaan ,
Any updates on this issue?  I'll close this issue if you don't have any further questions. 
Thanks !

closed 2019-12-02 by elieDaan