aws/amazon-freertos #1357: [GreenGrass] Can't create new MQTT connection if the previous one was not closed

ahmedwahdan, 2019-10-08:
**Describe the bug**
I'm using greengrass demo to connect to greengrass core over EC2 instance.
I'm using mobile phone as AP.
If the hardware was restarted without closing the connection, I have to reboot the mobile phone!

**System information**
- ESP32
- VSCode
- Operating System [Windows]
- Version of Amazon FreeRTOS (201906.00_Major-539-g3a519702)
- Project [Greengrass]

**Expected behavior**
This issue is not visible when using other demos, I think the issue may be related to using the MQTT_Agent instead of directly using MQTT v2 APIs directly

Thank you!

sarenameas, 2019-10-10:
Hello ahmedwahdan,

How do you restart the hardware? Do you power cycle? 

I am trying this out.

ahmedwahdan, 2019-10-10:
Hello Sarenameas,
Sometimes after successful demo complete, and some times if I restart the esp32 (power reset) before closing the connection, the connection can't established again with the greengrass core.

sarenameas, 2019-10-10:
Could you please turn on logging in the TCP/IP stack in 
**vendors\espressif\boards\esp32\aws_demos\config_files\FreeRTOSIPConfig.h** by setting:
```
#define ipconfigHAS_DEBUG_PRINTF    1
```
Then please post the logs here.

ahmedwahdan, 2019-10-12:
` make flash monitor
Flashing binaries to serial port COM6 (app at offset 0x20000 )...
esptool.py v2.6
Serial port COM6
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
MAC: 30:ae:a4:22:ce:80
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 23712 bytes to 14071...
Wrote 23712 bytes (14071 compressed) at 0x00001000 in 0.2 seconds (effective 1100.8 kbit/s)...
Hash of data verified.
Compressed 1099136 bytes to 669493...
Wrote 1099136 bytes (669493 compressed) at 0x00020000 in 10.6 seconds (effective 827.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 141...
Wrote 3072 bytes (141 compressed) at 0x00008000 in 0.0 seconds (effective 1804.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on COM6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6252
load:0x40078000,len:11276
ho 0 tail 12 room 4
load:0x40080000,len:6084
0x40080000: _WindowOverflow4 at D:/Wahdan/AFR_08_19/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

entry 0x4008032c
0x4008032c: _KernelExceptionVector at ??:?

I (30) boot: ESP-IDF v3.1.3-25-g36733db8d 2nd stage bootloader
I (30) boot: compile time 13:22:56
I (31) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (67) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (74) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (82) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (89) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (97) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (104) boot: End of partition table
I (108) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x2f4cc (193740) map
I (185) esp_image: segment 1: paddr=0x0004f4f4 vaddr=0x3ffc0000 size=0x00b1c (  2844) load
I (187) esp_image: segment 2: paddr=0x00050018 vaddr=0x400d0018 size=0xc376c (800620) map
0x400d0018: _flash_cache_start at ??:?

I (472) esp_image: segment 3: paddr=0x0011378c vaddr=0x3ffc0b1c size=0x02424 (  9252) load
I (476) esp_image: segment 4: paddr=0x00115bb8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at D:/Wahdan/AFR_08_19/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

[... 726 lines cut ...]
43 1842 [IP-task] FreeRTOS_closesocket[24324 to 36d5e2bcip:8883]: buffers 60 socks 0
44 1842 [iot_thread] Socket closed.
45 1842 [iot_thread] Stack high watermark for discovery helper task: 2768.
46 1842 [iot_thread] Greengrass device discovered.
47 1842 [iot_thread] Establishing MQTT communication to Greengrass...
48 1842 [iot_thread] prvProcessDNSCache: look-up: 'ec2-54-213-226-188.us-west-2.compute.amazonaws.com' @ 36d5e2bcip
49 1842 [iot_thread] FreeRTOS_gethostbyname: found 'ec2-54-213-226-188.us-west-2.compute.amazonaws.com' in cache: bce2d536ip
50 1843 [iot_thread] FreeRTOS_connect: 25012 to 36d5e2bcip:8883
51 1843 [iot_thread] Socket 25012 -> 36d5e2bcip:8883 State eCLOSED->eCONNECT_SYN
52 1843 [IP-task] prvSocketSetMSS: 1400 bytes for 36d5e2bcip:8883
53 1843 [IP-task] prvWinScaleFactor: uxRxWinSize 1 MSS 1400 Factor 0
54 1843 [IP-task] Connect[36d5e2bcip:8883]: next timeout 1: 3000 ms
55 1923 [IP-task] MSS change 1400 -> 1388
56 1923 [IP-task] Change mss 1400 => 1388
57 1923 [IP-task] TCP: active 25012 => 36d5e2bcip:8883 set ESTAB (scaling 1)
58 1923 [IP-task] Socket 25012 -> 36d5e2bcip:8883 State eCONNECT_SYN->eESTABLISHED
59 2438 [iot_thread] ERROR: Handshake failed with error code -29312
60 2438 [iot_thread] [ERROR][NET][24380] Failed to establish new connection.
61 2438 [IP-task] FreeRTOS_closesocket[25012 to 36d5e2bcip:8883]: buffers 60 socks 0
62 2439 [iot_thread] [ERROR][MQTT][24390] Failed to establish new MQTT connection, error NETWORK ERROR.
63 2439 [iot_thread] ERROR: Could not connect to the Broker.
64 2439 [iot_thread] Heap low watermark: 8848. Stack high watermark: 2768.
65 2439 [iot_thread] ----Demo finished----
I (83686) wifi: state: run -> init (7c0)
`

ahmedwahdan, 2019-10-15:
@sarenameas Any updates related to this issue?

sarenameas, 2019-10-15:
Hey ahmedwahdan,

Apologies for the delay. We are under investigation. I asked for logs earlier because I could reproduce your issue, but I wanted to see if you had the same one as me; and you did! 

This the log I wanted to see if you had also:
```
56 2123 [IP-task] eCONNECT_SYN: flags 0012 expected, not 0010
```
It appears the server is responding with just and ACK instead of a SYN-ACK for a new connection request.  We are working on this now. We will get back to you on our findings.

Thanks for your patience.

sarenameas, 2019-10-15:
Hello again,

This has been addressed now with PR #1413 .

Thanks for bringing this issue to the light.

closed 2019-10-16 by sarenameas

ahmedwahdan, 2019-10-16:
Thanks @sarenameas  for the fix.
Could you please explain what was the issue?

ahmedwahdan, 2019-10-18:
I got this crash sometimes
`10 892 [IP-task] vDHCPProcess: offer c0a80106ip
11 892 [IP-task] vDHCPProcess: acked c0a80106ip
12 893 [iot_thread] [INFO ][DEMO][8930] Successfully initialized the demo. Network type for the demo: 1
13 893 [iot_thread] [INFO ][MQTT][8930] MQTT library successfully initialized.
14 893 [iot_thread] [INFO ][DEMO][8930] Attempting automated selection of Greengrass device

15 923 [iot_thread] prvProcessDNSCache: add: 'a3rcy4vytvmu2y-ats.iot.us-west-2.amazonaws.com' @ 327087e0ip
16 923 [iot_thread] FreeRTOS_connect: 35869 to 327087e0ip:8443
17 923 [iot_thread] Socket 35869 -> 327087e0ip:8443 State eCLOSED->eCONNECT_SYN
18 923 [IP-task] prvSocketSetMSS: 1400 bytes for 327087e0ip:8443
19 923 [IP-task] prvWinScaleFactor: uxRxWinSize 1 MSS 1400 Factor 0
20 923 [IP-task] Connect[327087e0ip:8443]: next timeout 1: 3000 ms
21 1076 [IP-task] MSS change 1400 -> 1412
22 1076 [IP-task] TCP: active 35869 => 327087e0ip:8443 set ESTAB (scaling 1)
23 1076 [IP-task] Socket 35869 -> 327087e0ip:8443 State eCONNECT_SYN->eESTABLISHED
24 1076 [IP-task] MSS change 1400 -> 1412
25 1592 [iot_thread] ERROR: Handshake failed with error code -29312
26 1592 [iot_thread] About to close socket.
CORRUPT HEAP: multi_heap.c:474 detected at 0x3ffef288
abort() was called at PC 0x40093dc9 on core 0
0x40093dc9: multi_heap_assert at D:/Wahdan/AFR_08_19/amazon-freertos/vendors/espressif/esp-idf/components/heap/multi_heap.c:377
 (inlined by) multi_heap_free_impl at D:/Wahdan/AFR_08_19/amazon-freertos/vendors/espressif/esp-idf/components/heap/multi_heap.c:474

Backtrace: 0x400943e8:0x3ffddda0 0x400945b7:0x3ffdddc0 0x40093dc9:0x3ffddde0 0x400823de:0x3ffdde00 0x40091dca:0x3ffdde20 0x400ebb72:0x3ffdde40 0x400f129e:0x3ffdde60 0x400f14dc:0x3ffdde80 0x400
edf99:0x3ffddea0 0x400e9a4e:0x3ffddec0 0x400e9b8f:0x3ffddef0 0x400e9521:0x3ffddf20 0x400e996b:0x3ffddf50 0x400e2ff1:0x3ffddf80 0x400e309b:0x3ffddfc0 0x400efbf3:0x3ffddfe0 0x400edaa1:0x3ffde000

0x400943e8: invoke_abort at D:/Wahdan/AFR_08_19/amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:676

0x400945b7: abort at D:/Wahdan/AFR_08_19/amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:676

0x40093dc9: multi_heap_assert at D:/Wahdan/AFR_08_19/amazon-freertos/vendors/espressif/esp-idf/components/heap/multi_heap.c:377
 (inlined by) multi_heap_free_impl at D:/Wahdan/AFR_08_19/amazon-freertos/vendors/espressif/esp-idf/components/heap/multi_heap.c:474

0x400823de: heap_caps_free at D:/Wahdan/AFR_08_19/amazon-freertos/vendors/espressif/esp-idf/components/heap/heap_caps.c:123

0x40091dca: vQueueDelete at D:/Wahdan/AFR_08_19/amazon-freertos/freertos_kernel/queue.c:2364

0x400ebb72: C_CloseSession at D:/Wahdan/AFR_08_19/amazon-freertos/libraries/abstractions/pkcs11/mbedtls/iot_pkcs11_mbedtls.c:628

0x400f129e: prvFreeContext at D:/Wahdan/AFR_08_19/amazon-freertos/libraries/freertos_plus/standard/tls/src/iot_tls.c:129

0x400f14dc: TLS_Recv at D:/Wahdan/AFR_08_19/amazon-freertos/libraries/freertos_plus/standard/tls/src/iot_tls.c:886

0x400edf99: SOCKETS_Recv at D:/Wahdan/AFR_08_19/amazon-freertos/libraries/abstractions/secure_sockets/freertos_plus_tcp/iot_secure_sockets.c:231

0x400e9a4e: GGD_SecureConnect_Disconnect at D:/Wahdan/AFR_08_19/amazon-freertos/libraries/freertos_plus/aws/greengrass/src/aws_helper_secure_connect.c:201

0x400e9b8f: GGD_SecureConnect_Connect at D:/Wahdan/AFR_08_19/amazon-freertos/libraries/freertos_plus/aws/greengrass/src/aws_helper_secure_connect.c:165

0x400e9521: GGD_JSONRequestStart at D:/Wahdan/AFR_08_19/amazon-freertos/libraries/freertos_plus/aws/greengrass/src/aws_greengrass_discovery.c:251

0x400e996b: GGD_GetGGCIPandCertificate at D:/Wahdan/AFR_08_19/amazon-freertos/libraries/freertos_plus/aws/greengrass/src/aws_greengrass_discovery.c:174

0x400e2ff1: _discoverGreengrassCore at D:/Wahdan/AFR_08_19/amazon-freertos/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c:215

0x400e309b: vStartGreenGrassDiscoveryTask at D:/Wahdan/AFR_08_19/amazon-freertos/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c:284

0x400efbf3: runDemoTask at D:/Wahdan/AFR_08_19/amazon-freertos/demos/demo_runner/iot_demo_freertos.c:372

0x400edaa1: _threadRoutineWrapper at D:/Wahdan/AFR_08_19/amazon-freertos/libraries/abstractions/platform/freertos/iot_threads_freertos.c:318

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6252
load:0x40078000,len:11276
ho 0 tail 12 room 4
load:0x40080000,len:6084
0x40080000: _WindowOverflow4 at D:/Wahdan/AFR_08_19/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

entry 0x4008032c
0x4008032c: _KernelExceptionVector at ??:?

`