aws/amazon-freertos #1367: Build warnings in MDK

MichalChromecNXP, 2019-10-09:
Could you please fix these build warnings:

../../../../../rtos/amazon-freertos/demos/demo_runner/iot_demo_freertos.c(48): warning: macro expansion producing 'defined' has undefined behavior [-Wexpansion-to-defined]
#if MQTT_DEMO_TYPE_ENABLED
    ^
../../../../../rtos/amazon-freertos/demos/demo_runner/iot_demo_freertos.c(46): note: expanded from macro 'MQTT_DEMO_TYPE_ENABLED'
#define MQTT_DEMO_TYPE_ENABLED    ( defined( CONFIG_MQTT_DEMO_ENABLED ) || defined( CONFIG_SHADOW_DEMO_ENABLED ) || defined( CONFIG_DEFENDER_DEMO_ENABLED ) || defined( CONFIG_OTA_UPDATE_DEMO_ENABLED ) )

compiling iot_demo_freertos.c...
../../../../../rtos/amazon-freertos/libraries/abstractions/pkcs11/mbedtls/iot_pkcs11_mbedtls.c(700): warning: comparison of address of 'xP11Context.xMbedEntropyContext' not equal to a null pointer is always true [-Wtautological-pointer-compare]
        if( NULL != &xP11Context.xMbedEntropyContext )
            ~~~~     ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
../../../../../rtos/amazon-freertos/libraries/abstractions/pkcs11/mbedtls/iot_pkcs11_mbedtls.c(705): warning: comparison of address of 'xP11Context.xMbedDrbgCtx' not equal to a null pointer is always true [-Wtautological-pointer-compare]
        if( NULL != &xP11Context.xMbedDrbgCtx )
            ~~~~     ~~~~~~~~~~~~^~~~~~~~~~~~
../../../../../rtos/amazon-freertos/libraries/abstractions/pkcs11/mbedtls/iot_pkcs11_mbedtls.c(1045): warning: comparison of address of 'pxSession->xSHA256Context' not equal to a null pointer is always true [-Wtautological-pointer-compare]
        if( NULL != &pxSession->xSHA256Context )
            ~~~~     ~~~~~~~~~~~^~~~~~~~~~~~~~

sarenameas, 2019-10-10:
I am unable to see these warnings in our current UVision projects. 
Could you send me the project you are using?
Or tell me what configurations you enabled and UVision options you have enabled as well?

sarenameas, 2019-10-10:
PR #1367 has been merged.

For warnings: 
compiling iot_demo_freertos.c...
../../../../../rtos/amazon-freertos/libraries/abstractions/pkcs11/mbedtls/iot_pkcs11_mbedtls.c(700): warning: comparison of address of 'xP11Context.xMbedEntropyContext' not equal to a null pointer is always true [-Wtautological-pointer-compare]
if( NULL != &xP11Context.xMbedEntropyContext )
~~~~ ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
../../../../../rtos/amazon-freertos/libraries/abstractions/pkcs11/mbedtls/iot_pkcs11_mbedtls.c(705): warning: comparison of address of 'xP11Context.xMbedDrbgCtx' not equal to a null pointer is always true [-Wtautological-pointer-compare]
if( NULL != &xP11Context.xMbedDrbgCtx )
~~~~ ~~~~~~~~~~~~^~~~~~~~~~~~

These contexts will be deleted in: https://github.com/aws/amazon-freertos/pull/1372

closed 2019-10-10 by sarenameas