aws/amazon-freertos #1380: iot_demo_freertos.c _cleanup missing AwsIotNetworkManager_DisableNetwork( configENABLED_NETWORKS );

teuteuguy, 2019-10-10:
**Describe the bug**
When wanting to run the demo code multiple times, the _cleanup function does not cleanup the wifi network. Looks like it's simply missing a call to AwsIotNetworkManager_DisableNetwork( configENABLED_NETWORKS );

**System information**
- Which hardware board or part numbers? ESP32
- IDE used: vscode
- Operating System [Windows|Linux|MacOS] Mac
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it) latest
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server] MQTT
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
I expect the demo code to be able to be run multiple times

sarenameas, 2019-10-10:
How exactly are you trying to run the demo multiple times? Could you post your code snippet or changes?
I am able to rerun the same demo code multiple times by adding the following in **demos\demo_runner\iot_demo_freertos.c** :
```
#include "platform/iot_clock.h"
...
        while ( true )
        {
            /* Run the demo. */
            status = pContext->demoFunction( true,
                                            clientcredentialIOT_THING_NAME,
                                            pConnectionParams,
                                            pCredentials,
                                            pNetworkInterface );

            /* Log the demo status. */
            if( status == EXIT_SUCCESS )
            {
                /* DO NOT EDIT - This message is used in the test framework to
                * determine whether or not the demo was successful. */
                IotLogInfo( "Demo completed successfully." );
            }
            else
            {
                IotLogError( "Error running demo." );
            }
            IotClock_SleepMs( 1000 );
        }
```

teuteuguy, 2019-10-15:
Hi @sarenameas,

yes, your code will work, however this assumes that the _initialize has been executed before once, before a re-run.

My objective was to have a full init, and cleanup at each run of the demo code.
In the context of the networking stack / wifi etc ... if you want to re-run the _initialize function, you need the AwsIotNetworkManager_DisableNetwork( configENABLED_NETWORKS ); to be executed in the _cleanup.

My simple change is in the _cleanup function with the following code:
```
static void _cleanup( void )
{
    /* Remove network manager subscription */
    AwsIotNetworkManager_DisableNetwork( configENABLED_NETWORKS ); // line added here
    AwsIotNetworkManager_RemoveSubscription( subscription );
    IotSemaphore_Destroy( &demoNetworkSemaphore );
    IotSdk_Cleanup();
}
```

qiutongs, 2019-10-15:
OK, I see what you want to do here. And we are interested in why you try to run demos end-to-end multiple times? Is there any business requirement?

teuteuguy, 2019-10-16:
I'm using a dev kit i'm working on, and effectively, running the MQTT demo, but triggering running it from a button press on the dev kit.

Idea being, on every press of the button, run the MQTT demo.
Given the demo as is, runs, publishes a couple times on couple topics, then quits.

I think if the demo finishes, it should fully cleanup.

qiutongs, 2019-10-16:
After adding the "disable network", I ran it multiple time locally and it worked well.

qiutongs, 2019-10-16:
It is welcomed to create a PR if you have need to fix or improve the code in the future.

closed 2019-10-16 by qiutongs