aws/amazon-freertos #1385: MQTT Agent doesn't connect on MQTT v2

sumit612, 2019-10-11:
**Describe the bug**
MQTT agent was connecting with TLS certificates in MQTT v1. After porting the same application on MQTT v2 with same connect flags( mqttagentREQUIRE_TLS | mqttagentUSE_AWS_IOT_ALPN_443 ) and credentials MQTT_AGENT_CONNECT() fails to connect with AWS broker via wifi while MQTT Demo is working fine.

**System information**
- ESP32
- aws cli
- Windows
- Version of Amazon FreeRTOS : 201906.00_Major-921-g7852393c9
- Custom Application
- Connect Parameters:
static MQTTAgentConnectParams_t xConnectParameters =
{
    clientcredentialMQTT_BROKER_ENDPOINT,  
    democonfigMQTT_AGENT_CONNECT_FLAGS,           
    pdFALSE,                                        
    clientcredentialMQTT_BROKER_PORT,                   
    ( const uint8_t * ) ( clientcredentialIOT_THING_NAME ),
    ( uint16_t ) strlen( ( const char * ) clientcredentialIOT_THING_NAME ),  
    pdFALSE,                                            
    NULL,                                  
    MQTTAgentCallback,                                  
    NULL,                                            
    0                                                  
};

**Expected behavior**
Should connect with AWS mqtt broker

**Screenshots or console output**
37 1270 [MqttTask] ERROR: Handshake failed with error code -9984
38 1270 [MqttTask] [ERROR][NET][12700] Failed to establish new connection.
39 1270 [MqttTask] [ERROR][MQTT][12700] Failed to establish new MQTT connection, error NETWORK ERROR.

sarenameas, 2019-10-11:
Hello sumit612,

Using the latest in master of this post. I am able to make an MQTT connection with your exact xConnectParameters. 

I added to ESP32's aws_demo_config.h:
```
#define democonfigMQTT_AGENT_CONNECT_FLAGS             ( mqttagentREQUIRE_TLS | mqttagentUSE_AWS_IOT_ALPN_443 )
```
My quick and dirty code:
```
    MQTTAgentHandle_t xMQTTClientHandle;

    /* Create MQTT Client. */
    if( MQTT_AGENT_Create( &( xMQTTClientHandle ) ) == eMQTTAgentSuccess )
    {
        MQTTAgentConnectParams_t xConnectParams =
        {
            clientcredentialMQTT_BROKER_ENDPOINT,
            democonfigMQTT_AGENT_CONNECT_FLAGS,
            pdFALSE,
            clientcredentialMQTT_BROKER_PORT,
            ( const uint8_t * ) ( clientcredentialIOT_THING_NAME ),
            ( uint16_t ) strlen( ( const char * ) clientcredentialIOT_THING_NAME ),
            pdFALSE,
            NULL,
            NULL, /* No call back not needed just to connect. */
            NULL,
            0
        };

        if( MQTT_AGENT_Connect( xMQTTClientHandle,
                                &xConnectParams,
                                pdMS_TO_TICKS( 20000UL ) ) != eMQTTAgentSuccess )
        {
            configPRINTF( ( "ERROR: Could not connect to the Broker.\r\n" ) );
        }
        else
        {
            if( MQTT_AGENT_Disconnect( xMQTTClientHandle,
                                   pdMS_TO_TICKS( 20000UL ) ) == eMQTTAgentSuccess )
            {
                configPRINTF( ( "Disconnected from the broker.\r\n" ) );

                if( MQTT_AGENT_Delete( xMQTTClientHandle ) == eMQTTAgentSuccess )
                {
                    configPRINTF( ( "Deleted Client.\r\n" ) );
                }
                else
                {
                    configPRINTF( ( "ERROR:  MQTT_AGENT_Delete() Failed.\r\n" ) );
                }
            }
            else
            {
                configPRINTF( ( "ERROR:  Did not disconnected from the broker.\r\n" ) );
            }
        }
    }
```

my output:
```
5 303 [iot_thread] [INFO ][DEMO][3030] Successfully initialized the demo. Network type for the demo: 1
6 783 [iot_thread] [INFO ][MQTT][7830] Establishing new MQTT connection.
7 784 [iot_thread] [INFO ][MQTT][7830] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
8 784 [iot_thread] [INFO ][MQTT][7840] (MQTT connection 0x3ffe2f3c, CONNECT operation 0x3ffe210c) Waiting for operation completion.
9 811 [iot_thread] [INFO ][MQTT][8110] (MQTT connection 0x3ffe2f3c, CONNECT operation 0x3ffe210c) Wait complete with result SUCCESS.
10 811 [iot_thread] [INFO ][MQTT][8110] New MQTT connection 0x3ffe224c established.
11 811 [iot_thread] [INFO ][MQTT][8110] (MQTT connection 0x3ffe2f3c) Disconnecting connection.
12 811 [iot_thread] [INFO ][MQTT][8110] (MQTT connection 0x3ffe2f3c, DISCONNECT operation 0x3ffe210c) Waiting for operation completion.
13 813 [iot_thread] [INFO ][MQTT][8130] (MQTT connection 0x3ffe2f3c, DISCONNECT operation 0x3ffe210c) Wait complete with result SUCCESS.
14 813 [iot_thread] [INFO ][MQTT][8130] (MQTT connection 0x3ffe2f3c) Connection disconnected.
15 814 [iot_thread] [INFO ][MQTT][8140] (MQTT connection 0x3ffe2f3c) Network connection closed.
16 817 [iot_thread] [INFO ][MQTT][8170] (MQTT connection 0x3ffe2f3c) Network connection destroyed.
17 818 [iot_thread] Disconnected from the broker.
18 818 [iot_thread] Deleted Client.
19 818 [iot_thread] ----Demo finished----
20 818 [iot_thread] [INFO ][DEMO][8180] Demo completed successfully.
21 919 [iot_thread] [INFO ][INIT][9190] SDK cleanup done.
22 919 [iot_thread] [INFO ][DEMO][9190] -------DEMO FINISHED-------
```

```
37 1270 [MqttTask] ERROR: Handshake failed with error code -9984
```
This error translates to -2700 in libraries\3rdparty\mbedtls\include\mbedtls\x509.h. This means some certificate verification failed somewhere down the handshake process. 
Please check and make sure that the keys in demos\include\aws_clientcredential_keys.h are correct and in correct format. Also that those keys are associated with your thing.

sumit612, 2019-10-11:
But used the same credential in my previous code and it connected every time. The previous AFR version was "Amazon FreeRTOS V1.4.1". 
* Will the mbedTLS debug console data help you suggest me some help? 
* Or any configuration change in menuconfig?
I'm attaching here two screenshots from two project.

The previous config:
[image]

The new config:
[image]

* I'm also getting an error after restarting ESP32:
E (287) PKCS11: failed nvs get file size 4354 0

After that I enabled multi blob support. But no change.

*Else I only see the difference is the newer AFR is using new  vDevModeKeyProvisioning()

sarenameas, 2019-10-12:
Yes I think turning on mbedtls debug logs would help. Please send that log to us.
Are you using the default esp_config.h (under vendors\espressif\boards\esp32\aws_demos\application_code\espressif_code\mbedtls\port\include\mbedtls\esp_config.h)? If you changed your mbedtls config.h please post it here.
Also make sure that you are erasing the flash before reprogramming.

sumit612, 2019-10-12:
No, I didn't change anything on that file.
I also tried erasing flash.

Here is the debug log:

4819 16161 [MqttTask] mbedTLS: |2| 0x3ffd27f8: => handshake
4820 16161 [MqttTask] mbedTLS: |2| 0x3ffd27f8: client state: 0
4821 16161 [MqttTask] mbedTLS: |2| 0x3ffd27f8: => flush output
4822 16161 [MqttTask] mbedTLS: |2| 0x3ffd27f8: <= flush output
4823 16161 [MqttTask] mbedTLS: |2| 0x3ffd27f8: client state: 1
4824 16161 [MqttTask] mbedTLS: |2| 0x3ffd27f8: => flush output
4825 16161 [MqttTask] mbedTLS: |2| 0x3ffd27f8: <= flush output
4826 16161 [MqttTask] mbedTLS: |2| 0x3ffd27f8: => write client hello
4827 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, max version: [3:3]
4828 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, current time: 161
4829 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: dumping 'client hello, random bytes' (32 bytes)
4830 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: 0000:  00 00 00 a1 1c 3e ee 47 d7 d0 d3 e3 65 a5 31 a6  .....>.G....e.1.
4831 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: 0010:  9c 48 58 67 19 dd 5d 9f 42 8b 59 fc 76 5d 53 70  .HXg..].B.Y.v]Sp
4832 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, session id len.: 0
4833 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: dumping 'client hello, session id' (0 bytes)
4834 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c0ad
4835 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c09f
4836 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: 006b
4837 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c00a
4838 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c014
4839 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: 0039
4840 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c0af
4841 16161 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c0a3
4842 16162 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c02b
4843 16162 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c02f
4844 16162 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: 009e
4845 16162 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c0ac
4846 16162 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c09e
4847 16162 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c023
4848 16162 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c027
4849 16162 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: 0067
4850 16162 [MqttTask] mbedTLS: |3| 0x3ffd27f8: client hello, add ciphersuite: c009
4920 16548 [MqttTask] mbedTLS: |2| 0x3ffd27f8: in_left: 0, nb_want: 5
4921 16548 [MqttTask] mbedTLS: |2| 0x3ffd27f8: ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
4922 16548 [MqttTask] mbedTLS: |2| 0x3ffd27f8: <= fetch input
4923 16548 [MqttTask] mbedTLS: |4| 0x3ffd27f8: dumping 'input record header' (5 bytes)
4924 16548 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0000:  16 03 03 15 24                                   ....$
4925 16548 [MqttTask] mbedTLS: |3| 0x3ffd27f8: input record: msgtype = 22, version = [3:3], msglen = 5412
4926 16548 [MqttTask] mbedTLS: |2| 0x3ffd27f8: => fetch input
4927 16548 [MqttTask] mbedTLS: |2| 0x3ffd27f8: in_left: 5, nb_want: 5417
4928 16548 [MqttTask] mbedTLS: |2| 0x3ffd27f8: in_left: 5, nb_want: 5417
4929 16548 [MqttTask] mbedTLS: |2| 0x3ffd27f8: ssl->f_recv(_timeout)() returned 1395 (-0xfffffa8d)
4930 16584 [MqttTask] mbedTLS: |2| 0x3ffd27f8: in_left: 1400, nb_want: 5417
4931 16584 [MqttTask] mbedTLS: |2| 0x3ffd27f8: ssl->f_recv(_timeout)() returned 1400 (-0xfffffa88)
4932 16648 [MqttTask] mbedTLS: |2| 0x3ffd27f8: in_left: 2800, nb_want: 5417
4933 16648 [MqttTask] mbedTLS: |2| 0x3ffd27f8: ssl->f_recv(_timeout)() returned 1400 (-0xfffffa88)
I (167850) [vProcessMqttAgentDisconnect]: MQTT Return Value: 1
4934 16749 [MqttTask] mbedTLS: |2| 0x3ffd27f8: in_left: 4200, nb_want: 5417
4935 16749 [MqttTask] mbedTLS: |2| 0x3ffd27f8: ssl->f_recv(_timeout)() returned 1217 (-0xfffffb3f)
4936 16749 [MqttTask] mbedTLS: |2| 0x3ffd27f8: <= fetch input
4937 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: dumping 'input record from network' (5417 bytes)
4938 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0000:  16 03 03 15 24 02 00 00 51 03 03 d1 70 8d cf 6c  ....$...Q...p..l
4939 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0010:  af 5d e1 bf 1f 4b b2 50 5b d2 44 47 17 da fe 60  .]...K.P[.DG...`
4940 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0020:  9e 8e 55 5b f9 25 7f 0a b0 ef 48 20 63 71 7e 69  ..U[.H cq~i
4941 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0030:  2b 03 2f df d7 79 a3 1e 12 ad ac 33 43 17 f7 f6  +./..y.....3C...
4942 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0040:  0f 09 84 cb 75 6c ea a6 c6 54 43 17 c0 2f 00 00  ....ul...TC../..
4943 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0050:  09 ff 01 00 01 00 00 17 00 00 0b 00 13 10 00 13  ................
4944 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0060:  0d 00 05 a5 30 82 05 a1 30 82 04 89 a0 03 02 01  ....0...0.......
4945 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0070:  02 02 10 0c 62 ff 6c 06 6e ad 10 24 28 8a 6c f1  ....b.l.n..$(.l.
4946 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0080:  9d d4 dc 30 0d 06 09 2a 86 48 86 f7 0d 01 01 0b  ...0...*.H......
4947 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0090:  05 00 30 46 31 0b 30 09 06 03 55 04 06 13 02 55  ..0F1.0...U....U
[... cut ...]
4961 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0170:  19 78 ae 61 5d 62 db df 67 9e 90 8e 97 69 3f bb  .x.a]b..g....i?.
4962 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0180:  b6 a7 14 c4 82 af 46 b2 25 02 c8 cf 4c 06 30 44  ......F.
4963 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 0190:  da a4 7a bb 69 31 71 c2 9a 5c 1a a0 24 db c9 78  ..z.i1q..\..$..x
4964 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 01a0:  fa 57 28 93 9c c4 05 bf 0e d8 13 98 20 e0 ba df  .W(......... ...
4965 16749 [MqttTask] mbedTLS: |4| 0x3ffd27f8: 01b0:  3a 40 c0 0f 9a 30 e3 f3 75 bd d9 57 88 5d a0 86  :@...0..u..W.]..

sumit612, 2019-10-12:
My flash partition is like this:

# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x10000,  0x6000
otadata,  data, ota,     0x16000,  0x2000
phy_init, data, phy,     0x18000,  0x2000
ota_0,    0,    ota_0,   0x20000,  1M
ota_1,    0,    ota_1,   0x120000  ,       1M
factory,  0,    0,      0x220000 , 1M
storage,  data, nvs,     ,         0x10000
storage1,  data, spiffs,  ,        400K

sumit612, 2019-10-13:
I just found it connects from iot_mqtt_demo.c, but fails from any file from "vendors\espressif\boards\esp32\aws_demos\application_code\" directory.

sumit612, 2019-10-13:
*I found one big issue. It was mainly a stack overflow. After setting "Asymmetric in/out fragment length". The handshaking is complete. but after that there is an exception occurs regarding IOT thread. 

* Is it problem that I created task without using xTaskCreate() instead of Iot_CreateDetachedThread() function?

The exception console log is below

`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400901c2  PS      : 0x00060630  A0      : 0x8008d09d  A1      : 0x3ffdc7a0
0x400901c2: vListInsert at F:/solshare/amazon-freertos/freertos_kernel/list.c:150 (discriminator 1)

A2      : 0x3ffc70b0  A3      : 0x3ffd3ed0  A4      : 0x00007530  A5      : 0x3ffd965c
A6      : 0x3ffcfb7c  A7      : 0x3ffdc93c  A8      : 0x00000000  A9      : 0x3ffc70b8
A10     : 0x00000000  A11     : 0x00000014  A12     : 0x00000000  A13     : 0x400ed2cc
0x400ed2cc: _IotMqtt_ProcessSend at F:/solshare/amazon-freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_operation.c:832

A14     : 0x3ffcf990  A15     : 0x3ffcfb98  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

Backtrace: 0x400901c2:0x3ffdc7a0 0x4008d09a:0x3ffdc7c0 0x4008e7c3:0x3ffdc7e0 0x400e8ce2:0x3ffdc820 0x400e8cfd:0x3ffdc840 0x400eed89:0x3ffdc860 0x400eedd9:0x3ffdc880 0x400ecd68:0x3ffdc8b0 0x400eb2da:0x3ffdc8e0 0
x400e4b22:0x3ffdc910 0x400e3da2:0x3ffdc9b0 0x400e3f8d:0x3ffdc9d0 0x400e425e:0x3ffdca50
0x400901c2: vListInsert at F:/solshare/amazon-freertos/freertos_kernel/list.c:150 (discriminator 1)

0x4008d09a: vTaskPlaceOnEventList at F:/solshare/amazon-freertos/freertos_kernel/tasks.c:5068

0x4008e7c3: xQueueSemaphoreTake at F:/solshare/amazon-freertos/freertos_kernel/queue.c:2364

0x400e8ce2: prIotMutexTimedLock at F:/solshare/amazon-freertos/libraries/abstractions/platform/freertos/iot_threads_freertos.c:318

0x400e8cfd: IotMutex_Lock at F:/solshare/amazon-freertos/libraries/abstractions/platform/freertos/iot_threads_freertos.c:318

0x400eed89: IotTaskPool_Schedule at F:/solshare/amazon-freertos/libraries/c_sdk/standard/common/taskpool/iot_taskpool.c:850

0x400eedd9: IotTaskPool_ScheduleDeferred at F:/solshare/amazon-freertos/libraries/c_sdk/standard/common/taskpool/iot_taskpool.c:850

0x400ecd68: _IotMqtt_ScheduleOperation at F:/solshare/amazon-freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_operation.c:1072

0x400eb2da: IotMqtt_Connect at F:/solshare/amazon-freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c:1130

0x400e4b22: MQTT_AGENT_Connect at F:/solshare/amazon-freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_agent.c:608

0x400e3da2: connect_mqtt_agent at F:/solshare/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/mqttManager.c:1843

0x400e3f8d: vProcessWiFiConnect at F:/solshare/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/mqttManager.c:1843

0x400e425e: MqttTask at F:/solshare/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/mqttManager.c:1843

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6228
load:0x40078000,len:11684
ho 0 tail 12 room 4
load:0x40080000,len:6112
0x40080000: _WindowOverflow4 at F:/solshare/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

entry 0x40080330
0x40080330: _KernelExceptionVector at ??:?

`

sarenameas, 2019-10-14:
Hello Sumit612,

How did you discover you had a stack overflow? Did you end up in function: vApplicationStackOverflowHook()?
If your task did end up here, you could also update the stack in parameter *usStackDepth* instead of changing the TLS settings which may other have consequences. 

Also, changing any settings in this menu:

[image]

DOES change the mbedtls config file. If there any other settings in this menu you have updated, could you please send us the resulting sdkconfig.h generated file after building the code?  
- If you build using vendors\espressif\boards\esp32\aws_demos\Makefile (it looks like you are), then that will be under vendors\espressif\boards\esp32\aws_demos\build\include.

Also your mbedtls log seems to be missing information? Could you post the logs all the way up until the following log?
```
37 1270 [MqttTask] ERROR: Handshake failed with error code -9984
```

sumit612, 2019-10-14:
* When I ran the mqtt task with"Iot_CreateDetachedThread()" alike function "Iot_CreateMqttThread()" with other tasks running I got an error

`mbedTLS: |1| 0x3fffcc30: alloc(16701 bytes) failed`

* After commenting out other tasks allocation was successful. But with stack size of 10KB I got this

`7 822 [iot_thread] [INFO ][DEMO][8220] Successfully initialized the demo. Network type for the demo: 1
8 903 [iot_thread] mbedTLS: |2| 0x3ffe3424: => handshake
9 903 [iot_thread] mbedTLS: |2| 0x3ffe3424: client state: 0
10 903 [iot_thread] mbedTLS: |2| 0x3ffe3424: => flush output
11 903 [iot_thread] mbedTLS: |2| 0x3ffe3424: <= flush output
12 903 [iot_thread] mbedTLS: |2| 0x3ffe3424: client state: 1
13 903 [iot_thread] mbedTLS: |2| 0x3ffe3424: => flush output
14 903 [iot_thread] mbedTLS: |2| 0x3ffe3424: <= flush output
15 903 [iot_thread] mbedTLS: |2| 0x3ffe3424: => write client hello
16 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, max version: [3:3]
17 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, current time: 9
18 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: dumping 'client hello, random bytes' (32 bytes)
19 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: 0000:  00 00 00 09 aa f2 85 23 2d 15 59 b2 cc 72 05 41  .......#-.Y..r.A
20 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: 0010:  a9 87 e2 ec 09 c3 cf a5 32 d1 c0 fd 06 46 a5 da  ........2....F..
21 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, session id len.: 0
22 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: dumping 'client hello, session id' (0 bytes)
23 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c0ad
24 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c09f
25 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: 006b
26 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c00a
27 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c014
28 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: 0039
29 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c0af
30 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c0a3
31 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c02b
32 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c02f
33 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: 009e
34 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c0ac
35 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c09e
36 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c023
37 903 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c027
38 906 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: 0067
39 906 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c009
40 906 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c013
41 906 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: 0033
42 906 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c0ae
43 906 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c0a2
44 906 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: c09d
45 906 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: 003d
46 906 [iot_thread] mbedTLS: |3| 0x3ffe3424: client hello, add ciphersuite: 0035
93 909 [iot_thread] mbedTLS: |4| 0x3ffe3424: 00f0:  17 00 00 00 10 00 11 00 0f 0e 78 2d 61 6d 7a 6e  ..........x-amzn
94 909 [iot_thread] mbedTLS: |4| 0x3ffe3424: 0100:  2d 6d 71 74 74 2d 63 61 00 23 00 00              -mqtt-ca.#..
95 909 [iot_thread] mbedTLS: |2| 0x3ffe3424: => flush output
97 910 [iot_thread] mbedTLS: |2| 0x3ffe3424: ssl->f_send() returned 268 (-0xfffffef4)
109 935 [iot_thread] mbedTLS: |2| 0x3ffe3424: in_left: 0, nb_want: 5
110 935 [iot_thread] mbedTLS: |2| 0x3ffe3424: ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
111 935 [iot_thread] mbedTLS: |2| 0x3ffe3424: <= fetch input
112 935 [iot_thread] mbedTLS: |4| 0x3ffe3424: dumping 'input record header' (5 bytes)
113 935 [iot_thread] mbedTLS: |4| 0x3ffe3424: 0000:  16 03 03 15 24                                   ....$
114 935 [iot_thread] mbedTLS: |3| 0x3ffe3424: input record: msgtype = 22, version = [3:3], msglen = 5412
115 935 [iot_thread] mbedTLS: |2| 0x3ffe3424: => fetch input
116 935 [iot_thread] mbedTLS: |2| 0x3ffe3424: in_left: 5, nb_want: 5417
117 935 [iot_thread] mbedTLS: |2| 0x3ffe3424: in_left: 5, nb_want: 5417
118 935 [iot_thread] mbedTLS: |2| 0x3ffe3424: ssl->f_recv(_timeout)() returned 1395 (-0xfffffa8d)
119 955 [iot_thread] mbedTLS: |2| 0x3ffe3424: in_left: 1400, nb_want: 5417
120 955 [iot_thread] mbedTLS: |2| 0x3ffe3424: ssl->f_recv(_timeout)() returned 1400 (-0xfffffa88)
121 986 [iot_thread] mbedTLS: |2| 0x3ffe3424: in_left: 2800, nb_want: 5417
122 986 [iot_thread] mbedTLS: |2| 0x3ffe3424: ssl->f_recv(_timeout)() returned 1400 (-0xfffffa88)
123 1006 [iot_thread] mbedTLS: |2| 0x3ffe3424: in_left: 4200, nb_want: 5417
124 1006 [iot_thread] mbedTLS: |2| 0x3ffe3424: ssl->f_recv(_timeout)() returned 1217 (-0xfffffb3f)
125 1006 [iot_thread] mbedTLS: |2| 0x3ffe3424: <= fetch input
126 1006 [iot_thread] mbedTLS: |4| 0x3ffe3424: dumping 'input record from network' (5417 bytes)
127 1006 [iot_thread] mbedTLS: |4| 0x3ffe3424: 0000:  16 03 03 15 24 02 00 00 51 03 03 08 42 41 50 b8  ....$...Q...BAP.
[... cut ...]
#define CONFIG_ESPTOOLPY_BAUD_921600B 1
#define CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED 1
#define CONFIG_SPI_FLASH_ROM_DRIVER_PATCH 1
#define CONFIG_SPIFFS_PAGE_SIZE 256
#define CONFIG_MBEDTLS_ECP_DP_SECP192K1_ENABLED 1
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 1
#define CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT 3072
#define CONFIG_MONITOR_BAUD_OTHER_VAL 115200
#define CONFIG_NEWLIB_STDOUT_LINE_ENDING_CRLF 1
#define CONFIG_ESPTOOLPY_PORT "COM3"
#define CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS 1
`

sarenameas, 2019-10-15:
Hello Sumit,

I noticed you increased the TLS fragment length:
```
#define CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN 16384
```
This is double the size that Amazon FreeRTOS tests with. Is there a business need for this update? If there is not, I would recommend using all of the default mbedtls settings by undoing your changes to sdkconfig. 

Also my other question has not been answered. Did you reach the vApplicationStackOverflowHook() when you set a break point in there? 

Unfortunately the mbedtls logs can lag behind the application. To get exact error information at which part in the handshake we will need to insert a loop before your application ends:
```
while(1)
{
    vTaskDelay( 1000 );
}
```
And keep running until all of the mbedtls logs to to the handshake error have finished. There is no need to keep playing with the stack size.

Also on another note, some platforms can only support one MQTT connection active a time. Please make sure you only have ONE connection at a time in the system.

sumit612, 2019-10-15:
I got the issue. 

* If I connect to using my own it's not connecting. The is connecting wifi on it's own. So there must be a conflict.
* I had to look for different approach for connecting the wifi. Do you have suggestions on that? Should I start a new issue closing this current thread?

* I didn't change
```
#define CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN 16384

```
it was by default in the Kconfig in this directory "amazon-freertos\vendors\espressif\boards\esp32\aws_demos\application_code\espressif_code\mbedtls"

qiutongs, 2019-10-15:
Please open a new issue and close this one.

closed 2019-10-15 by sumit612