aws/amazon-freertos #1404: PC project: using random numbers before CRYPTO_Init() is called

htibosch, 2019-10-15:
**Describe the bug**
When running the PC demo project, I get an error saying:

`ERROR: DRBG is not initialized`

FreeRTOS+TCP want to start DHCP discovery, and therefore it needs a random ID. The random ID can not be given because `CRYPTO_Init()` has not been called yet.

The application then stops.

**System information**
- An ASUS laptop
- IDE : VS 2017
- Windows 10
- AWS/FreeRTOS sources: today's version (15 oct 2019)
- Project : PC ( projects\pc\windows\visual_studio\aws_demos )
`Code to reproduce the bug`: just enable `ipconfigUSE_DHCP` ( which is enabled by default )

**Expected behavior**
That a random number is obtained in the early DHCP stage

Possible solution: I called `CRYPTO_Init()` before calling `vTaskStartScheduler()` in stead of calling it when the IP-stack is up and running ( called by `SYSTEM_Init()` ).

The seems to work, but I don't know the dependencies of the `CRYPTO_Init()` on the scheduler and on a working TCP/IP stack.

Thanks, Hein

alexa-noxon, 2019-10-15:
Hi Hein, 

Thanks for pointing that out.  

#1401 should take care of the issue.   As you suspected, CRYPTO_Init() is needed before the DRBG can be initialized, and the DRBG must be initialized before ulRand may be called.

alexa-noxon, 2019-10-23:
Closing this issue - 11a517db38b99d5492361308fa94e2bdacd3ad50 reverts the CRYPTO updates and causes CRYPTO_Init to be called at time of first PKCS #11 use.

closed 2019-10-23 by alexa-noxon